guysoft/OctoPi #711: Making OctoPi build on Ubuntu 64bit so we have 64bit builds

guysoft, 2021-01-11:
Ok, so following what was going on in #692 . We are thinking about using Ubuntu to make 64bit builds. And also testing to see how using a distro with predictable release cycles works with OctoPi.

Current status:
* There is a build at: 
http://unofficialpi.org/Distros/OctoPi/nightly-arm64/2021-01-11_octopi-20.10-preinstalled-server-arm64+raspi-0.18.0.zip
md5: [``b0296e50919cc16d1d84bfb0e15e0b2c``](http://unofficialpi.org/Distros/OctoPi/nightly-arm64/2021-01-11_octopi-20.10-preinstalled-server-arm64+raspi-0.18.0.zip.md5)

It seems to load but
1. There is an Ubuntu user and also the Pi user (can be fixed). 
2. Pi user has no sudo
3. HA proxy does not load. I get:
```
/usr/sbin/haproxy -f /etc/haproxy/haproxy.cfg -S /run/haproxy-master.sock
[NOTICE] 010/111929 (4074) : haproxy version is 2.2.3-2
[NOTICE] 010/111929 (4074) : path to executable is /usr/sbin/haproxy
[ALERT] 010/111929 (4074) : parsing [/etc/haproxy/haproxy.cfg:35] : The 'reqrep' directive is not supported anymore since HAProxy 2.1. Use 'http-request replace-path', 'http-request replace-uri' or 'http-request replace-header' instead.
[ALERT] 010/111929 (4074) : parsing [/etc/haproxy/haproxy.cfg:36] : The 'reqadd' directive is not supported anymore since HAProxy 2.1. Use 'http-request add-header' instead.
[ALERT] 010/111929 (4074) : parsing [/etc/haproxy/haproxy.cfg:37] : The 'reqadd' directive is not supported anymore since HAProxy 2.1. Use 'http-request add-header' instead.
[ALERT] 010/111929 (4074) : parsing [/etc/haproxy/haproxy.cfg:43] : The 'reqrep' directive is not supported anymore since HAProxy 2.1. Use 'http-request replace-path', 'http-request replace-uri' or 'http-request replace-header' instead.
[ALERT] 010/111929 (4074) : Error(s) found in configuration file : /etc/haproxy/haproxy.cfg
[ALERT] 010/111929 (4074) : Fatal errors found in configuration.
```
Possible fix by @CRCinAU at: https://github.com/guysoft/OctoPi/issues/692#issuecomment-757891021

CRCinAU, 2021-01-11:
So what's broken / known problems at this point? Only the haproxy? Just about to give it a whirl...

guysoft, 2021-01-11:
I haven't tested more. Its about an hour old

CRCinAU, 2021-01-11:
Heh - noted - I won't overwrite my working RC setup then ;)

CRCinAU, 2021-01-11:
Ok, good news is that it seems to work well enough that I'm going to back up my RC1 config and load it into this on my faster USB stick.... It boots fine, resizes ok, starts octoprint ok, haproxy starts with my modified config as per the quoted post, connects to the printer ok....

CRCinAU, 2021-01-11:
Extra packages that can be removed:
* snapd (and rm -fR /run/snapd)
* squashfs-tools
* bcache-tools
* cloud-* (and rm -fR /etc/cloud)
* cryptsetup-*
* *nginx*

There may be more - but that's a quick overview...

DrDrache, 2021-01-11:
I will download this and test as well - I have been running RC1 64bit since you released it (and nightly before)
not having an aarch64 image (ubuntu or otherwise) for a pi4 host is a bad idea in my personal opinion.

the speed of the system and the interface IS significantly better (faster?); I do understand the comments about support/etc that was indicated in the previous thread. 

I only want to say, in 2021; sticking to 32bit only is short-sighted (PI Foundation, for instance).

testing here I go! :P

DrDrache, 2021-01-11:
based on my limited testing, I am unable to get my webcam working with mpegstreamer. I will spend more time on it... just adding to the discussion.

XboxOldie, 2021-01-11:
I have downloded the 64bit version and could start it on my raspberry Pi4 8GB.
With Putty I can access the system as user pi.
I was able to change the password but thats it.
After this I can't continue with any configuration because user pi is not allowed for root access.
Is there a Ubuntu admin user which will have this access or how should I be able to test further when I can't further configure my rasbperry to work correctly.
Any feedback would be great helping you to continue with testing.

dan-and, 2021-01-12:
@XboxOldie  see: https://github.com/guysoft/OctoPi/issues/712  ... there is also the ubuntu/ubuntu user on the current images. that user is allowed to sudo.

guysoft, 2021-01-12:
Next nightly should have the pi user able to run sudo. I hope so at least, testing appreciated!

XboxOldie, 2021-01-13:
@guysoft 
I have installed the latest nightly build now from 12.01.2021 and the user pi is having now sudo rights.
Thanks for the change

I could do then the password change , the sudo apt update and upgrade like always under the user pi.
All was running with a view messages and ended after a view minutes.
sudo raspi-config does not exist anymore on the RC2 so nothing i can do there.

However after all the system updates I was not able to open the octoprint webinterface on any browser like Opera or Edge.
Not with the IP nor with http;//octopi

Hope you can still solve the issue that I can continue testing.
I added the whole installation procedure I did as a text file for your reference

[Install octopi RC2.txt](https://github.com/guysoft/OctoPi/files/5805772/Install.octopi.RC2.txt)

Hope that helps.

PS: So far I have the RC1 with the 64bit running and all is working well on my Pi 4B 8GB

guysoft, 2021-01-13:
@XboxOldie See the first post. HAproxy needs to be fixed because it is not working on ubuntu, I don't know why:
https://github.com/guysoft/OctoPi/issues/711#issue-783301756

guysoft, 2021-01-13:
Seems like HAproxy on Raspbian/Rpios is ``1.8.19`` and on Ubuntu ``2.2``, which means the settings need to be upgraded.

CRCinAU, 2021-01-13:
> Seems like HAproxy on Raspbian/Rpios is `1.8.19` and on Ubuntu `2.2`, which means the settings need to be upgraded.

If you swap out the original two backend blocks and replace them with the two blocks I posted on the original ticket we discussed this, it works fine.

XboxOldie, 2021-01-13:
@guysoft Please keep me informed when HAproxy is solved into a new nightly distribution to continue testing.
I'm not a linux specialist only a normal user trying to inform you when all is working as expected.

guysoft, 2021-01-13:
@CRCinAU 

Ok, just to recap its comment https://github.com/guysoft/OctoPi/issues/692#issuecomment-757891021:

```
backend octoprint
        acl needs_scheme req.hdr_cnt(X-Scheme) eq 0

        http-request replace-path ^([^\ :]*)\ /(.*) \1\ /\2
        http-request add-header X-Scheme https if needs_scheme { ssl_fc }
        http-request add-header X-Scheme http if needs_scheme !{ ssl_fc }
        option forwardfor
        server octoprint1 127.0.0.1:5000
        errorfile 503 /etc/haproxy/errors/503-no-octoprint.http

backend webcam
        http-request replace-path ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8080
        errorfile 503 /etc/haproxy/errors/503-no-webcam.http
```
I am swamped and hope to test that soon. If anyone wants to test it here it would help me. I can also just push it to the nightly for now.

DrDrache, 2021-01-13:
> @CRCinAU
> 
> Ok, just to recap its comment [#692 (comment)](https://github.com/guysoft/OctoPi/issues/692#issuecomment-757891021):
> 
> ```
> backend octoprint
>         acl needs_scheme req.hdr_cnt(X-Scheme) eq 0
> 
>         http-request replace-path ^([^\ :]*)\ /(.*) \1\ /\2
>         http-request add-header X-Scheme https if needs_scheme { ssl_fc }
>         http-request add-header X-Scheme http if needs_scheme !{ ssl_fc }
>         option forwardfor
>         server octoprint1 127.0.0.1:5000
>         errorfile 503 /etc/haproxy/errors/503-no-octoprint.http
> 
> backend webcam
>         http-request replace-path ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
>         server webcam1  127.0.0.1:8080
>         errorfile 503 /etc/haproxy/errors/503-no-webcam.http
> ```
> 
> I am swamped and hope to test that soon. If anyone wants to test it here it would help me. I can also just push it to the nightly for now.

I  can verify that those codeblock changes allowed full access to the UI. I was unable to get my webcam working at all - so I cannot verify the webcam codeblock is 100%.  
in addition, I think the same type of changes should be done on the HLS webcam block? I am not sure.

XboxOldie, 2021-01-16:
@DrDrache  or @CRCinAU , Every day I set up a new system with the latest nightly build, but until now no nightly build was able to start the octoprint UI on the browser.
Are you able to make some changes in the build that my setups are  at least able to start Octoprint?
I also have a PI webcam running on my life system and will be able to provide feedback if accessable or not.

cp2004, 2021-01-18:
@XboxOldie If you need something stable without issues, don't use the nightly. I wouldn't expect every build to 'just work', especially considering it has just switched to Ubuntu rather than Raspberry Pi OS. There is a reason they are called 'nightly'.

Good feedback is 'It won't start OctoPrint and I got errors XYZ', or 'ABC will not load, it says D' - that will help them solve the issue quicker.

guysoft, 2021-01-18:
@XboxOldie Yes I can make the changes, however I am swamped this week. So unless someone else PRs (which they are certainly welcome). You will have to wait.

deejay2, 2021-01-19:
I've just installed the latest nightly build and got to do that to make octoprint start:

- Copy /etc/securetty from another ubuntu version cause it's not present on 20.10?
- Modify /etc/haproxy/haproxy.conf:
backend octoprint
        acl needs_scheme req.hdr_cnt(X-Scheme) eq 0

        http-request replace-path ^([^\ :]*)\ /(.*) \1\ /\2
        http-request add-header X-Scheme:\ https %[src] if needs_scheme { ssl_fc }
        http-request add-header X-Scheme:\ http %[src] if needs_scheme !{ ssl_fc }
        option forwardfor
        server octoprint1 127.0.0.1:5000
        errorfile 503 /etc/haproxy/errors/503-no-octoprint.http

backend webcam
        http-request replace-path ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8080
        errorfile 503 /etc/haproxy/errors/503-no-webcam.http

In these 2 sections, regexp stuff need to use new haproxy's way with http-request...

deejay2, 2021-01-20:
May I suggest to add this to /etc/sysctl.conf
vm.swappiness=0

Swapping should be used as a very last resort.

CRCinAU, 2021-01-21:
Ok - I'm starting not to be sold on the Ubuntu idea.... Mainly because as a desktop OS, there's a LOT that needs to be stripped out to get it anywhere near the Rasbian base.... Lots....

So, what other options are there? What is the official problem with the 64bit official distro? Are they just slow in updating?

guysoft, 2021-01-21:
@CRCinAU 
1. We are using the server version, most of it is already in place
2. There is no option to use the Raspios, please read the https://github.com/RPi-Distro/pi-gen/issues/481 . They are not to be used ATM.

The current blocker to get Ubuntu read is that I am swamped and have no time to fix and test the HAproxy settings to version 2.

CRCinAU, 2021-01-21:
I think there's a difference between "not to be used" and "are not considered stable".

Given the massive amount of work to get Ubuntu anywhere close to the current (even beta) pios distro, and the lack of time available, getting the fine tuning required in Ubuntu -> OctoPi seems like an even further away stretch goal.

guysoft, 2021-01-24:
@CRCinAU 64 bit builds are planned for next release, not the current RC. They are nightly builds, aka they are not tested, I don't know what happens when you run them. The nightly builds tend to become more stable as we near a new release. I am now planning to release 0.18.0 RC2 Monday, so I am not touching the Ubuntu builds just yet.
If you want to help getting them stable then you can make PRs and then others can test them too. I am quite confident we can reach stability as good as the RpiOS. We have had quite a lot of turbulence from the RpiOS unorganized release scheduling. And adding Ubuntu as a possibility might actually make things more stable in the log run. Unlike Ubuntu, the Rpi foundation have cleanly said they are not planing to maintain the 64bit just yet. So the Ubuntu 64bit is at the moment more stable and up to date than RpiOS.

Willmac16, 2021-01-24:
Changes I made to get the Ubuntu 64 bit version working:
+ `sudo apt update && sudo apt upgrade` I am pretty sure this didn't make a difference, but I did do it.
+ modify the haproxy file (`backend octoprint` matches above, `backend webcam` is different)
```
backend octoprint
        acl needs_scheme req.hdr_cnt(X-Scheme) eq 0

        http-request replace-path ^([^\ :]*)\ /(.*) \1\ /\2
        http-request add-header X-Scheme https if needs_scheme { ssl_fc }
        http-request add-header X-Scheme http if needs_scheme !{ ssl_fc }
        option forwardfor
        server octoprint1 127.0.0.1:5000
        errorfile 503 /etc/haproxy/errors/503-no-octoprint.http

backend webcam
        http-request replace-path /webcam/(.*) /\1
        server webcam1  127.0.0.1:8080
        errorfile 503 /etc/haproxy/errors/503-no-webcam.http

```
+ comment out `ConditionPathExists=/etc/octopi_streamer/mjpeg` in /etc/systemd/system/webcamd.service
+ reboot the pi

I will work to figure out how to get a pr together for this.

XRyu, 2021-02-11:
> Changes I made to get the Ubuntu 64 bit version working:
>
> * `sudo apt update && sudo apt upgrade` I am pretty sure this didn't make a difference, but I did do it.
> * modify the haproxy file (`backend [...]

Did exactly the same, can confirm interface works - going to try further.
Made the mentioned changes in ```/etc/haproxy/haproxy.cfg``` (i am totally new with this, so i had to figure out the path, adding here for completeness :D)

i would love to help with the PR, but i even dont know where in the repo i would need to edit / change these files. Maybe i find out :)

cp2004, 2021-02-11:
Haproxy config file is here:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/root/etc/haproxy/haproxy.cfg

XRyu, 2021-02-11:
Thank you @cp2004 ! :) Am i right when i am guessing - adding anything for the ubuntu-variant below the variant folder, adding some scripts to overwrite the raspios-default files to the bash-files in there?

XRyu, 2021-02-11:
I hope i did everything correctly :D PR incoming

cp2004, 2021-02-11:
> Thank you @cp2004 ! :) Am i right when i am guessing - adding anything for the ubuntu-variant below the variant folder, adding some scripts to overwrite the raspios-default files to the bash-files in there?

I don't know, I just know my way around all things OctoPrint to guide others... We will see what Guy says to review the PR.

guysoft, 2021-02-11:
It should do that. I am merging this, looks right. 
There is an issue that the ftp server for the nightlies is down. So will try see if I have a workaround so its easy to test things out.

Willmac16, 2021-02-11:
Thank you guys for figuring out how to make the PR (I got stuck trying to build the OS and worked on other projects for the past couple of weeks).

guysoft, 2021-02-21:
Was wondering if anyone found the time to test this in the end?

Willmac16, 2021-02-21:
I'll try it out now.

Willmac16, 2021-02-21:
UI and Webcam (mjpg) are up.

CRCinAU, 2021-02-21:
I did - but ended up going back to the Raspbian based images for 64 bit - they were just faster....

So I installed the last 64 bit raspbian image and then  just updated it to the latest of everything...

XRyu, 2021-02-21:
I did too, worked fine so far - I don’t know, after some fails with the x64
RC the Ubuntu build feels so much more stable. But totally subjective. But
did not have any downtime ever since! :) UI very responsive and Prints
absolutely flawless.

Willmac16, 2021-02-22:
The only issues I am having with the current ubuntu 64 bit build are trying to get WS281x LED setup since ubuntu doesn't have the gpio permissions baked in.

cp2004, 2021-02-22:
@Willmac16 interesting, if you could let me know when you do get it sorted I might be able to adjust some things to make it easier, or at least document it. And if you have more questions you know where I am...

I've not tried my plugin on Ubuntu yet, next release does make it more platform-agnostic hopefully.

XRyu, 2021-02-22:
Maybe you can try this: https://askubuntu.com/a/1313156
If this works for you, we might add it to the Ubuntu-Variant :D

guysoft, 2021-02-22:
Pushed to nightly, we can test

XboxOldie, 2021-02-22:
@guysoft 
Have installed nightly 2021-02-22_octopi-20.10-preinstalled-server-arm64+raspi-1.0.0
After copy of system to the SD and adding my WLAN access into the "octopi-wpa-supplicant.txt" the system started immediately and very fast into the WEB interface with the further setup screens.
After the plugin installation of "Creality-2x-temperature-reporting-fix" for my Ender 3 Pro, all is working perfect and fast.
Also camera is recognized and working normal.
I also did sudo update and upgrade +reboot without errors.
So far all works perfect and I will continue print tests with this new nightly version.

XRyu, 2021-02-22:
i can't tell how great it feels to read that - i can rarely contribute. This feels great to be a part of a nice community getting projects like this forward :D

Thank you for testing and the feedback @XboxOldie :D
Mine is running also fine, no stuttering of Octoprint crashing since then - with raspiOS i had regular downtimes and had to ssh into and needed to reboot.

Willmac16, 2021-02-23:
I made a pr #724 with the minimum set of changes I believe is required to get (the current version of) ws281x to work. @guysoft @cp2004

guysoft, 2021-02-23:
Tested here too. Boots, works well and the raspberry pi camera I have also works.

ping-localhost, 2021-02-24:
I'm currently testing/using the latest nightly build `2021-02-23_octopi-20.10-preinstalled-server-arm64+raspi-1.0.0.zip` on my RPI4 8GB and PSU control (and GPIO Control/anything that uses `RPi.GPIO`).

I've some logs available as a gist here: https://gist.github.com/ping-localhost/aad12f9619b13deaf5e2d6810407a0f5

guysoft, 2021-03-01:
@ping-localhost That error seem seems to be from the serial terminal or something like that and is unreleated.
Its crashing on some unicode error:
https://gist.github.com/ping-localhost/aad12f9619b13deaf5e2d6810407a0f5#file-octoprint-log-L21
This line says its using a [Unicode Character 'LEFT SINGLE QUOTATION MARK' (U+2018)](https://www.fileformat.info/info/unicode/char/2018/index.htm).
I am not sure where its coming from, but its not an OctoPi issue. Should fail anywhere.

Will try and see if RPi.GPIO can be shipped with the image.

ping-localhost, 2021-03-05:
> @ping-localhost That error seem seems to be from the plugin manager console or something like it might be unrelated.

I think you are right, because when I install the plugin directly through SSH, on the `pi` account and in the home dir, using `pip install "https://github.com/kantlivelong/OctoPrint-PSUControl/archive/master.zip"`, everything works as expected. 

However, I get a massive error when I run the following (which is what OctoPrint runs, but without the no cache dir):
```
/home/pi/oprint/bin/python -m pip --disable-pip-version-check install https://codeload.github.com/kantlivelong/OctoPrint-PSUControl/zip/master
```

Error it outputs: https://gist.github.com/ping-localhost/aad12f9619b13deaf5e2d6810407a0f5#file-terminal-log

guysoft, 2021-03-05:
@foosel looks like a plugin manager bug?

foosel, 2021-03-05:
@guysoft Negative, the plugin manager isn't involved anymore at this point, that's a failing `pip install` during intake of a dependency.

ping-localhost, 2021-03-05:
Here are my version, in case it helps (they are the same):

```
pi@octopi:~$ /home/pi/oprint/bin/python -m pip --version
pip 21.0.1 from /home/pi/oprint/lib/python3.8/site-packages/pip (python 3.8)
pi@octopi:~$ /home/pi/oprint/bin/python --version
Python 3.8.6 
```

```
pi@octopi:~$ python3 -m pip --version
pip 21.0.1 from /home/pi/.local/lib/python3.8/site-packages/pip (python 3.8)
pi@octopi:~$ python3 --version
Python 3.8.6 
```

werdnum, 2021-04-03:
> I'm currently testing/using the latest nightly build `2021-02-23_octopi-20.10-preinstalled-server-arm64+raspi-1.0.0.zip` on my RPI4 8GB and PSU control (and GPIO Control/anything that uses `RPi.GPIO`) fails to be installed due to the following error:
> 
> ```
> 2021-02-23 21:52:16,545 - octoprint.plugins.pluginmanager - ERROR - Reason: UnicodeEncodeError('latin-1', '  source/c_gpio.c: In function ‘setup’:\n', 31, 32, 'ordinal not in range(256)')
> ```
> 
> I've some logs available as a gist here: https://gist.github.com/ping-localhost/aad12f9619b13deaf5e2d6810407a0f5

As far as I can tell, this seems to be a silly error wherein the command-line output is being converted from Unicode to `latin1` to strip out the colour control characters, and the string doesn't work as `latin1` because the output contains non-`latin1` characters. I hacked around it in my installation by just editing the offending file (`/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py`) to change line 57 from this:

```python
 return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
```

to this:

```python
 return _ANSI_REGEX.sub(b"", line.encode("latin1", "namereplace")).decode("latin1", "namereplace")
```

and restarting octoprint. You could also use `ignore` instead of `namereplace` probably, but I didn't test that because it doesn't actually matter very much what the command-line output is, as long as it doesn't throw an error.

guysoft, 2021-04-07:
This is part of the nightly build and is planned to be released when we release.  So new bugs should go in a separate issue. Closing.

closed 2021-04-07 by guysoft