guysoft/OctoPi #712: nightly build: 64 bit 2021-01-12 - pi user not in sudoers

dan-and, 2021-01-12:
#### What were you doing?

I tried to install the new 0.18 RC2 64 Bit version: http://unofficialpi.org/Distros/OctoPi/nightly-arm64/2021-01-12_octopi-20.10-preinstalled-server-arm64+raspi-0.18.0.zip  on a raspberry pi4 

Booting and logging in as user pi works fine

#### What did you expect to happen?

I can login with the user pi and sudo to root  

#### What happened instead?

The user pi is not allowed to sudo 

I loaded the sdcard on another linux machine and added the user pi to /etc/sudoers, but I think that should not be the right way. 

Did I miss something?

#### Version of OctoPi

0.18.0 RC2 "nightly Build" http://unofficialpi.org/Distros/OctoPi/nightly-arm64/2021-01-12_octopi-20.10-preinstalled-server-arm64+raspi-0.18.0.zip

I have read the FAQ.

guysoft, 2021-01-12:
Yep, its an issue. I think the ubuntu user is generated from cloudinit. Will have to test how to remove it, and give user pi sudo.

https://askubuntu.com/questions/1183370/ubuntu-1910-server-install-on-raspberry-pi-4-hangs-on-cloud-init

I think this should make it work, will commit and you can test the nightly/executing in your account as ubuntu:
```
usermod -a -G sudo pi
```

chrislavender, 2022-02-01:
None of the above worked for me. Including trying to disable cloud init via the referenced askubuntu link. However, what did work was switching to the root user (after setting up a password for root), cd-ing into `/etc/sudoers.d` and adding the logged in user to the `90-cloud-init-users` file in the same way one would add them to the regular sudoers file. 

From what I can surmise it appears that `90-cloud-init-users` is overriding anything else with regard to sudo access.

guysoft, 2022-02-02:
Hey,
So I tested the latest nightly from today and sudo works on the pi user. So unless anyone has any other input I am closing this.
Also note that if you flash form rpi-imager with custom settings the image has booting issues and reaches this issue: https://github.com/raspberrypi/linux/issues/4458

guysoft, 2024-10-25:
Hey, this has been merged and tested, closing, reopen if there is an issue

closed 2024-10-25 by guysoft