guysoft/OctoPi #713: OctoPi 0.18.0rc2 Locale settings are GB

jvdbrg, 2021-01-19:
The new image locale settings are GB.

I've found out the hard way this can cause some trouble when generating passwords using special characters if you are doing this whith a screen and keyboard connected and then you'll spend ages finding a solution when trying to ssh into your PI.

Maybe it would be an idee to change this to US?

guysoft, 2021-01-19:
It was always GB because that is the default of Raspbian/Rpi OS.
I see no reason to change it to US over GB, and I am not in GB.

I am closing, since this is not a RC issue. If you feel like there is a reason to make is US or any other layout then you are welcome to comment.

closed 2021-01-19 by guysoft

jvdbrg, 2021-01-19:
In my country we use US keyboards (I live in NL, don't ask me why we use US keyboards :) ) so when I changed the pi password using a keyboard and monitor I had quite a struggle getting to log in using SSH because the characters above the numers change quite a bit when switching between US and GB.

But if Raspbian ships using GB you're right then there is no real need to change this.