guysoft/OctoPi #717: wpa_supplicant config is broken with Realtek driver

marcolivierroy, 2021-01-26:
#### What were you doing?

The Wifi stopped connecting updating from OctoPi 0.18.0 RC1 to OctoPi 0.18.0
I'm using a Realtek RTL8192cu usb dongle on a RPi 2.

#### What did you expect to happen?
It should have connected. The config is good and works manually.
#### What happened instead?
It did not but I found a workaround.
Add a Cron job as root to manually connect on boot:

```
sudo crontab -e
@reboot /sbin/wpa_supplicant -c /etc/wpa_supplicant/wpa_supplicant.conf -i wlx8cae4cf876bc -D wext
```
#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPi
OctoPi 0.18.0

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2021-01-26:
What is the output of dmesg when the wifi fails?

marcolivierroy, 2021-01-27:
It seems to stick to negociating an IPv6 address but IPv6 is not enabled on my network (and never will be).
It seems to associate to the AP but never get an IP addres from the DCHP server...

[    8.572456] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
[    8.622129] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
[    8.732825] usbcore: registered new interface driver r8188eu
[    9.041160] r8188eu 1-1.5:1.0 wlx8cae4cf876bc: renamed from wlan0
[   11.457737] random: crng init done
[   11.457777] random: 7 urandom warning(s) missed due to ratelimiting
[   12.000569] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   12.148697] 8021q: 802.1Q VLAN Support v1.8
[   12.543340] smsc95xx 1-1.1:1.0 enxb827ebf24489: hardware isn't capable of remote wakeup
[   13.230399] MAC Address = 8c:ae:4c:f8:76:bc
[   14.943136] R8188EU: assoc success
**[   14.949656] IPv6: ADDRCONF(NETDEV_CHANGE): wlx8cae4cf876bc: link becomes ready**
[   38.925461] uart-pl011 3f201000.serial: no DMA platform data
[36253.689000] R8188EU: ap recv disassoc reason code(8) sta:80:2a:a8:51:c8:cc
[36253.698866] R8188EU: indicate disassoc
[36255.413827] R8188EU: assoc success

guysoft, 2021-01-27:
1. Try connecting with an Ethernet cable - because that would make sure the DHCP is working correctly between the Pi and the router.
2. Make sure you are using a good power supply and a good cable, the versions might be drawing different power supplies and the wifi module might be failing at the moment it actually needs to communicate. Saw that happen before.

marcolivierroy, 2021-01-27:
DHCP works with a cable and power supply has been tested and this WiFi module has worked flawlessly for 2 months. It comes from distro updates. RC1 was working flawlessly until I apt dist-upgrade, decided to fix with flashing Release, no apt update and did not work out of the box. It really seems to be something with the driver/distro. I fixed it with _**wext**_, what is the default driver for wpa_supplicant ?

marcolivierroy, 2021-01-27:
Forgot to mention I had a usb wifi module that was pulling too much power from a Pi before and it was connecting properly but when the Pi came under a certain load, too much power was required by the usb controller and it made sd card read/write crash and the Pi at the same time to become unresponsive. To sum up, it would get a dhcp address.

trajesus, 2021-02-03:
Had the same problem, your solution with wext fixed it for me too, thank you.

guysoft, 2021-02-04:
> what is the default driver for wpa_supplicant ? nl80211 ?

Yes I think so. Also just noting that's a userspace driver and not a kernel driver.