guysoft/OctoPi #718: [latest nightly 64bit] Interface not reachable

MatzeWI, 2021-01-30:
#### What were you doing?

1. Flashed fresh image from nightly build
2. edited wpa settings
3. Turn on ras pi

#### What did you expect to happen?

Interface to show up

#### What happened instead?

State "Connection refused"

#### Did the same happen when running OctoPrint in safe mode?

Fresh Install so no plugin installed

#### Version of OctoPi

021-01-27_octopi-20.10-preinstalled-server-arm64+raspi-0.18.0.zip.md5

#### Printer model & used firmware incl. version

Snapmaker 2.0 A350 Latest FW

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

MatzeWI, 2021-01-30:
Some notes:
Octoprint seams to be running:
ps -ef@grep octoprint
pi          1686       1  3 14:47 ?        00:00:04 /home/pi/oprint/bin/python /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000

When wget index.html and then cat everything seems fine

I think haproxy is not working, as i can not find it in ps -ef

guysoft, 2021-01-30:
Please use the stable build 0.18.0 just came out. This is an out of date nightly

guysoft, 2021-02-22:
Try the latest nightly, this has been fixed

closed 2021-02-22 by guysoft