guysoft/OctoPi #720: wpa_supplicant works if called manually but not automatically on boot

mnesarco, 2021-02-03:
#### What were you doing?

Installing OctoPi on RaspBerryPi 3B Rev 1.2

1. Image flashed onto micro sd
2. /boot/octopi-wpa-supplicant.txt edited with correct network and country
2. Boot with HDMI monitor and keyboard
3. Wifi is not working so
4. I did: 
```
sudo killall wpa_supplicant
sudo wpa-supplicant -c/boot/octopi-wpa-supplicant.txt -Dwext -iwlan0
```
5. After that, WIFI works as expected
6. But after restart it didn't work untill I repeat step 4 manually

#### What did you expect to happen?

I want to use the PI in headless mode, so i need it to connect to wifi automatically

#### What happened instead?

It requires manual commands:
```
sudo killall wpa_supplicant
sudo wpa-supplicant -c/boot/octopi-wpa-supplicant.txt -Dwext -iwlan0
```

#### Did the same happen when running OctoPrint in safe mode?

I have not tested in safe mode, but i have a vanilla installation.

#### Version of OctoPi

octopi-buster-armhf-lite-0.18.0.zip

#### Printer model & used firmware incl. version

Tronxy XY2-pro, but it is irrelevant to the wifi problem.

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

trajesus, 2021-02-03:
Issue #717 already says that wpa-supplicant may not be working correctly.
It wasn't working for me either and that solved it. Try it as it could solve your case too 👍  (replacing wlx8cae4cf876bc in the command by the name of your interface which seems to be wlan0)

mnesarco, 2021-02-03:
Thank you, I will try it, but I am not using an external USB dongle.

trajesus, 2021-02-03:
I think it should work even if you are not using a dongle. You just need to run both commands described in the issue I mentioned (with the correct interface name) and reboot.

mnesarco, 2021-02-04:
It is very strange. It didn't work with the `cron` workaround, but it still works if done `manually`. I have double checked the `cron job`.

mnesarco, 2021-02-04:
I have solved it with a different approach. I just copied the modified `/boot/octopi-wpa-supplicant.txt` to `/boot/wpa_supplicant.conf` and it works, without any cron job.

closed 2021-02-04 by mnesarco