guysoft/OctoPi #721: [Request] Disable wifi sleep

hawkeyexp, 2021-02-09:
During using octoscreen i noticed sometimes reconnects by octoscreen which was caused by wifi sleep policy.

I disabled it with "iw dev wlan0 set power_save off" inside /etc/rc.local and the disconnects was gone.
I recommend disabling wifi sleep in sdcard image possible as ifup script etc :-)

guysoft, 2021-02-10:
That's really cool. I am trying to figure out where is the best place t place this.
[Here](https://forum.clockworkpi.com/t/how-to-disable-wifi-power-save-to-prevent-disconnects/933) its suggested at ``/etc/network/if-up.d/powesave_off`` before the ``exit 0`` line.

guysoft, 2021-02-10:
After trying to get it to work with ``/etc/network/if-up.d/powesave_off`` it seems that method just doesn't work anymore. So using rc.local. Hope some brave contributor comes along and figures out how to do this nicer, possibly taking other interface names in to account.

guysoft, 2021-02-10:
Fixed, needs to be tested

closed 2021-02-10 by guysoft