guysoft/OctoPi #725: [Request] ESXI-Arm Fling Support on 64 Bit Nightlies

MarkProminic, 2021-02-27:
#### What were you doing?

Desription:

I have currently setup the Raspberry Pi 4 that I was going to use for my  Ender 3 V2 using the ESXI-Arm fling for other testing and development purposes. I was able to get the OctoPi to boot (mostly) following these instructions:

https://www.virtuallyghetto.com/2020/10/how-to-run-raspberry-pi-os-as-a-vm-on-esxi-arm.html
https://www.virtuallyghetto.com/2020/11/how-to-run-kali-linux-on-esxi-arm.html

I eventually got to the OS booting, but it stops at an Initramfs prompt.  I tried to install linux-image-arm64, but it doesn't appear to be available for Ubuntu 20.10 for Arm, I am likely using the wrong package, and I have a hunch that if I get the right one, following the steps in the above guide, pluse using the right kernel image, would get me to being able to boot this in ESXI-Arm::

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

MarkProminic, 2021-02-28:
Hello,

I have actually got this to run, I used the wrong Nightly release. I wasn't sure what the difference was between:

2021-01-11_2020-08-20-octopi-buster-arm64-lite-0.18.0.zip

and 

2021-02-28_octopi-20.10-preinstalled-server-arm64+raspi-1.0.0.zip

with: 2021-01-11_2020-08-20-octopi-buster-arm64-lite-0.18.0.zip

 I was able to get OctiPi up and working on ESXI-Arm for Raspberry Pi. I will followup with more details.

closed 2021-02-28 by MarkProminic

guysoft, 2021-02-28:
Cool, thanks for updating