guysoft/OctoPi #727: Pluggin in BigTreeTech PiTFT v1.0 causes Raspberry Pi to turn off WiFi

NovaViper, 2021-03-05:
#### What were you doing?

1. Unplug Pi from power
2. Plug in TFT ribbon cable to the screen and pi
3. Power on Pi
4. Notice how the wifi doesn't turn on (no ip address appears on the terminal screen)

#### What did you expect to happen?
The pi should boot like normal to connect to wifi

#### What happened instead?
Whenever I have the TFT screen plugged into the pi when I boot, the wifi just completely turns off with no way to turn it back on

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPi
0.18.0

#### Printer model & used firmware incl. version
Creality Ender 3, Klipper

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2021-03-06:
1. How are you powering your pi?
2. which version of raspberrypi?
It sounds like a power issue. Where rhe screen draws power and then rhe wifi is the first thing to go down when under-powered.

NovaViper, 2021-03-06:
I'm using the official raspberry pi 2.5A psu. One thing I noticed was that when I unplugged the screen while it was turned on, the wifi immediately turned back on. I ended up grabbing another pi as I tried multiple different powers (even took some of my high powered type c chargers, converted them to micro USB via an adapter) and the same issue happened, even tried my NAS Pi 4 and it worked just fine with the screen.. I think my pi 3 might be defective

closed 2021-03-06 by NovaViper

NovaViper, 2021-03-08:
Wanted to give an update to this. I discovered that it wasn't my pi or the software, it was in fact the screen itself, it's not properly shielded and it interfered with the pi's wifi antenna

guysoft, 2021-03-08:
That was my second guess, but saw it was designed for the Pi. Guess that's not tested :-/