guysoft/OctoPi #735: [Request] support Raspbian Desktop.

spacecabbie, 2021-05-12:
Any change for Raspbian Desktop support ?
The Pi is to limited the more you do with it (like klipper) 

Raspbian desktop with for example (https://learn.pi-supply.com/make/getting-started-with-the-rtk-gpio-board/)
Can be a good alternative.

guysoft, 2021-05-12:
There is a script to install raspsbian desktop on OctoPi.

[/home/pi/scripts/install-desktop](https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/home/pi/scripts/install-desktop)

guysoft, 2021-05-23:
Closing no reply

spacecabbie, 2021-05-24:
Oww sorry din't see your reply before but its not the same I am talking about: Raspberry Pi OS:
https://www.raspberrypi.org/software/operating-systems/

guysoft, 2021-05-24:
OctoPi is already built on top of RaspberryPi OS using CustomPiOS. I has a script that lets you install the full desktop of RPiOS of you need it. And its not shipped by default because it would bloat the image by a few GB and would slow performance.

So I am not sure what exactly you are asking for apart from that.

spacecabbie, 2021-05-24:
Yes but I do not mean the default raspberrypi Os but the one for desktop "Raspberry Pi Desktop" mean to be installed on x86 architecture.

Willmac16, 2021-05-24:
Octoprint will run on pretty much anything. You just need to do a manual install [click here and scroll down to installing manually](https://octoprint.org/download). The gpio isn't used by octoprint core; however, many plugins are written with the target user-base of raspberry pi users, so you may need to find ways to get specific plugins to work with that external gpio usb device that you linked.

spacecabbie, 2021-05-24:
From the run around I am getting I take it means no it will not be supported just say that then.

closed 2021-05-24 by spacecabbie