guysoft/OctoPi #737: wpa_supplicant missing from /boot

mathpunk, 2021-05-28:
#### What were you doing?

Trying to set up WiFi after having flashed an SD card with Etcher.

1. download OctoPi stable
2. install Etcher
3. flash 32gb SD card with the unzipped img
4. mount the Linux partition
5. cd to the mount point
6. examine boot and etc/wpa_supplicant directories

#### What did you expect to happen?
I expected to find a file that looked similar to [cp2004/octopi-wpa-supplicant.txt] 
(https://gist.github.com/cp2004/5cb0361fb872fc779fb8272ad7f4887f).

#### What happened instead?

```
$ ls boot
```
yielded no output.

```
$ ls -l etc/wpa_supplicant/
total 40
-rwxr-xr-x 1 root root   937 Mar 24  2020 action_wpa.sh
-rw-r--r-- 1 root root 25569 Mar 24  2020 functions.sh
-rwxr-xr-x 1 root root  4696 Mar 24  2020 ifupdown.sh
lrwxrwxrwx 1 root root    31 Jan 10 18:02 wpa_supplicant.conf -> /boot/octopi-wpa-supplicant.txt
```

My console highlights that link in angry colors. Should it be pointing at `boot/...` rather than `/boot/...`?

#### Did the same happen when running OctoPrint in safe mode?

n/a

#### Version of OctoPi

OctoPi 0.18.0

#### Printer model & used firmware incl. version

n/a

#### Screenshot(s)/video(s) showing the problem:

n/a

I have read the FAQ, in particular https://community.octoprint.org/t/wifi-setup-and-troubleshooting/184

guysoft, 2021-05-28:
You go to either /boot or boot/ depending on where you are accessing it from.
/boot assumes you have booted the Pi with OctoPrint and that you have /boot mounted.

boot/ assumes you have thr sd card mounted to a system and you are editing it. Which is the common issue because people set up wifi before your boot thr device.

Closing because this is documented on countless guides, videos and the Readme.

closed 2021-05-28 by guysoft

mathpunk, 2021-05-28:
I'm looking at the readme, and what I can see is, 

> 2. Configure your WiFi by editing octopi-wpa-supplicant.txt on the root of the flashed card when using it like a thumb drive

I'd like to zoom in on my steps here just so we're sure we're speaking the same language. I'll be happy to submit a pull request with more documentation for people who got stuck here.

After flashing the drive:

```
$ cd ext
$ ls boot
$ ls etc/wpa_supplicant
action_wpa.sh  functions.sh  ifupdown.sh  wpa_supplicant.conf
$ cat etc/wpa_supplicant/wpa_supplicant.conf
cat: etc/wpa_supplicant/wpa_supplicant.conf: No such file or directory
```

Is my confusion, there isn't supposed to be anything and I'm meant to add it? If so, can there be a link to an example wpa_supplicant.conf?

mathpunk, 2021-05-28:
I'm going to try pasting in https://gist.githubusercontent.com/cp2004/5cb0361fb872fc779fb8272ad7f4887f/raw/5407a8121e6d5724a8bb9568930c7df0b08aa8b3/octopi-wpa-supplicant.txt to `boot` and see how that goes. I do see that it is labeled as being for 0.17.0 though.

mathpunk, 2021-05-28:
That didn't go so well and I'm trying to install 0.18.0, rather than the 0.15.0 that I arrived on via the https://octoprint.org/download/ page

guysoft, 2021-05-28:
> 
> 
> I'm looking at the readme, and what I can see is,
> 
> > 1. Configure your WiFi by editing octopi-wpa-supplicant.txt on the root of the flashed card when using it like a thumb drive
> 
> I'd like to zoom in on my steps here just so we're sure we're speaking the same language. I'll be happy to submit a pull request with more documentation for people who got stuck here.
> 
> After flashing the drive:
> 
> ```
> $ cd ext
> $ ls boot
> $ ls etc/wpa_supplicant
> action_wpa.sh  functions.sh  ifupdown.sh  wpa_supplicant.conf
> $ cat etc/wpa_supplicant/wpa_supplicant.conf
> cat: etc/wpa_supplicant/wpa_supplicant.conf: No such file or directory
> ```
> 
> Is my confusion, there isn't supposed to be anything and I'm meant to add it? If so, can there be a link to an example wpa_supplicant.conf?

I gave an explanation about paths when on the Pi and not on it, you are just pasting paths without understanding how they work. I really don't think a PR is required because I have seen already hundred of thousands people managing to Install this.

mathpunk, 2021-05-28:
Are you telling me to edit this file only when it is already on the Pi, and NOT when the card is mounted on my laptop?

mathpunk, 2021-05-28:
There's no file to edit, when it's mounted on my laptop, is what I'm trying to tell you, while the directions I read *seemed* to indicate that I'm supposed to set up WiFi before the card goes into the Pi.

mathpunk, 2021-05-28:
For others who got stuck here: I had mounted and searched the Linux partition -- hey, Raspbian is Linux right? Instead, mount the W95 partition.