guysoft/OctoPi #739: Missing logrotate rules for kern.log and syslog

tkolo, 2021-06-16:
#### What were you doing?
I'm running octopi on raspbbery backed by 64GB sdcard for quite a while now. Today, refused to turn on. After a brief troubleshooting session, I noticed it's out of space. The main culprit:

```
pi@octopi:~ $ ls -alhS /var/log/ 
total 55G
-rw-r-----  1 root adm   28G Jun 14 12:17 kern.log
-rw-r-----  1 root adm   27G Jun 16 17:03 syslog.1
...
```

#### What did you expect to happen?
I expected logrotate rules to be in place. Alternatively, perhaps whole /var/log should be mounted as tmpfs as not to wear out SD card.

#### What happened instead?
It ran out of space

#### Did the same happen when running OctoPrint in safe mode?
N/A

#### Version of OctoPi
1.6.1, according to latest log file

#### Printer model & used firmware incl. version
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.

cp2004, 2021-06-19:
AFAIK, logrotate is already in place, and besides, this is not managed by OctoPi and instead the underlying Raspberry Pi OS. Run `cat /etc/logrotate.d/rsyslog` and have an explore through `/etc/logrotate.d/` & stuff generally

This is a common issue with broken wifi connections, something in the drivers/rpi kernel somewhere *fills* the log files with endless error on error as it is broken. Don't ask why, how, or when it will be fixed but this has turned up on the forums/discord every now and then. You can usually delete the files, reboot and you will be OK.

guysoft, 2021-06-19:
Indeed, I am not sure what can be added here to fix it

tkolo, 2021-06-25:
Thanks guys, I don't know how I've missed the logrotate rules for those files the first time.
Indeed I have Wi-Fi issues (for quite some time now) and the log was full of driver-related errors. As resolving isn't related to this issue, I'll close this now, the logrotate rules seem fine to me.

closed 2021-06-25 by tkolo