guysoft/OctoPi #745: Wi-Fi settings not working on fresh 0.18.0 installation

patoberli, 2021-07-04:
Hello
I've done a fresh installation of Octoprint 0.18.0 from here https://octoprint.org/download/. I've burned it with BalenaEtcher to SD, put it into PI, booted it up, logged in and edited with VIM (command vi) the /boot/octopi-wpa-supplicant.txt and rebooted. Wireless didn't work. I did various variants (with spaces in front of ssid and psk, tabs, commented my config out and removed the comments from the sample and configured that) but none worked.
In the end I run raspi-config and made my wireless configuration there and it instantly worked and is working since. No idea what went wrong, maybe the octopi-wpa-supplicant.txt file is only read at the very first boot?
Oh and sorry for deleting the template. I just posted this in case to see if anybody else might also have this.

Regards
Patrick

guysoft, 2021-07-04:
If you can't provide it, I have no way of determining what is wrong.
The file has been used thousands of time at this point, so I would imagine there is a high chance of a syntax error or typo.

patoberli, 2021-07-04:
The file (I removed the password):
`cat /boot/octopi-wpa-supplicant.txt 
# Use this file to configure your wifi connection(s).
#
# Just uncomment the lines prefixed with a single # of the configuration
# that matches your wifi setup and fill in SSID and passphrase.
#
# You can configure multiple wifi connections by adding more 'network'
# blocks.
#
# See https://linux.die.net/man/5/wpa_supplicant.conf
# (or 'man -s 5 wpa_supplicant.conf') for advanced options going beyond
# the examples provided below (e.g. various WPA Enterprise setups).
#
# !!!!! HEADS-UP WINDOWS USERS !!!!!
#
# Do not use Wordpad for editing this file, it will mangle it and your
# configuration won't work. Use a proper text editor instead.
# Recommended: Notepad++, VSCode, Atom, SublimeText.
#
# !!!!! HEADS-UP MACOSX USERS !!!!!
#
# If you use Textedit to edit this file make sure to use "plain text format"
# and "disable smart quotes" in "Textedit > Preferences", otherwise Textedit
# will use none-compatible characters and your network configuration won't
# work!

## WPA/WPA2 secured
network={
  ssid="ludicrous name"
  psk="password"
}
#network={
# ssid="ludicrous name"
# psk="password"
#}

## Open/unsecured
#network={
#  ssid="put SSID here"
#  key_mgmt=NONE
#}

## WEP "secured"
##
## WEP can be cracked within minutes. If your network is still relying on this
## encryption scheme you should seriously consider to update your network ASAP.
#network={
#  ssid="put SSID here"
#  key_mgmt=NONE
#  wep_key0="put password here"
#  wep_tx_keyidx=0
#}

# Uncomment the country your Pi is in to activate Wifi in RaspberryPi 3 B+ and above
# For full list see: https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
country=GB # United Kingdom
#country=CA # Canada
#country=DE # Germany
#country=FR # France
#country=US # United States

### You should not have to change the lines below #####################

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1`

And this is the wpa_supplicant.conf which was probably created by raspi-config (it was not containing the network part before using raspi-config):
`cat /etc/wpa_supplicant/wpa_supplicant.conf 
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=GB

network={
        ssid="ludicrous name"
        psk="password"
}`

The only difference I see here is the sequence of the comands in the files.

guysoft, 2021-07-04:
Can't see anything, I dont know

patoberli, 2021-07-05:
No problem, I also didn't know what was/is the reason. Anyway, I hope it helps somebody else if it isn't working.
Simply use 'sudo raspi-config' and setup the wireless configuration there.

closed 2021-07-05 by patoberli