guysoft/OctoPi #746: incorrect path for octopi (or link missing) on 64b Nightlies

Thelvaen, 2021-07-17:
Hello,

The 64b Octopi distribution, is Ubuntu based, hence the "boot" partition, is mounted in /boot/firmware and not in /boot.

Thus system unit/script that relies on the /boot/octopi.txt like webcamd or streamer_select will not find the file where it's expected.

The easiest solution, is to symlink /boot/octopi.txt to /boot/firmware/octopi.txt.

A maybe more reliable solution would be to have scripts like /root/bin/streamer_select and /root/bin/webcamd include a list of possible directory, test where the file is and then use the correct file.

guysoft, 2021-07-17:
It would br better practice to have the file at /boot/octopi.txt and link to that, so when you open thr file on windows its in the right place

Thelvaen, 2021-07-17:
yeah, but as I said, the /boot does not match the boot FAT partition you can access from windows, as it is mounted by Ubuntu as /boot/firmware. I'm not sure it can be modified on Ubuntu.

```
root@mk3s:~# mount | grep boot
/dev/mmcblk0p1 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
root@mk3s:~#
```

guysoft, 2021-07-20:
Ok, then as I can see it not something I can fix, you should talk to Ubuntu about that. 
All I can do is use their existing image.
Anyway you should be editing the boot partition before booting the pi. Because otherwise how would you connect to it?
There in both cases the files are located in the root of the boot folder.

Thelvaen, 2021-07-20:
yeah but you can still easily create a simlink /boot/octopi.txt during first boot, that will point to /boot/firmware/octopi.txt

guysoft, 2021-07-20:
Yea, I guess we could do that. Since in the unix context it should work

guysoft, 2021-08-08:
THe commit broke the build and I had to revert it.

guysoft, 2021-08-08:
So I checked, this issue is this.
CustomPiOS build mounts partition 0 in to ``/boot``. But on ubuntu on the pi it mounts to /firmware/boot.

If you open the sdcard partition 0 on windows or something like that the files you want to edit show up on the root location of the partition, like in all OctoPi installer.

However, if you boot he device and edit the wifi settings from there its in ``/boot/firmware.txt``.

In order to solve this issue it will be needed to change the [``mount_image``](https://github.com/guysoft/CustomPiOS/blob/devel/src/common.sh#L168) function in CustomPiOS to handle Ubuntu and mount it in to a different location, then also handle changes in the [``network``](https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/network/start_chroot_script) module because of that. So its a slight rewrite of that is going on there.

Will have to make the time for that, and it might break other modules so will need to test it out.

commit 55344968 (2021-07-26): Fixes #746