guysoft/OctoPi #747: incorrect option in webcamd service unit in the Nightlies 64b

Thelvaen, 2021-07-17:
Hello,

/etc/systemd/system/webcamd.service is defined as follow :
```
[Unit]
Description=the OctoPi webcam daemon with the user specified config
# ConditionPathExists=/etc/octopi_streamer/mjpeg

[Service]
WorkingDirectory=/root/bin
StandardOutput=append:/var/log/webcamd.log
StandardError=append:/var/log/webcamd.log
ExecStart=/root/bin/webcamd
Restart=always
Type=forking
RestartSec=1

[Install]
WantedBy=multi-user.target
```

but process `/root/bin/webcamd` does not fork, hence provoking systemd to think that the unit has failed and restarting it over and over again.

```
syslog:Jul 17 15:30:40 mk3s systemd[1]: webcamd.service: Succeeded.
syslog:Jul 17 15:32:10 mk3s systemd[1]: webcamd.service: start operation timed out. Terminating.
syslog:Jul 17 15:32:10 mk3s systemd[1]: webcamd.service: Failed with result 'timeout'.
syslog:Jul 17 15:32:11 mk3s systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 1.
syslog:Jul 17 15:33:41 mk3s systemd[1]: webcamd.service: start operation timed out. Terminating.
syslog:Jul 17 15:33:41 mk3s systemd[1]: webcamd.service: Failed with result 'timeout'.
syslog:Jul 17 15:33:43 mk3s systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 2.
syslog:Jul 17 15:35:13 mk3s systemd[1]: webcamd.service: start operation timed out. Terminating.
syslog:Jul 17 15:35:13 mk3s systemd[1]: webcamd.service: Failed with result 'timeout'.
syslog:Jul 17 15:35:14 mk3s systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 3.
syslog:Jul 17 15:36:44 mk3s systemd[1]: webcamd.service: start operation timed out. Terminating.
syslog:Jul 17 15:36:44 mk3s systemd[1]: webcamd.service: Failed with result 'timeout'.
syslog:Jul 17 15:36:46 mk3s systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 4.
syslog:Jul 17 15:38:16 mk3s systemd[1]: webcamd.service: start operation timed out. Terminating.
syslog:Jul 17 15:38:16 mk3s systemd[1]: webcamd.service: Failed with result 'timeout'.
syslog:Jul 17 15:38:17 mk3s systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 5.
syslog:Jul 17 15:39:47 mk3s systemd[1]: webcamd.service: start operation timed out. Terminating.
syslog:Jul 17 15:39:47 mk3s systemd[1]: webcamd.service: Failed with result 'timeout'.
syslog:Jul 17 15:39:49 mk3s systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 6.
syslog:Jul 17 15:41:19 mk3s systemd[1]: webcamd.service: start operation timed out. Terminating.
syslog:Jul 17 15:41:19 mk3s systemd[1]: webcamd.service: Failed with result 'timeout'.
syslog:Jul 17 15:41:20 mk3s systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 7.
```

Removing `Type=forking` from the unit definition will avoid that behavior.

cp2004, 2021-07-17:
Possibly a change introduced in #741?

Thelvaen, 2021-07-17:
Possible, but I'm not familiar enough with the code to be sure of that.

guysoft, 2021-07-18:
@cp2004 @Thelvaen If its introduced by #741, then this nightly should not have the issue. I twas built a month ago and that commit is from 19 days ago:
https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2021-06-17_octopi-20.10-preinstalled-server-arm64+raspi-1.0.0.zip
(md5 [848cc709ab5f2c5eb0c6819636f80c36](https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2021-06-17_octopi-20.10-preinstalled-server-arm64+raspi-1.0.0.zip.md5))
So if you have the time to test @Thelvaen we will know.

Thelvaen, 2021-07-18:
I'm getting the same behavior with that build.

```
syslog:Jul 18 10:58:26 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 74.
syslog:Jul 18 10:58:28 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:29 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 75.
syslog:Jul 18 10:58:30 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:31 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 76.
syslog:Jul 18 10:58:32 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:33 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 77.
syslog:Jul 18 10:58:34 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:35 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 78.
syslog:Jul 18 10:58:37 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:38 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 79.
syslog:Jul 18 10:58:39 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:40 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 80.
syslog:Jul 18 10:58:41 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:42 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 81.
syslog:Jul 18 10:58:43 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:44 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 82.
syslog:Jul 18 10:58:46 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:47 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 83.
syslog:Jul 18 10:58:48 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:49 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 84.
syslog:Jul 18 10:58:50 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:51 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 85.
syslog:Jul 18 10:58:52 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:53 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 86.
syslog:Jul 18 10:58:55 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:56 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 87.
syslog:Jul 18 10:58:57 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:58:58 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 88.
syslog:Jul 18 10:58:59 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:59:00 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 89.
syslog:Jul 18 10:59:01 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:59:02 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 90.
syslog:Jul 18 10:59:04 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:59:05 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 91.
syslog:Jul 18 10:59:06 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:59:07 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 92.
syslog:Jul 18 10:59:08 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:59:09 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 93.
syslog:Jul 18 10:59:10 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:59:11 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 94.
syslog:Jul 18 10:59:13 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:59:14 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 95.
syslog:Jul 18 10:59:15 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:59:16 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 96.
syslog:Jul 18 10:59:17 octopi systemd[1]: webcamd.service: Succeeded.
syslog:Jul 18 10:59:18 octopi systemd[1]: webcamd.service: Scheduled restart job, restart counter is at 97.
syslog:Jul 18 10:59:19 octopi systemd[1]: webcamd.service: Succeeded.
```

Thelvaen, 2021-07-18:
Ok, I've just tested on the 32b stable version.

The `/root/bin/webcamd` exits after forking, but it does not give systemd proper information, hence doing a `systecmtl stop webcamd` does not kill the mjpg_streamer process. Hence, I do prefer the new `/root/bin/webcamd` script because at least systemctl is able to properly kill the process.

Do you want me to do a PR to correct the Unit file, or are you going to do it directly (a PR for just that seems overkill)?

guysoft, 2021-07-18:
If you know how to solve this in a PR that would be great!
I am not sure what you are going to change in order to fix it, so worst case I will have input after review.

Thelvaen, 2021-07-18:
I've close my PR, i've realized something was not working as it should (the process detection is not good).

I'll dive in the script and see what I can do from there.

guysoft, 2022-02-27:
Hey, just tested and it seems like the webcamd works both for Rpi cameras and also usb cameras