guysoft/OctoPi #749: [enhancement] rearrange project

Thelvaen, 2021-07-18:
Hello,

In order to not use /root/bin & /home/pi, it would be interesting to rearrange the project to deploy in /opt instead of /home.

It could also be done with a dedicated user instead of using the "pi" user which might have extended privileges, not needed for the application.

guysoft, 2021-08-08:
I understand that complies better with debian directory tree convention ([source](https://www.debian.org/releases/stable/amd64/apcs02.en.html)). However OctoPi has been using this structure for over 7 years and changing it would mean breaking all guides.

I would have considered doing that, but I see no benefit to the user of changing the folder structure.

Thelvaen, 2021-08-08:
one clear benefit from using a dedicated user instead of the user "pi", would be to make the Octopi distribution more robust against an external attacker, user "pi" is usually sudo without password, hence it can do whatever it wants on the system.

I know that the feeling is that once the bad guys are inside the Octopi, they can burn your house and most people working around 3D printer feel like it is the worse case scenario, but it might not be, someone taking access to your Raspberry Pi, and having full access to it, could also install a rogue DHCP/DNS server and get all your data/personnal information leaked might be a disaster too.

Thelvaen, 2021-08-08:
besides, those guides are not adapted to something other than the Raspberry Pi, hence standardising the installation process, would also brings benefits, as it would allow users to more easily install on other hardware (other SBC, PCEngines APU, or an old NUC/Laptop for exemple).

guysoft, 2021-08-08:
@foosel any preference?

foosel, 2021-08-09:
I've been thinking about this again and again throughout the years myself and always decided against it because of the whole "these instructions used to work and now they don't" overhead this would create. 

It would be the clean way to do it, but I'm frankly full of utter dread at the thought of the support intense switchover period that is probably going to stretch over at least a year if not more.

hbast, 2021-08-31:
Just because you've always done it that way, it shouldn't prevent you from doing something new. I support the proposal to create a clean project structure and move octoprint out of the user directory.

In the meantime you can help yourself with hints at the ssh login, or with softlinks to the new directory. An empty file with the hint "look under /opt..." would also be an option.

foosel, 2021-09-14:
You overestimate the ability or willingness of users to read, and you severely underestimate the kind of abuse frustrated users will sling at maintainers. I've gotten requests to off myself over minor workflow changes, I don't want to imagine what would happen here.

I agree that a clean(er) project structure would be nice, but as I said, I feel utter dread at the thought of the switchover period and the associated storm of support overhead. And frankly, the people asking for changes like this one are usually not there to witness the consequences, making it easy to say stuff like "Just because you've always done it that way, it shouldn't prevent you from doing something new".

IF we decide to burn bridges like that, we might as well think whether the current approach of basing stuff off the Raspberry Pi Lite image is really the best approach, or if we maybe rather want to go with an absolute minimal debootstrap instead that gives us *full* control over installed packages (in the past the lite image has gotten more and more bloat added to it again), or if possibly there might be even better solutions with easier upgrade paths out there. My point being, if I absolutely have to suffer through it, I really don't want to go through this horror of a transition period more than once if I can help it.