guysoft/OctoPi #751: OctoPi starts with an IP (169....) when using LAN interface and does not DHCP

PatMis16, 2021-09-15:
#### What were you doing?

1. Wrote the OctoPi Image to an MicroSD Card with the Raspberry Imager software
2. Started the Raspberry Pi (A Raspberry Pi 2B which was laying arround) with LAN connected
3. Tried to access the RPI via SSH, did not work
4. Connected HDMI and checked the system. The check revealed that the interface eth0 is using an IP 169.xxx.xxx.xxx and not DHCP provided by my router (address range 192.168.8.0/24).
5. Tried to enable DHCP
6. Tried to use a fixed IP in range 192.168.8.0/24

None of the steps worked. 

#### What did you expect to happen?
The interface eth0 gets an IP assigned by DHCP in the given range. 

#### What happened instead?
DHCP did not work properly, when assigning a fixed IP the octoPi was not accessible either. 

#### Did the same happen when running OctoPrint in safe mode?
Did not run OctoPrint in safe mode, since it is not accessible in the first place.

#### Version of OctoPi
0.18.0

#### Printer model & used firmware incl. version
I intend to use it with a Creality Ender 3 V2.

#### Screenshot(s)/video(s) showing the problem:
None

I have read the FAQ.

cp2004, 2021-09-15:
I don't think this is really an OctoPi issue, but more one with the Raspberry Pi OS image that it is built on. You should test if you have the same issue with Raspberry Pi OS, instead of OctoPi. It is a newer image, so if you want to test like-for-like test a nightly build of OctoPi as well.

cp2004, 2021-09-15:
A quick check of Google actually showed that a 169.x address would be assigned when the computer cannot connect to the network. So you should check that that's configured correctly as well. If you need help with that, reach out on the OctoPrint community forums or discord server.

guysoft, 2021-11-10:
Can this be closed? was it solved?

PatMis16, 2021-11-11:
Hi Guy

Yes, this can be closed. 

Regards,
Patrick

closed 2021-11-11 by guysoft