guysoft/OctoPi #756: "No wireless interfaces found" on RPi 3B v1.2

qwertymodo, 2021-11-08:
#### What were you doing?

1. Fresh install of OctoPi 0.18.0 from the official installation image on a Raspberry Pi 3 Model B (v1.2)
2. Configure octopi-wpa-supplicant.txt with network config copied from a previous working installation (running 0.17.0)
3. Boot the Pi
4. No network connectivity
5. Run raspi-config > 5 Localisation Options > L4 WLAN Country
6. Result is "No wireless interface found"
7. Run ifconfig
8. Interfaces eth0 and lo are the only ones listed, no listing for wlan0 (or any wlanX interfaces)

#### What did you expect to happen?
I expected WiFi to work properly, or at the very least for the wlan0 device to be detected so I could troubleshoot further.

#### What happened instead?
WiFi does not connect, and the OS doesn't seem to believe that any wireless interface exists at all, despite the one built into the board.  This seems to be a problem with the underlying OS image used, and nothing to do with OctoPrint.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPi

0.18.0

#### Printer model & used firmware incl. version

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

[X] I have read the FAQ.

guysoft, 2021-11-08:
1. What do you get if you run plain Raspberypi OS? Because this sounds like a hardware issue.
2. Are you using a good power supply and good power cable?

qwertymodo, 2021-11-08:
I just went looking for the older version of RaspiOS that OctoPi 0.18.0 is built on so I could test on that (looks like it was [2020-12-02-raspios-buster-armhf-lite](http://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-12-04/)).  I'll report back once I'm able to get that installed.  This pi has been running off of WiFi just fine on an old Raspbian build for a couple of years now until I pulled it down to install OctoPi this morning.  Power supply and power cable are both tested as good, with this Pi on its previous OS, as well as another Pi 3 just to be sure.

qwertymodo, 2021-11-09:
I've just confirmed that the issue also exists in the upstream image 2020-12-02-raspios-buster-armhf-lite.  Going to try the latest upstream image and see if that fixes it, and if not, I'll try rolling back and see if I can figure out where it breaks.

guysoft, 2021-11-09:
Awsome! Should I close this?

qwertymodo, 2021-11-09:
No, I would still consider this an open issue.  Even if the root problem is caused upstream, if there is a working version of the upstream image, that image should be used here.  I'm still working on trying out different versions to figure out where the break occurred.

gregor2004ua, 2021-11-10:
I have the same problem. Tried on 2 different Raspberry PI 3B
Also just tried installing latest Raspbian on one of them and having same problem...

guysoft, 2021-11-10:
One of my 3B+ just died, made me think it might be also that the chip died.

jusemon, 2021-12-04:
Same problem here, testing with a Raspberry Pi 3B v1.2 from 2015, the wireless interface works with an image of Retropie but not of OctoPi.

qwertymodo, 2022-01-14:
In my case, at least, this appears to be a hardware issue after all.  Swapped the card to another 3B, and it works fine there.  I'm going to close this issue, but if anybody else with the same issue is able to track down any more information on their system, feel free to post it.

closed 2022-01-14 by qwertymodo