guysoft/OctoPi #758: Bullseye support

guysoft, 2021-11-09:
Bullsye is out. Need to see if anything is broken:
https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-11-08/

Cc @foosel if you have ane foresight

foosel, 2021-11-10:
[Latest build here on GitHub Actions](https://github.com/guysoft/OctoPi/actions/runs/1441942732) at least went through after you merged #757 last night and the `regex` issue was solved upstream.

andylittle, 2021-11-19:
Running this with my Ender3/Marlin 2.0.9.1, will be printing shortly. Not sure if this is the proper place for feedback.

- 2021-11-18_2021-10-30-octopi-bullseye-armhf-lite-1.0.0.zip
- 2021-10-30-octopi-bullseye-armhf-lite-1.0.0.img
- 5.10.63-v7+
- 11.1
- Raspberry Pi 2 Model B Rev 1.1

guysoft, 2021-11-19:
@andylittle Its good, post if it works for you, also would make others reply too

andylittle, 2021-12-09:
This has been working fine, no issues

guysoft, 2021-12-09:
Indeed,  closing

closed 2021-12-09 by guysoft