guysoft/OctoPi #762: wpa_supplicant seg fault

youngd24, 2022-01-18:
While trying to troubleshoot getting wifi to work I found that /usr/sbin/wpa_supplicant would produce a segmentation fault. I first checked wpa via the systemctl command which is where I initially found the SEGV return from the attempted service startup.

Trying to run wpa_supplicant manually:

```
root@octoprint:/usr/sbin# ./wpa_supplicant --help
Segmentation fault
```
I ran an apt-get update & apt-get upgrade to make sure I had latest and continued to observe the segfault.

File details:

```
root@octoprint:/usr/sbin# ls -l wpa_supplicant 
-rwxr-xr-x 1 root root 2244528 Apr 16  2021 wpa_supplicant

root@octoprint:/usr/sbin# file wpa_supplicant     
wpa_supplicant: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=43fecfb2463035479e1cd23af281f23bea0d31da, stripped

root@octoprint:/usr/sbin# ldd wpa_supplicant     
        linux-vdso.so.1 (0x7eb15000)
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76ec1000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76eaa000)
        libnl-3.so.200 => /lib/arm-linux-gnueabihf/libnl-3.so.200 (0x76e7f000)
        libnl-genl-3.so.200 => /lib/arm-linux-gnueabihf/libnl-genl-3.so.200 (0x76e6a000)
        libnl-route-3.so.200 => /lib/arm-linux-gnueabihf/libnl-route-3.so.200 (0x76df7000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76de4000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76d62000)
        libpcsclite.so.1 => /lib/arm-linux-gnueabihf/libpcsclite.so.1 (0x76d49000)
        libssl.so.1.1 => /lib/arm-linux-gnueabihf/libssl.so.1.1 (0x76cca000)
        libcrypto.so.1.1 => /lib/arm-linux-gnueabihf/libcrypto.so.1.1 (0x76ab9000)
        libdbus-1.so.3 => /lib/arm-linux-gnueabihf/libdbus-1.so.3 (0x76a68000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x7691a000)
        /lib/ld-linux-armhf.so.3 (0x76ed6000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x768f0000)
        libsystemd.so.0 => /lib/arm-linux-gnueabihf/libsystemd.so.0 (0x76853000)
        liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x76822000)
        liblz4.so.1 => /lib/arm-linux-gnueabihf/liblz4.so.1 (0x767f6000)
        libgcrypt.so.20 => /lib/arm-linux-gnueabihf/libgcrypt.so.20 (0x7671c000)
        libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0x766f2000)
```
Ran it through strace once as well:

```
root@octoprint:/usr/sbin# strace ./wpa_supplicant 
execve("./wpa_supplicant", ["./wpa_supplicant"], 0x7ed317b0 /* 18 vars */) = 0
brk(NULL)                               = 0x2014000
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76fd1000
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0x76fd0000
close(3)                                = 0
readlink("/proc/self/exe", "/usr/sbin/wpa_supplicant", 4096) = 24
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\3\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=17708, ...}) = 0
mmap2(NULL, 81964, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76f8e000
mprotect(0x76f92000, 61440, PROT_NONE)  = 0
mmap2(0x76fa1000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x76fa1000
close(3)                                = 0
munmap(0x76fd0000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=48123, ...}) = 0
mmap2(NULL, 48123, PROT_READ, MAP_PRIVATE, 3, 0) = 0x76fc5000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/librt.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0 \30\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=26600, ...}) = 0
mmap2(NULL, 90648, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76f77000
[... 155 lines cut ...]
Segmentation fault
```
I manually copied the wpa_supplicant file over from another (generic Raspbian) Pi which resolved the issue:

```
pi@octoprint:/usr/sbin $ ./wpa_supplicant -v    
wpa_supplicant v2.8-devel
Copyright (c) 2003-2019, Jouni Malinen <j@w1.fi> and contributors
```

After reboot wifi works.

Hardware is a Raspberry Pi 3B+

Kernel info:

```
pi@octoprint:/usr/sbin $ uname -a
Linux octoprint 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux
```

OctoPi 0.18.0
OctoPrint 1.7.2

I have a previous core dump and can generate new ones if you need it.

guysoft, 2022-01-18:
Not sure how to reproduce it.
If you copy the 0.18.0 binary back does it produce the segfault again?

youngd24, 2022-01-23:
Yes, it still segfaults, saved it to a .orig file before copying the other one in.

guysoft, 2022-02-04:
Hey, question , how come the hostname is octoprint and not octopi? Did you change anything worth mentioning?
How did you flash the image?

guysoft, 2022-05-09:
Closing if no reply