guysoft/OctoPi #770: OctoPi 1.0.0 RC1 Status

guysoft, 2022-02-27:
First release candidate for OctoPi 0.18.0

There are both 32bit and 64bit images available. Which give support to new Raspberry Pi 4B hardware pis that have been shipping out there. 
The 64bit image is based on Ubuntu 20.04.4, since it seemed more stable than RaspberryPi OS when testing 64bit, it might improve later on.

Raspsberrypi 3 and up can try the 64bit version. No performance gain in normal OctoPi is expected. It might help future plugins.

Please try the release candidate so we know it works.

32bit armf:
Download it at:
https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2022-02-27_octopi-20.04.4-preinstalled-server-arm64+raspi-1.0.0.zip

Md5: [``a78e25bda751e0253981020d2ddf4298``](https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2022-02-27_octopi-20.04.4-preinstalled-server-arm64+raspi-1.0.0.zip.md5).

64bit arm64/aarch64:
Download it at:
https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2022-02-27_octopi-20.04.4-preinstalled-server-arm64+raspi-1.0.0.zip

Md5: [``a78e25bda751e0253981020d2ddf4298``](https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2022-02-27_octopi-20.04.4-preinstalled-server-arm64+raspi-1.0.0.zip.md5).

### Changes in the image

* OctoPrint 1.7.3
* Version is 1.0.0, it seem like its about time https://github.com/guysoft/OctoPi/commit/b6b7f3623aa59021513693596537543b725aff04
* Raspbery PI OS uses bullseye
* rpi.gpio-common has been added
*  Fix webcamd daemon loop & handling of non-usb v4l2 devices #741  (@foosel)
* Add a network monitoring for wifi connections   #729 (thanks @hawkeyexp)
* [Delete octopi-network.txt](https://github.com/guysoft/OctoPi/commit/ca2840ced1dc6ae398e4c693927a4527514b042a)
*  picam-bullseye-fix #763  (Thanks @PowerWiesel)

cp2004, 2022-02-27:
I will admit I'm still not completely sold on the version number being 1.0.0. We get enough people already confusing OctoPi and OctoPrint, I feel like this might make it worse?

What about just 'OctoPi 19.0' - there's no API involved, and it's not possible to update it, so the SemVer type versioning doesn't need to apply? Just specifying that it's the release after 0.18.

guysoft, 2022-02-27:
@cp2004 I had a talk about this, which I can't seem to find anywhere now.
The argument is people confuse OctoPi and OctoPrint anyway. So it doesn't really change the situation.
And I don't want to that already existing confusion effect version naming. Worst case if it goes crazy (which I doubt, but open to be corrected), wWe can jump to version 2 or 3. But it seems to me to remain at 0.x for this long is just too much. Also its shorter.
We could have named OctoPi differently, but its too of a good pun IMHO.

Taomyn, 2022-02-28:
Great news and really want to try it - will backup my current 0.18 on an RPi-4 and restore it to a new SD card to see how it goes.

Will we be able to upgrade to the full release once it's out, or will we have to install/restore it again?

foosel, 2022-02-28:
Reflash and restore, there's no way to upgrade from one version of Octo*Pi* to another.

ytwytw, 2022-02-28:
And 32bits and 64 bits are the same MD5 as well

matonb, 2022-02-28:
I would hope so, they're pointing at the same file

guysoft, 2022-02-28:
My bad fixing the links for 32bit. Pasting error

QuinnDamerell, 2022-02-28:
Congrats on 1.0! Well deserved! 🥳

x40-Community, 2022-02-28:
I just tested the 32bit distribution with the Weedo X40 printer. As with all previous distributions, the connection will not be established due to the outdated CH340 driver. 
Please update the CH340 driver.
Have a look at:
https://www.weedo3dprinter.com/doku.php/x40/updatech340

Thanks

eblieb, 2022-03-01:
Tried to restore a backup and got
Installing plugin "PrintTimeGenius"...
> /home/pi/oprint/bin/python -m pip install https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip --no-cache-dir
< Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
< Collecting https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip
<   Downloading https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip (7.8 MB)
Removing temporary unpacked folder
Error while running restore
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/backup/__init__.py", line 1260, in _restore_backup
    on_install_plugins(known_plugins)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/backup/__init__.py", line 292, in on_install_plugins
    self.__class__._install_plugin(
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/backup/__init__.py", line 914, in _install_plugin
    cls._pip_caller.execute(*pip)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/pip.py", line 208, in execute
    return self.call(command, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/commandline.py", line 244, in call
    all_stdout += process_stdout(p.stdout.readlines(timeout=0.5))
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/commandline.py", line 236, in process_stdout
    return process_lines(lines, self._log_stdout)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/commandline.py", line 229, in process_lines
    processed = self._preprocess_lines(
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/pip.py", line 498, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/pip.py", line 517, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode characters in position 5-44: ordinal not in range(256)
 
Restore failed! Check the above output and octoprint.log for reasons as to why.

eblieb, 2022-03-01:
Looks like I just had to rebuild my Locales afte rthe flash

robness, 2022-03-01:
For plugins you have to reinstall some of them because of the differences in Python versions: https://community.octoprint.org/t/pios-bullseye-killed-octoprint/38790

Basically restore the backup to the plugin error point. install the plugin manually and then restore the backup again after the reboot and rinse and repeat for each failed plugin.

plduganaman, 2022-03-01:
The Navbar Temp plugin no longer appears.  I tried uninstalling and reinstalling but it doesn't show.  Should I contact the creators of the plugin?

robness, 2022-03-01:
Etcher had problems verifying the 64 image on Windows 11, but I was still able to start the import of the back up and almost got through installing all the plugins including those that I had to manually install. One of the last few plugins but I thought it might be something I needed to update on the OS level. SSH connection was refused. I tried enabling it manually through the ssh file with no luck. Plugged it into a keyboard and monitor and quickly realized the raspi-config couldn't be found either after finally being able to ssh. Back to the nightly builds I go: https://github.com/guysoft/OctoPi/actions/workflows/build.yml

rerickson1, 2022-03-01:
> Etcher had problems verifying the 64 image on Windows 11, but I was still able to start the import of the back up and almost got through installing all the plugins including those that I had to [...]

Sounds like a windows issue. I used Raspberry Pi Imager tool on macOS with the 64-bit .img and SSH was setup and works just fine.

cp2004, 2022-03-01:
> Tried to restore a backup and got
> Installing plugin "PrintTimeGenius"...
> < Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> < Collecting [...]

This `UnicodeDecodeError` is a problem with compatibility between Pip 22.0 and OctoPrint at the moment. It's been fixed in 1.8.0 which will be starting release candidate soon as well. When that's released this issue won't happen anymore.

guysoft, 2022-03-01:
@robness The issue that was linked about dealing with an upgraded version of python is not relevant in this case, because it talks about having an existing python environment, and then upgrading it and fixing all the paths. But the case here is a brand new python environment that was built. So the path changes should not be an issue.
Python has been upgraded to 3.9 on 32bit, but that is not supposed to break compatibility (unless there is something new I am not aware of).

guysoft, 2022-03-01:
@x40-Community Is there some move to get that driver in to [mainline kernel](https://lore.kernel.org/all/?q=CH340G) or [raspberrypi/linux](https://github.com/raspberrypi/linux)?
I can't seem to find any, that would be the appropriate channel before we patch it on top with OctoPi.
Another issue is the driver linked is just code in a .zip file without any version control. I can't see what has been changed and why. So shipping it could cause other issues.

foosel, 2022-03-01:
No license attached to this arbitrary driver either, so it *can't* be included.

robness, 2022-03-01:
> Sounds like a windows issue. I used Raspberry Pi Imager tool on macOS with the 64-bit .img and SSH was setup and works just fine.

It was with Etcher. No problems using the imager tool. I need to figure out how to get that working with another PI project I am involved with. So convenient to use.

eblieb, 2022-03-01:
I have never had an issue with Rufus.

robness, 2022-03-01:
Arc Welder also had the UnicodeDecodeError, but when I tried to install it manually I got the following error:

Installing plugin "Arc Welder" from https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip...
/home/pi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpyi1j5pdp/ArcWelderPlugin-master.zip --no-cache-dir
Processing /tmp/tmpyi1j5pdp/ArcWelderPlugin-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: OctoPrint>1.3.8 in /home/pi/oprint/lib/python3.8/site-packages (from Arc-Welder==1.0.0+u.bb71e8f) (1.7.3)
Requirement already satisfied: setuptools>=6.0 in /home/pi/oprint/lib/python3.8/site-packages (from Arc-Welder==1.0.0+u.bb71e8f) (44.0.0)
Requirement already satisfied: six in /home/pi/oprint/lib/python3.8/site-packages (from Arc-Welder==1.0.0+u.bb71e8f) (1.16.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.0)
Requirement already satisfied: feedparser<7,>=6.0.8 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (6.0.8)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2021.10.11)
Requirement already satisfied: zipstream-new<1.2,>=1.1.8 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.8)
Requirement already satisfied: sarge==0.1.6 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.1.6)
Requirement already satisfied: psutil<6,>=5.8 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (5.9.0)
Requirement already satisfied: pathvalidate<3,>=2.4.1 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.5.0)
Requirement already satisfied: websocket-client<1,>=0.59 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.59.0)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.4)
Requirement already satisfied: immutabledict<3,>=2.1 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.2.1)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.0.10)
Requirement already satisfied: unidecode in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.3.3)
Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.10.4)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.18.2)
Requirement already satisfied: emoji<2,>=1.4.2 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.6.3)
Requirement already satisfied: netifaces<1,>=0.11 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.11.0)
Requirement already satisfied: zeroconf<0.34,>=0.33 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.33.4)
Requirement already satisfied: netaddr<0.9,>=0.8 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.8.0)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.9.0)
Requirement already satisfied: tornado<7,>=6 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (6.1)
Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.1.1)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (7.1.2)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.1)
Requirement already satisfied: OctoPrint-PiSupport>=2021.10.28 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2021.10.28)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (3.1.1)
Requirement already satisfied: Jinja2<3,>=2.11.3 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.11.3)
Requirement already satisfied: colorlog<6,>=5.0.1 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (5.0.1)
Requirement already satisfied: PyYAML<6,>=5.4.1 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (5.4.1)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.0.1)
[... cut ...]
Requirement already satisfied: ifaddr>=0.1.7 in /home/pi/oprint/lib/python3.8/site-packages (from zeroconf<0.34,>=0.33->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.1.7)
Building wheels for collected packages: Arc-Welder
Building wheel for Arc-Welder (setup.py): started
Building wheel for Arc-Welder (setup.py): finished with status 'error'
Running setup.py clean for Arc-Welder
Error!
Could not install plugin from https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip, see the log for more details

robness, 2022-03-01:
> Arc Welder also had the UnicodeDecodeError, but when I tried to install it manually I got the following error:
>
> Installing plugin "Arc Welder" from [...]

Still a Latin-1 error from logs:
2022-03-01 07:26:25,739 - octoprint.plugins.pluginmanager - ERROR - Reason: UnicodeEncodeError('latin-1', '  │ exit code: 1\n', 2, 3, 'ordinal not in range(256)')
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 900, in _command_install_archive
    returncode, stdout, stderr = self._call_pip(pip_args)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 1631, in _call_pip
    return self._pip_caller.execute(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/pip.py", line 208, in execute
    return self.call(command, **kwargs)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 243, in call
    all_stderr += process_stderr(p.stderr.readlines(timeout=0.5))
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 239, in process_stderr
    return process_lines(lines, self._log_stderr)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 229, in process_lines
    processed = self._preprocess_lines(
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/pip.py", line 498, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/pip.py", line 517, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2502' in position 2: ordinal not in range(256)

cp2004, 2022-03-01:
@robness As I explained above it is not an issue with specific plugins it's a problem with the current version of OctoPrint and the new version of pip released a few weeks ago, already fixed but not released, and it will impact any plugin that requires downloading dependencies.

robness, 2022-03-01:
> @robness As I explained above it is not an issue with specific plugins it's a problem with the current version of OctoPrint and the new version of pip released a few weeks ago, already fixed but not released, and it will impact any plugin that requires downloading dependencies.

I figured that, but for this one, even trying to manually install it, it wasn't working. It was meant more as a heads up that the plugin wouldn't even work in a manual install because of the newer version of pip.

Thanks for everything that you do to support the project.

JoveToo, 2022-03-01:
Would it be possible to copy the `octopi-wpa-supplication.txt` file to `/etc/wpa_supplication`, restrict access and then delete the file from boot? 

It is not good policy to have files with passwords readable by everyone.

guysoft, 2022-03-01:
@JoveToo This is unrelated to the release candidate. It has OctoPi has been behaving sinc Aug 2016: https://github.com/guysoft/OctoPi/commit/326bbdfa73093bc9afe82737e733386c457a4fd6 . Its not going to change in this release.

foosel, 2022-03-02:
> Cannot check for throttling
> OctoPrint cannot check for throttling situations reported by your Pi. vcgencmd doesn't work as expected. Make sure the system user OctoPrint is running under is a member of the "video" group.

That indicates that the 64bit build is still missing a working `vcgencmd`, either it's not there at all (which would be bad) or the user is missing some permissions (which could be fixed)

@guysoft do you know if the `vcgencmd` family is available on the Ubuntu images you use? If not, that's a big issue as that's the only way OctoPrint (or rather the PiSupport plugin) can currently check for undervoltage issue, which are sadly so common that they have caused a ridiculous support overhead in the past before this check was introduced. An image without the ability of OctoPrint to check for throttling is something that is going to cause us *significant* support overhead again while hunting down ghosts of issues that actually are due to the Pi browning out.

Grey-Sheep, 2022-03-02:
> > Cannot check for throttling
> > OctoPrint cannot check for throttling situations reported by your Pi. vcgencmd doesn't work as expected. Make sure the system user OctoPrint is running under is a member of the "video" group.
> 
> That indicates that the 64bit build is still missing a working `vcgencmd`, either it's not there at all (which would be bad) or the user is missing some permissions (which could be fixed)
> 
> @guysoft do you know if the `vcgencmd` family is available on the Ubuntu images you use? If not, that's a big issue as that's the only way OctoPrint (or rather the PiSupport plugin) can currently check for undervoltage issue, which are sadly so common that they have caused a ridiculous support overhead in the past before this check was introduced. An image without the ability of OctoPrint to check for throttling is something that is going to cause us _significant_ support overhead again while hunting down ghosts of issues that actually are due to the Pi browning out.
> 
> _edit_ According to https://2byt.es/post/vcgencmd/ it should be as easy as
> 
> 1. `sudo apt-get install libraspberrypi-bin`
> 2. `sudo usermod -aG video pi` (if not already done)
> 
> if building against an Ubuntu base image.

rerickson1, 2022-03-02:
One downside of using Ubuntu as the base image for 64-bit is that it is not as easy to add a shutdown button, compared to using this on Raspberry Pi OS:
```
sudo nano /boot/config.txt
```
 Add this to the file:
```
# shutdown with GPIO3 low
dtoverlay=gpio-shutdown,gpiopin=3,active_low="y",gpio_pull="up",debounce=2000
```

I tried getting a python script going, but didn't have luck auto launching the script at startup from `/etc/rc.local`.

yasars, 2022-03-02:
I have the same problem with the backup and restore on new.1.0 i have downgrage pip.after that restore is working , after then reboot make update pip ... :)

wlayher, 2022-03-02:
I just updated successfully from my previous version (already with Python 3) to the new version (32-bit). Normally, if you do the same thing repeatedly, like installing an update or a plugin, but each time expect a different result, that is considered a sign of insanity! :-)
But hey, don't knock it if it works! :-)
The webcam stream still freezes frequently for my HQ PiCam at FullHD, same as on the last version. I guess I have to change to ustreamer again.

guysoft, 2022-03-02:
Just an update, Ubuntu 64bit will surely get an RC2 and likely OctoPi 32bit might also (didn't decide what to do in the case of only one image that needs to be update). Due to missing ``vcgencmd`` permissions

guysoft, 2022-03-02:
@rerickson1 Is there any difference between the 32bit and 64bit that requires you to add that line for a shutdown button?

eblieb, 2022-03-02:
Guessing you mean OctoPi 64 bit

eblieb, 2022-03-02:
Is it possible to downgrade PIP in the images so restore can function
smoothly and then PIP can be upgraded?

b-morgan, 2022-03-02:
I'm confused about how to configure WiFi. Should I use ctrl-shift-x on the RPi imager or should I edit octopi-wpa-supplicant.txt?

foosel, 2022-03-02:
Recommended way by OctoPrint's setup guide is to use rpi imager.

About the latin-1 encoding error on restore, as already said I'm currently fully focusing on getting 1.8.0rc1 out of the door which will contain a fix for that. But based on past experiences it will be at least a month until stable 1.8.0 gets released (we have seen 4 weeks at least of RC phase for the past years now). So IF the wait for that is too long I suggest to roll pip back to version 21 for the second OctoPi RC (both 32 and 64 bit) to get that going ASAP.

eblieb, 2022-03-02:
> @foosel <https://github.com/foosel> I can wait, though if others voice
> something different here I could speed it up.
> I have other work on FullPageOS, RealtimePi and others that I can do in
> the meantime. Possible release a new CustomPiOS stable.

Guy... any chance you want to make a custom program to compile marlin
firmware on the RPI in a simple to use interface :)

rerickson1, 2022-03-02:
> @rerickson1 Is there any difference between the 32bit and 64bit that requires you to add that line for a shutdown button?

@guysoft On the 64-bit Ubuntu based build, `/boot/config.txt` didn't exist so I assumed setting the DT overlay for a shutdown button wasn't supported in Ubuntu. Maybe my assumption is incorrect?
On the 32-bit Raspberry Pi OS build, that file is there so you can setup a shutdown IO very easily.

guysoft, 2022-03-02:
@eblieb I see no reason how that is related to current RC1 status. aka not sure why you asked it in this issue and not in a separate issue. or the community pages, or in a private email asking to hire me :)

eblieb, 2022-03-02:
Sorry was a bad joke because you were talking about other things you can do
in the meantime.

guysoft, 2022-03-02:
@eblieb Gotcha. Its kinda hard to tell apart due to Poe's law. I actually got people asking me to do stuff because they need it for their jobs before.

JeremyLaurenson, 2022-03-04:
Also seeing the Unicode error; suggest we wait till thats fixed and incorporate that fix into the image and not publish as-is.

ChrisHeerschap, 2022-03-06:
Installed 32 bit on a Pi 2 (trying to find a Pi 4, hah) with no problems. Did a little of my own customizing afterwards and am setting up Octoprint plugins and all. So far no issues seen.

matonb, 2022-03-07:
I needed to:
* add the 'video' group to the user pi to fix the 'Cannot check for throttling' error
* install g++ package to install OptoLapse plugin

I haven't been able to resolve a serial port permissions is yet though.  I'm using the GPIO serial pins instead of a USB connection.

Odd permissions on dev/ttyS0
```
ls -la /dev/ttyS0
crw--w---- 1 root tty 4, 64 Mar  7 10:41 /dev/ttyS0
```

If I allow the `tty` group read access `chmod g+r /dev/ttyS0`, the problem goes away temporarily but something is reverting the permissions...

guysoft, 2022-03-07:
@matonb 32 bit right?

matonb, 2022-03-07:
@guysoft ,  Ah I missed a critical bit of info there...

No, I was testing the 64bit version

SiliconExarch, 2022-03-10:
Downgraded to 0.18.0 again as the webcam stream was freezing every minute or so requiring me to refresh OctoPrint and snapshots were also failing intermittently with my PlayStation Eye webcam using the suggested arguments for mjpeg-streamer.

This is using the 64bit version on a Pi 3B+. Also I had to install raspi-config myself as it wasn't in the image.

foosel, 2022-03-14:
For the record, both `octopi-password.txt` and `octopi-hostname.txt` do not work on the 64bit build, probably because the services in question do not run as they are still implemented as init scripts and systemd on the ubuntu build used here simply ignores them. At least the password one is also something we recommend people do if they've managed to forget the system password (requires physical access to the Pi to solve, but at least *can* be solved without having to fire up a unix system and a chroot environment). 

I noticed this while running an automated update test for the imminent release of OctoPrint 1.8.0rc1 against the 64bit image - or tried to run it that is, since the provisioning of the image didn't work as expected and thus the rest of the run couldn't complete.

[See here](https://github.com/guysoft/OctoPi/tree/devel/src/modules/octopi/filesystem/root/etc/init.d) for the services I'm talking about - they need a conversion to systemd for 64bit to achieve feature parity in this regard.

Personally btw I'm not planning on pushing the 64bit build on people on the download page - I don't see any advantages from it for the general public, and the completely different base systems will be a nightmare in end-user support 😐

guysoft, 2022-03-14:
The network settings should work in the ubuntu build. I tested them a few months ago. That is something I should fix.

foosel, 2022-03-14:
Yeah, network does work, but neither password nor hostname

guysoft, 2022-03-14:
I am considering adding a setting to [my fork of rpi imager](https://github.com/guysoft/pi-imager), which would let us disable those options.
OctoPi and many other images are a single-user operating system in effect, and there is really no need to be able to change the username. 
The password makes sense though to have an option. I could look in to that.

eblieb, 2022-03-23:
Still having camera issues where the camera freezes in the control panel

guysoft, 2022-03-23:
@eblieb What camera? details please. I got both a pi camera and a webcam working here, so I need information to differentiate

eblieb, 2022-03-23:
@guysoft I am getting this error a lot in DMESG
```

[224158.023005] uvcvideo: Failed to resubmit video URB (-1).
[225621.991796] uvcvideo: Failed to resubmit video URB (-1).
[226994.479849] uvcvideo: Failed to resubmit video URB (-1).
[227726.474051] uvcvideo: Failed to resubmit video URB (-1).
[227817.973307] uvcvideo: Failed to resubmit video URB (-1).
[229830.969270] uvcvideo: Failed to resubmit video URB (-1).
[230013.970997] uvcvideo: Failed to resubmit video URB (-1).
[231111.962102] uvcvideo: Failed to resubmit video URB (-1).
[232850.452275] uvcvideo: Failed to resubmit video URB (-1).
[232850.452303] uvcvideo: Failed to resubmit video URB (-1).
[234131.455648] uvcvideo: Failed to resubmit video URB (-1).
[235595.448098] uvcvideo: Failed to resubmit video URB (-1).
[236327.446014] uvcvideo: Failed to resubmit video URB (-1).
[237242.447494] uvcvideo: Failed to resubmit video URB (-1).
[237333.948067] uvcvideo: Failed to resubmit video URB (-1).
[243464.413076] uvcvideo: Failed to resubmit video URB (-1).
[247307.410226] uvcvideo: Failed to resubmit video URB (-1).

and
[37042.653540] usb 1-1.3: New USB device strings: Mfr=2, Product=1, SerialNumber                                                                                                                           =3
[37042.653557] usb 1-1.3: Product: HDA Webcam USB
[37042.653572] usb 1-1.3: Manufacturer: HDA Webcam USB
[37042.653587] usb 1-1.3: SerialNumber: HDA Webcam USB
[37042.659485] uvcvideo: Found UVC 1.00 device HDA Webcam USB (0c45:6366)
[37042.681116] input: HDA Webcam USB: HDA Webcam USB as /devices/platform/soc/3f                                                                                                                           980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input7
[37042.718769] usb 1-1.3: 3:1: cannot get freq at ep 0x84
[37048.443644] dwc_otg_handle_wakeup_detected_intr lxstate = 2
[37048.996063] usb 1-1.2: new full-speed USB device number 17 using dwc_otg
[37049.106057] usb 1-1.2: device descriptor read/64, error -71
[37049.336041] usb 1-1.2: device descriptor read/64, error -71
[37049.566041] usb 1-1.2: new full-speed USB device number 18 using dwc_otg
[37049.676090] usb 1-1.2: device descriptor read/64, error -71
[37049.906016] usb 1-1.2: device descriptor read/64, error -71
[37050.026227] usb 1-1-port2: attempt power cycle
[37050.705964] usb 1-1.2: new full-speed USB device number 19 using dwc_otg
[37051.145991] usb 1-1.2: device not accepting address 19, error -71
[37051.255986] usb 1-1.2: new full-speed USB device number 20 using dwc_otg
[37051.695971] usb 1-1.2: device not accepting address 20, error -71
[37051.696249] usb 1-1-port2: unable to enumerate USB device
[37068.575396] usb 1-1.2: new full-speed USB device number 21 using dwc_otg
[37068.685448] usb 1-1.2: device descriptor read/64, error -71
[37068.915415] usb 1-1.2: device descriptor read/64, error -71
[37069.145433] usb 1-1.2: new full-speed USB device number 22 using dwc_otg
[37069.255434] usb 1-1.2: device descriptor read/64, error -71
[37069.485435] usb 1-1.2: device descriptor read/64, error -71
[37069.605707] usb 1-1-port2: attempt power cycle
[37070.285383] usb 1-1.2: new full-speed USB device number 23 using dwc_otg
[37070.735390] usb 1-1.2: device not accepting address 23, error -71
[37070.845403] usb 1-1.2: new full-speed USB device number 24 using dwc_otg
[37071.285381] usb 1-1.2: device not accepting address 24, error -71
[37071.285558] usb 1-1-port2: unable to enumerate USB device

guysoft, 2022-03-23:
@eblieb 
1. has the camera ever worked? 
2. this looks like a power supply issue, make sure you are using a good power supply and a GOOD USB cable.

eblieb, 2022-03-23:
@guysoft yes the camera works it just randomly freezes in the web viewer. It works fine with OctoApp and never freezes in that.

antony, 2022-04-04:
Doesn't seem to be able to connect to my wifi network on a pi 3. Wired works without issue, and have configured `/etc/wpa_supplicant/wpa_supplicant.conf` to provide my ssid and password, but it never connects. `iwlist` shows the interface as ready, but it never gets an IP address.

The pi was previously running a very old build of octopi from 2020 and I just swapped out the SD card.

Not sure what other information I can provide but happy to provide whatever helps.

rexit1982, 2022-04-14:
Late to the party but it seems like input_raspicam.so is not present on 0.18 so webcamd is failing to start up/work with camera="raspi".  Was that intentional and I'm jsut not finding it?

[12 more replies]

closed 2022-08-04 by guysoft