guysoft/OctoPi #771: Cannot check for throttling

salomonje, 2022-03-02:
#### What were you doing?

#### Version of OctoPi

#### Printer model & used firmware incl. version

I have read the FAQ.

guysoft, 2022-03-02:
64bit or 32bit?

salomonje, 2022-03-02:
64Bit

cp2004, 2022-03-02:
This seems to be an issue with the 64 bit builds, there was another report on the forum about this, they didn't know they were running 1.0.0 64 bit nightly builds but were experiencing random crashes that were likely due to undervoltage, but the `vcgencmd get_throttled` command is not available to find out if throttling is occurring.

https://community.octoprint.org/t/cannot-check-for-throttling/42545?u=charlie_powell

foosel, 2022-03-02:
See my comment here for solution: https://github.com/guysoft/OctoPi/issues/770#issuecomment-1057175274