guysoft/OctoPi #773: octopi-wpa-supplicant.txt should not remain world readable past first boot

JoveToo, 2022-03-02:
Hi,

currently, the wireless password is passed into a fresh image by using octopi-wpa-supplicant.txt, this is fine. However, this file is world readable and can be read by anyone that can log into the device. Having files with passwords being world-readable is bad practice.

I would like to propose that on first boot, this file is copied to /etc/wpa_supplicant directory, make readable only by wpa_supplicant (root) and then removed from the /boot partition.

Changes for this should be trivial.

foosel, 2022-03-02:
Please never ever state that any kind of changes on a project you are not intimately familiar with should be trivial.

JoveToo, 2022-03-02:
Point taken.

Whatever boot up script exists for wpa_supplication could check for this and execute the commands.

JoveToo, 2022-03-02:
It may be better to just do this if wpa_supplicant successfully associated to the network: then it will only disappear if it works. 

It is more complex though.

guysoft, 2022-03-02:
I am strongly against the idea of user putting their time to write a file, make sure its formatted correctly and then delete it. I speak from painful experience of filling in these config files for testing.