guysoft/OctoPi #774: Settings in octopi.txt not being used

stormbard, 2022-03-08:
#### What were you doing?

Trying to setup Raspberry Pi Camera and the settings in that file are not being used. If I wait until the system is up and then restart the webcamd service the settings in octopi.txt are used. 

I netboot my RPi4. Since octopi.txt lives on the boot partition which is mounted; I can only assume, after webcamd starts.

1. Modify octopi.txt to have desired camera settings
2. Reboot octopi
3. Camera settings not being used.
4. Manually restart webcamd `sudo systemctl restart webcamd` and setting in file are used

#### What did you expect to happen?

Webcam settings in octopi.txt to be used when starting up.

#### What happened instead?

Default settings were used instead at boot and had to manually restart webcamd to get the desired ones.

#### Did the same happen when running OctoPrint in safe mode?

Yes, same thing happened when I booted with OctoPrint safe mode enabled.

#### Version of OctoPi
```
OctoPrint version : 1.7.3
OctoPi version    : 0.18.0
```

#### Printer model & used firmware incl. version
N/A

#### Screenshot(s)/video(s) showing the problem:

Attached is the octopi.txt and webcamd log file. In the log file the first start of webcamd is on boot and the second is after I manually restarted it.

[webcamd.log](https://github.com/guysoft/OctoPi/files/8206882/webcamd.log)
[octopi.txt](https://github.com/guysoft/OctoPi/files/8206883/octopi.txt)

I have read the FAQ.

guysoft, 2022-03-08:
I suggest you pot to https://community.octoprint.org/
It might depend on which camera you are using and I have no way of testing it here on my end without the hardware.

stormbard, 2022-03-08:
Posted there. Forgot to mention I am using the Raspberry Pi Camera v2.

guysoft, 2022-03-10:
Ok, closing since its not an OctoPi installation

closed 2022-03-10 by guysoft

guysoft, 2022-03-17:
Ok, If you want to add better features to OctoPi that is a different matter. I hope you understand though that I can't really support OctoPi components that are ripped out of the image and placed somewhere else.

Would you like to add a PR to the line ``ExecStartPre=/usr/bin/test -f /boot/octopi.txt``?