guysoft/OctoPi #776: [Request] webcamd should allow video devices other than `/dev/videoN`

bendavis78, 2022-03-20:
Video devices under `/dev/videoN` are not always guaranteed to be enumerated in a predictable manner. UDEV no longer allows overriding NAME, so the only solution is to create your own symlink (for example `SYMLINK+=logitech-c310` would create `/dev/logitech-c310`). Unfortunately, `webcamd` is hard-coded to expect devices under `/dev/videoN`, and will break if it encounters otherwise.

cp2004, 2022-03-20:
I don't think this is completely true - I've been able to successfully use `/dev/v4l/by-id` to address the camera device.

Having said that... Is it this regex that breaks when trying to use something else that's not either `/dev/video*` or `/dev/v4l` https://github.com/guysoft/OctoPi/blob/4a92bdf8dcb52611258ded74dd797888bb545ee0/src/modules/octopi/filesystem/home/root/bin/webcamd#L59

guysoft, 2022-03-20:
I don't know how I can account automatically for stuff you changed in the udev rules manually.
But it seems @cp2004 is right - you could just type that variable out by hand, since you are already patching in custom udev code, and it should work.