guysoft/OctoPi #777: Python 3.8

tigeryell99, 2022-03-22:
#### What were you doing?

The OctoPrint-Wyze plugin requires Python 3.8, the current OctoPi image has Python 3.7.3
Are you working on upgrading to Python 3.8 or 3.9? 

#### What did you expect to happen?
Plugin successfully install.
#### What happened instead?
Plugin Install failed
[plugin_pluginmanager_console.log](https://github.com/guysoft/OctoPi/files/8328022/plugin_pluginmanager_console.log)
#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

OctoPrint 1.7.3 Python 3.7.3 OctoPi 0.18.0

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2022-03-22:
The image is based on RPi OS, which in turn uses Debian. So it is based on the Debian Python version, not controlled by OctoPi.

The OctoPi 1.0.0rc in #770 is running Python 3.9 already.

guysoft, 2022-03-22:
Indeed, also Wyze explicitly stated in their documentation to use OctoPi 1.0.0. I can't fix the past. The future is tricky enough.