guysoft/OctoPi #781: "systemctl start octoprint.service" does not fail even though /home/pi/oprint/bin/octoprint fails to start

hlovdal, 2022-04-13:
#### What were you doing?

Updating an older octopi installation which resulted in some failure, **however this bug report is only focused on the behaviour of systemctl**.

#### What did you expect to happen?

`sudo systemctl status octoprint.service` should report a failed state.

#### What happened instead?

```bash
(pi) octopi: ~/oprint/bin>sudo systemctl status octoprint.service
● octoprint.service - LSB: OctoPrint daemon
     Loaded: loaded (/etc/init.d/octoprint; generated)
     Active: active (exited) since Wed 2022-04-13 21:53:40 CEST; 43min ago
       Docs: man:systemd-sysv-generator(8)
    Process: 955 ExecStart=/etc/init.d/octoprint start (code=exited, status=0/SUCCESS)
        CPU: 135ms

Apr 13 21:53:40 octopi.andeby.dns systemd[1]: Starting LSB: OctoPrint daemon...
Apr 13 21:53:40 octopi.andeby.dns octoprint[955]: Starting OctoPrint Daemon: OctoPrint.
Apr 13 21:53:40 octopi.andeby.dns systemd[1]: Started LSB: OctoPrint daemon.
(pi) octopi: ~/oprint/bin>
```

Notice that systemctl is happy and reports no problems even though octoprint died immediately. If I manually try to start octoprint it exits with a non-zero return code:

```bash
(pi) octopi: ~>/home/pi/oprint/bin/octoprint --host=127.0.0.1 --port=5000 || echo "failed to start"
Traceback (most recent call last):
  File "/home/pi/oprint/bin/octoprint", line 5, in <module>
    from octoprint import main
ModuleNotFoundError: No module named 'octoprint'
failed to start
(pi) octopi: ~>
```

The underlying issue here is old packages installed under /home/pi/oprint/lib/python3.7/site-packages while python has been updated to 3.9. However that is only indirectly related to this bug. *Regardless of why octoprint fails to start, if it does then systemctl start should also fail*.

#### Did the same happen when running OctoPrint in safe mode?

Not applicable since it fails before any mode comes into play.

#### Version of OctoPi

Raspbian GNU/Linux 11 (bullseye) - updated from 10/buster.
Octoprint 1.7.3
I am not sure what created `/etc/init.d/octoprint` initially, `dpkg -S /etc/init.d/octoprint` fails to associate it with any installed program.

#### Printer model & used firmware incl. version

Not applicable.

#### Screenshot(s)/video(s) showing the problem:

Not applicable.

I have read the FAQ.

cp2004, 2022-04-13:
If it is running through init.d, then this is way too old to be supported for bug reports. For years OctoPi has run through systemd, so there is no point debugging the situation.

hlovdal, 2022-04-14:
Closing since init.d script is removed from later releases.

For reference, the underlying issue was solved by running `pip install -U octoprint` (based on [this topic](https://community.octoprint.org/t/how-can-i-update-the-octoprint-installation-on-my-octopi-image/207/5)) and then reinstalling the missing plugins (comparing `/home/pi/oprint/lib/python3.9/site-packages` to `/home/pi/oprint/lib/python3.7/site-packages`). And with that octoprint worked like it should.

However I have trouble getting a camera module working, so I might bite the bullet and reinstall octopi from scratch for that reason.

closed 2022-04-14 by hlovdal