guysoft/OctoPi #782: Possible PR proposal: Webcamd with support for multiple streaming implementations

cp2004, 2022-04-18:
a 'webcamd v2', if you like.

A while ago I had an idea, with a lot of talk around streaming software about on discord (adding ustreamer, webrtc etc.) that webcamd would need a bit of work to support multiple streamers.

Currently, it supports multiple webcams OK, but the code is a bit confusing (at least for me) so the thought of adding an extra dimension didn't make me feel that good. So I came up with this 'idea' that seemed to make sense to me.

## Core `webcamd`

What is currently known as `webcamd` remains the main service. It iterates over the configuration files provided (both `/boot/octopi.txt` and `/boot/octopi.conf.d/*.txt` as it does now. Each file can have a new `camera_streamer` variable, that is either `mjpg`, `hls`, `ustreamer`, `webrtc` etc. etc.

It iterates through the files and calls a subscript, such as `webcamd-mjpg`, which handles the rest of the configuration and actually starting mjpg-streamer. This can be done for as many config files as there are.

## Prototype

I also played around with this idea a few evenings ago. I have a concept here https://github.com/cp2004/OctoPi/commit/03228838fedf9677489d0afb32a9491a95e7d2fa - it does not completely work but I hope it illustrates the idea I'm working on

## Benefits

* Easier to maintain webcamd, as there is less code needed for what (I think) seems to be the same functionality - the commit above has net reduction in lines of code if I remember right
* No need for the `streamer_select` service, which can be confusing and hard to setup for users (at least, it was not as straightforward as I would like)
* If someone else wants to add another streaming option down the road, there is an easy way.

## Thoughts?

I don't have a huge amount of time to convert it into a working PR, but I thought I would post this issue to get some feedback on the idea, to see if it is worth me working on a proper implementation and submitting a PR in the future.

Of course, the answer could be 'no' as well 🙂 - that's why I'm asking before properly working on such a large rewrite of the setup.

guysoft, 2022-04-19:
Looks cleaner, did you test it?

cp2004, 2022-05-08:
I am working on this, but progress will be very slow until later in the summer. It is not far from being able to work flawlessly (I hope!), but I got distracted trying to integrate HLS as well. Branch is [`cp2004/OctoPi/webcamd-v2`](https://github.com/cp2004/OctoPi/tree/webcamd-v2). Developing it with an actual camera attached now...

I may make a PR for the updated webcamd for just mjpg streamer initially, then follow up with adding HLS and also ustreamer. Eventually I would also like to add a webrtc stream option, but I haven't manged to get anything functional other than a slightly dodgy stream from aiortc.

guysoft, 2024-01-16:
Guess this got stuck?