guysoft/OctoPi #796: OctoPi 1.0.0 RC3 Status

guysoft, 2022-10-27:
Third release candidate for OctoPi 1.0.0

As before, there are both 32bit and 64bit images available. 
The 64bit image is based on on RpiOS 64bit.

Raspsberrypi 3 and up can try the 64bit version. No performance gain in normal OctoPi is expected. It might help future plugins.

Please try the release candidate so we know it works.

32bit armf:
Download it at:
https://unofficialpi.org/Distros/OctoPi/nightly/2022-10-27_2022-09-22-octopi-bullseye-armhf-lite-1.0.0.zip

Md5: [``88135c1af2491aa4f114190c728d5fa2``](https://unofficialpi.org/Distros/OctoPi/nightly/2022-10-27_2022-09-22-octopi-bullseye-armhf-lite-1.0.0.zip.md5).

64bit arm64/aarch64:
Download it at:
https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2022-10-27_2022-09-22-octopi-bullseye-arm64-lite-1.0.0.zip

Md5: [``a0fe9dd406f7bb4f8bf0a3d0af1d4582``](https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2022-10-27_2022-09-22-octopi-bullseye-arm64-lite-1.0.0.zip.md5).

### Changes since RC2
* OctoPrint 1.8.6
* This release deals with the userless images changes ( https://github.com/guysoft/CustomPiOS/issues/163 ), updates OctoPrint to latest
* Uses Debian "bullseye" RpiOS 2022-09-22

### Changes in the image since RC1

* OctoPrint 1.8.1
* ships ffmpeg with HLS support at ``/opt/ffmpeg-hls/ffmpeg``  #784
* 64 bit build has moved to raspios (#785)
* Uses new Rpi userless system
* Uses Debian "bullseye" RpiOS 2022-04-04

### Changes in the image

* Version is 1.0.0, it seem like its about time https://github.com/guysoft/OctoPi/commit/b6b7f3623aa59021513693596537543b725aff04
* Raspbery PI OS uses bullseye
* rpi.gpio-common has been added
*  Fix webcamd daemon loop & handling of non-usb v4l2 devices #741  (@foosel)
* Add a network monitoring for wifi connections   #729 (thanks @hawkeyexp)
* [Delete octopi-network.txt](https://github.com/guysoft/OctoPi/commit/ca2840ced1dc6ae398e4c693927a4527514b042a)
*  picam-bullseye-fix #763  (Thanks @PowerWiesel)

foosel, 2022-12-08:
As already told you via DM, I totally lost sight of this with everything else going on and only now finally got around to testing this. Big big sorry for that. 

However, I sadly have to report, that `octopi-password.txt` does *not* work (32bit), contrary to your comment https://github.com/guysoft/OctoPi/issues/788#issuecomment-1285508388. It may very well be showing as being executed, but it is not doing anything. My testrig setup involves flashing an image and then setting the hostname and password via `octopi-hostname.txt` and `octopi-password.txt`. Hostname setting **worked**. However, upon attempting to log into the machine with the newly set password, access was denied. Using the default `raspberry` instead however worked, so something is broken there.

Can also be confirmed by these lines in the motd:

```
SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.
```

And OctoPrint also warns about this:

[image]

Meanwhile:

```
root@octopia:/home/pi# systemctl status change_password | cat
● change_password.service - LSB: Change pi's password via /boot/octopi-password.txt
     Loaded: loaded (/etc/init.d/change_password; generated)
     Active: active (exited) since Thu 2022-09-22 01:23:58 BST; 2 months 16 days ago
       Docs: man:systemd-sysv-generator(8)
      Tasks: 0 (limit: 1472)
        CPU: 0
     CGroup: /system.slice/change_password.service

Sep 22 01:23:56 octopia systemd[1]: Starting LSB: Change pi's password via /boot/octopi-password.txt...
Sep 22 01:23:58 octopia systemd[1]: Started LSB: Change pi's password via /boot/octopi-password.txt.
```

```
-- Journal begins at Thu 2022-09-22 01:23:40 BST, ends at Thu 2022-12-08 11:19:00 GMT. --
Sep 22 01:23:46 octopi systemd[1]: Starting LSB: Change pi's password via /boot/octopi-password.txt...
Sep 22 01:23:48 octopi systemd[1]: change_password.service: Succeeded.
Sep 22 01:23:48 octopi systemd[1]: change_password.service: Unit process 442 (change_password) remains running after unit stopped.
Sep 22 01:23:48 octopi systemd[1]: change_password.service: Unit process 444 (chpasswd) remains running after unit stopped.
Sep 22 01:23:48 octopi systemd[1]: Stopped LSB: Change pi's password via /boot/octopi-password.txt.
Sep 22 01:23:49 octopi chpasswd[444]: pam_unix(chpasswd:chauthtok): password changed for pi
Sep 22 01:23:49 octopi change_password[442]: Password for user pi changed and change file deleted.
```

The `change_password` service basically consists of this:

``` bash
do_start () {
  text_file="/boot/octopi-password.txt"
  if [ ! -f "$text_file" ]
  then
    exit 0
  fi

  new_password=`head -n1 "$text_file" | sed -e 's/^[[:space:]]*//' -e 's/[[:space:]]*$//' | tr -d '\n'`
  if [ ! -n "$new_password" ]
  then
    log_failure_msg "No new password provided, refusing to change to empty password"
    exit 1
  fi

  (echo "pi:$new_password" | chpasswd && rm "$text_file" && log_success_msg "Password for user pi changed and change file deleted") || log_failure_msg "Could not change password"
}
```

Running `echo "pi:newpassword" | chpasswd` manually from a root prompt works and does what it's supposed to do, so I frankly have NO idea right now what's going wrong here. I wonder if it could be solved by getting rid of the init script and converting that into a proper unit instead.

A quick test with this seems to work at least:

`/etc/systemd/system/change-password`
``` systemd
[Unit]
Description=Allow changing the main user's password via /boot/octopi-password.txt

DefaultDependencies=no

Before=network-pre.target
Wants=network-pre.target

After=local-fs.target
Wants=local-fs.target

[Service]
Type=oneshot
ExecStart=/root/bin/change-password

[Install]
WantedBy=multi-user.target
```

`/root/bin/change-password`
``` bash
#!/usr/bin/env bash

set -e

USERID=1000
FIRSTUSER=`getent passwd $USERID | cut -d: -f1`
PASSWORD_FILE="/boot/octopi-password.txt"

if [ ! -f "$PASSWORD_FILE" ]
then
        exit 0
fi

# read password from file
new_password=`head -n1 "$PASSWORD_FILE" | sed -e 's/^[[:space:]]*//' -e 's/[[:space:]]*$//' | tr -d '\n'`
if [ ! -n "$new_password" ]
then
        echo "No new password provided, refusing to change to empty password"
        exit 1
fi

# change password
(echo "$FIRSTUSER:$new_password" | chpasswd && rm "$PASSWORD_FILE" && echo "Password for user $FIRSTUSER changed and change file deleted") || echo "Could not change password for $FIRSTUSER"

```

```
-- Journal begins at Thu 2022-09-22 01:23:40 BST, ends at Thu 2022-12-08 11:20:00 GMT. --
Dec 08 11:14:11 octopia systemd[1]: Starting Allow changing the main user's password via /boot/octopi-password.txt...
Dec 08 11:14:13 octopia chpasswd[273]: pam_unix(chpasswd:chauthtok): password changed for pi
Dec 08 11:14:13 octopia change-password[271]: Password for user pi changed and change file deleted
Dec 08 11:14:13 octopia systemd[1]: change-password.service: Succeeded.
Dec 08 11:14:13 octopia systemd[1]: Finished Allow changing the main user's password via /boot/octopi-password.txt.
```

I haven't yet been able to test this against a completely fresh image however and also don't know if I will get around to doing that.

foosel, 2022-12-08:
Just tested it again. So, it looks like it fails only on the initial boot. A subsequent creation of another `/boot/octopi-password.txt` file and then rebooting changes the password. Consider me even more confused now.

foosel, 2022-12-08:
For reference, flashed and provisioned via my testrig (basically a scripted `dd` followed by some file copying to `/boot`):

```
[pi@flashhost.lan] out: 2906653696 bytes (2.9 GB, 2.7 GiB) copied, 181.438 s, 16.0 MB/s
[pi@flashhost.lan] out:

Flashing done, giving the system a bit to recover...
... done
[pi@flashhost.lan] sudo: mount /dev/disk/by-id/usb-LinuxAut_sdmux_HS-SD_MMC_000000000498-0:0-part1 /home/pi/mount/pia
[pi@flashhost.lan] put: <file obj> -> /home/pi/mount/pia/octopi-wpa-supplicant.txt
[pi@flashhost.lan] put: <file obj> -> /home/pi/mount/pia/octopi-network.txt
[pi@flashhost.lan] put: <file obj> -> /home/pi/mount/pia/octopi-hostname.txt
[pi@flashhost.lan] put: <file obj> -> /home/pi/mount/pia/octopi-password.txt
[pi@flashhost.lan] sudo: echo 'boot_delay=3' >> "$(echo /home/pi/mount/pia/config.txt)"
[pi@flashhost.lan] sudo: umount /home/pi/mount/pia
[pi@flashhost.lan] sudo: /home/pi/tools/usbsdmux/bin/usbsdmux /dev/usb-sd-mux/id-000000000498 dut
[pi@flashhost.lan] sudo: /home/pi/tools/ykush/bin/ykushcmd -u 3
[pi@flashhost.lan] sudo: systemctl restart um25c_octopia
[pi@flashhost.lan] run: /home/pi/tools/mqtt_annotation.sh pia "Switched pia to DUT mode"

Done.
Disconnecting from flashhost.lan... done.
```

Then SSH attempt after first boot, new password doesn't work, old one still in effect:

```
Linux octopia 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

------------------------------------------------------------------------------
Access OctoPrint from a web browser on your network by navigating to any of:

    http://octopia.local
    http://192.168.1.221

https is also available, with a self-signed certificate.
------------------------------------------------------------------------------
This image comes without a desktop environment installed because it's not
required for running OctoPrint. If you want a desktop environment you can
install it via

    sudo /home/pi/scripts/install-desktop
------------------------------------------------------------------------------
OctoPrint version : 1.8.6
OctoPi version    : 1.0.0
------------------------------------------------------------------------------

pi@octopia:~ $ sudo journalctl -u change_password

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for pi:
-- Journal begins at Thu 2022-09-22 01:23:40 BST, ends at Thu 2022-12-08 16:17:28 GMT. --
Sep 22 01:23:46 octopi systemd[1]: Starting LSB: Change pi's password via /boot/octopi-password.txt...
Sep 22 01:23:48 octopi systemd[1]: change_password.service: Succeeded.
Sep 22 01:23:48 octopi systemd[1]: change_password.service: Unit process 451 (change_password) remains running after unit stopped.
Sep 22 01:23:48 octopi systemd[1]: change_password.service: Unit process 455 (chpasswd) remains running after unit stopped.
Sep 22 01:23:48 octopi systemd[1]: Stopped LSB: Change pi's password via /boot/octopi-password.txt.
Sep 22 01:23:50 octopi chpasswd[455]: pam_unix(chpasswd:chauthtok): password changed for pi
Sep 22 01:23:50 octopi change_password[451]: Password for user pi changed and change file deleted.
-- Boot c7e5d321d7194b689ef598ef20d91d0d --
Sep 22 01:23:57 octopia systemd[1]: Starting LSB: Change pi's password via /boot/octopi-password.txt...
Sep 22 01:23:58 octopia systemd[1]: Started LSB: Change pi's password via /boot/octopi-password.txt.
```

Create new password file, reboot:

```
pi@octopia:~ $ sudo -s
root@octopia:/home/pi# cat > /boot/octopi-password.txt
test
root@octopia:/home/pi# sudo reboot
```

Password now works?!

foosel, 2022-12-08:
I'm wondering if we might see some interaction between the new userless stuff or something. Note the second boot in between the two successful runs of the script. I didn't trigger those. If it only fails on the very first boot but works on any after, I think this can be ignored for now. As long as users have a chance to reset their password once they've forgotten it, the thing works as intended.

guysoft, 2022-12-11:
Ok good to know. Also I thought it worked for me when testing by hand, so you guess its something related to the userless stuff makes sense.

I think a possible fix would be to try and create a systemd unit and then make it run only after the userless unit. For not I rather just let it go since this image has been held long enough.

wtremmel, 2022-12-23:
In octopi.txt, example for "Configuration of network monitoring" should also have a line for IPv6, like

`destination_host=fdd1:9494:7978:15::1`

wtremmel, 2022-12-23:
32bit version worked fine on my Pi4 using an 8GB SD-card. Network is IPv6 only, after changing _dhcpcd.conf_ to use the hardware address for SLAAC everything was fine (make this a config option for /boot ?)

Another suggestion would be a plugin so the .ssh/authorized_keys are included (and restored) using OctoPi backup.

64bit version did not work, RasPi  was not accessible via Wifi, to debug I need to detach the Raspi from my printer which I might do next week.
Now I will try to print something.

guysoft, 2022-12-25:
@wtremmel Strange, I use here the 64bit version on a dual-stack network and it reaches using the ipv6.
Are you using the ip directly or via mDNS hostname?

wtremmel, 2022-12-25:
I tried the IPv6 IP directly. I will try again after the holidays when I can debug properly.

guysoft, 2022-12-30:
I think ill release this when I get time to do it.

wtremmel, 2022-12-31:
ok, finally got a HDMI cable to attach my RasPi to debug the 64bit problem, here is what I found so far:

- wpa_supplicant gets started twice, once via systemctl, once from dhcpcd (no idea if this can be a problem)
- dhcpcd reports error: dhcpcd_prestartinterface: wlan0: Operation not possible due to RF_kill

CoryCharlton, 2023-01-13:
Thanks! There was a restore option when I first logged in so that worked. 

I installed the 64bit version and everything worked fine out of the box.

badnetmask, 2023-01-18:
I've been considering installing the RC image, because I would like to run 64-bits, but I have some questions, if you don't mind:

- Is it better to install the base RC3 image, then upgrade, or just install using the latest nightly?
- Is it worth installing either the RC3 or the nightly at this point? How far is it from the actual release?

Thanks for your great work!

dchauran, 2023-01-19:
As this is external, leaving this comment mostly as an FYI in case anyone else hits the issue.

I had a persistent USB issue with 0.18, and it was resolved by updating the kernel. After installing rc3 64 bit, the issue is back, even after updating the kernel. This happens often enough that I cannot use my pi to print:

ch341-uart ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32

This appears to be related to an issue that should be resolved in current kernels, but seems it is not at least in 64 bit.

https://github.com/raspberrypi/linux/issues/5211
https://github.com/raspberrypi/linux/issues/5088

Broken Version --> Linux octopi 5.15.89-v8+ #1620 SMP PREEMPT Wed Jan 18 12:26:08 GMT 2023 aarch64 GNU/Linux
Working Version --> Linux octopi 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux

rubyredrick, 2023-01-22:
Hi.

I'd like to try this in order to get a Raspberry Pi Camera Module.V3 working.

I've been googling around and can't find a concise description of how to install a beta.  My Debian skill's are a bit rusty, and although I've used imager to install octopi before the raspberry pi imager support came along, the memories are vague.

I'm assuming that the first step would be to use apt-get to update the base system on the pi.
I guess I'd then want to make a new sd card with the new image.  Is there a good guide on how to do that.

Can someone point me to a good walkthrough on how to install the beta?

Thanks.

CoryCharlton, 2023-01-22:
@rubyredrick you can take a backup from the Octoprint UI. Then download the .img from the links above and use the Raspberry PI imager to install it.

rubyredrick, 2023-01-22:
> @rubyredrick you can take a backup from the Octoprint UI. Then download the .img from the links above and use the Raspberry PI imager to install it.

Thanks. Will that update raspian or do I need to do that first?

CoryCharlton, 2023-01-22:
> > @rubyredrick you can take a backup from the Octoprint UI. Then download the .img from the links above and use the Raspberry PI imager to install it.
> 
> Thanks. Will that update raspian or do I need to do that first?

The Raspberry PI imager will wipe your SD card and install the image you downloaded. I didn't see any "upgrade" path for the RC other than "start fresh and restore backup" as I outlined but maybe I missed something 🤷 

If you have a second SD card available you could always install the image on that so you could easily swap back to your current installation if you run into problems.

loskexos, 2023-02-06:
Downloaded both images (32/64 bit). Put them on 32GB mSD cards via raspberry imager.
Adjusted in config text files the Wlan things and Camera options.

next: 
put the 64bit image into my Raspbi 3B+. Powered on. Never went reachable on network, activity LED blinking a lot.
put the 32bit image into same Raspbi 3B+. Powered on. Never went reachable on network, activity LED blinking occasionally.

First time ever attached a monitor to a raspberry :). Then checked what is going on.
64 bit image -> bootloop (last console messages seem to be around changing CPU governor and starting ssh)
32bit image -> did sit in logon prompt. after logon -> wlan disabled from rfkill. -> rfkill enable 0 -> reboot -> bootloop. Looks similar than other image.

Nothing connected to raspi initially except power supply (although through gpio due to broken microSD port). Later added hdmi-monitor and usb-keyboard.

I'll try on the other raspi once that is free (finished printing) to see if it is something weird local with that one pi or reproducible on others.

kevinwright, 2023-02-10:
64bit version doesn't include the `input_raspicam.so` library.  I know the camera can be used as though it were a USB device via `input_uvc.so`, but that sacrifices the ability to rotate the view (my cam is mounted upside down on an arm)

Looks like it's back to 32bit for me

dchauran, 2023-02-10:
IIRC, there's a kernel overlay you have to apply. I know I got it working in 64 bit, but I rolled back to 32 bit due to a USB issue.

kevinwright, 2023-02-11:
Didn't work for me on 32bit either :(

I'm going back to stable for now, and resigning myself to a script to keep the network up as I wait for an RC4 that works with the pi camera

CoryCharlton, 2023-02-12:
I opened issue #804 related to network monitoring. 

It might not be specific to 1.0.0 RC3 but that is where I am currently experiencing it. 

I've fixed the script on my local system and will send over a PR shortly.

foosel, 2023-02-13:
> I'm going back to stable for now, and resigning myself to a script to keep the network up as I wait for an RC4 that works with the pi camera

As you stated yourself, the pi camera *does* work, albeit with `input_uvc`. From my understanding `input_raspicam` cannot be run nor even built on RaspberryPi OS Bullseye (which OctoPi 1.0.0 is based off) since the necessary library bindings are no longer present there ([which is why the module doesn't even get built](https://github.com/jacksonliam/mjpg-streamer/blob/310b29f4a94c46652b20c4b7b6e5cf24e532af39/mjpg-streamer-experimental/plugins/input_raspicam/CMakeLists.txt#L2)). My guess is that this is due to switch over to [`libcamera`](https://www.raspberrypi.com/documentation/computers/camera_software.html#libcamera-and-libcamera-apps), which is also the only thing that AFAIK supports the latest PiCam (and which will most likely require a switch to a different mjpg-streamer fork to work at all).

wtremmel, 2023-02-13:
Exactly - I got my Raspberry Pi Camera v3 working following these instructions:
https://community.octoprint.org/t/add-support-for-raspberry-pi-camera-v3/49052/3

foosel, 2023-02-13:
Yes, and that's precisely the fork I have in mind for patching this post-release with [OctoPi-UpToDate](https://github.com/OctoPrint/OctoPi-UpToDate/), since it's really really time to get this build released.

kevinwright, 2023-02-13:
> My guess is that this is due to switch over to [`libcamera`](https://www.raspberrypi.com/documentation/computers/camera_software.html#libcamera-and-libcamera-apps), which is also the only thing that AFAIK supports the latest PiCam

Okay, so it sounds like this needs to go through at least one more round of release candidacy, with `libcamera` in the image and `octopi.txt` updated to reference `libcamera` instead of `raspicam`.

foosel, 2023-02-13:
In the end it's something that @guysoft has to decide on, but frankly I don't see a reason why we should now do yet another round of RC on 1.0.0 just because of input-raspicam, when something that already works fine for all the hardware that existed when this RC phase started is available in the shape of input-uvc.so. This particular RC phase has been going on since August 2022, thanks to a TON of changes on bullseye that required lots of debugging. 

Given that we have the option to add picam3 support AFTER the release on an updated image, without having to delay the release of 1.0.0 which works fine for the majority of existing use cases, and that will automatically fix this problem as well, personally I'd consider this not a reason to do an RC4. We already decided against that for some other issue (password change file not working on first boot).

Considering that Guy has a ton of stuff on his plate I can imagine that putting in the work you are asking for is something that would cause a *severe* delay here. And I'd be happy to do this work instead, but rather via CustoPiZer in OctoPi-UpToDate since frankly that allows me a WAY faster turn-around and my resources for OctoPi stuff are severely limited as well given that I have all of OctoPrint already to manage. For the end-user, there's no difference, we'd simply backport things then from there.

kevinwright, 2023-02-13:
My concern is that this gets released and someone new will come along, see the comments in `octopi.txt`, update their config to use `raspicam`, then wonder why it's broken.

If it's acceptable to change just those comments in going from RC to release then that would would work, but I'm very nervous at the thought of a release that actively tells people to use a known-broken strategy. I'd like to see all references to `raspicam` ripped out, and maybe add a URL to a page outlining how to use `libcamera`.

bsimmo, 2023-02-13:
Release, pop a warning up. 
Get a new testing version going with any new updates, a new lts kernel is right around the corner. (6.1)
By the end of summer it'll be Bookworm update time.

Especially if as mentioned the back porting stuff is fine.
Otherwise it'll be RC8 and you'll never be out of the RC loop.

I assume the rest needs to be, does it work on all current RPi's upto before Camera3 was released.

foosel, 2023-02-13:
> Otherwise it'll be RC8 and you'll never be out of the RC loop.

That is precisely my fear here since that has already been going on so long. I'd rather fight the fear stated by @kevinwright with documentation (and another quick update script) than causing yet MORE delay here for things that frankly don't even are issues for most of the users out there since they'd never ever touch `octopi.txt`, don't have an issue with the existing workaround, and couldn't care less about details like this. At some point it become way more problematic to explain why something is NOT released vs why something works slightly different than before.

kevinwright, 2023-02-13:
> does it work on all current RPi's upto before Camera3 was released.

My testing strategy was:

- flash the new image
- paste my previously-good config into `octopi.txt`
- discover the camera to be broken

I would content that many others would upgrade via the exact same approach, and that it therefore **doesn't** work on all current RPi's from before the camera 3 was released.

All it takes is an update **in the comments** in that file to say "raspicam now doesn't work, and here's a link to a page stating why". That could be safely done without an RC but would still save a lot of people a lot of pain.

loskexos, 2023-02-13:
32bit image: 
yes, can confirm got a puicture with raspicam hardware using input_uvc. 
Building the input_raspi is possible and easy though (install buildutils, clone source, make, no dependency issues). 

64bit image: 
building mjpeg_streamer indeed hits dependencies and got no usable input_raspi without need to further look into other library issues.

btt:
Considering octopi 1.0.0 rc is essentially not supporting raspi cam via input_raspi in the "old" variant anymore (as the library is not included) -> why keep the choice of "usb|raspi" in octopi.txt at all? 
Maybe quick way here is to add a few lines of comment into octopi.txt around the setting to clarify status around cams allows to move on to release.

PS: whishlist would be a config that just detects what is there and enumerates. To allow >1 cam. 
(had something modded into 0.18 with camera "both" ... liked to have the light raspicam attached to the heatbed and a usb cam giving overview vom above).

kevinwright, 2023-02-13:
> Considering octopi 1.0.0 rc is essentially not supporting raspi cam via input_raspi in the "old" variant anymore (as the library is not included) -> why keep the choice of "usb|raspi" in octopi.txt at all?

Exactly. I'm not concerned about support for the v3 camera, I'm not even concerned about the removal of raspicam library. What really bothers me is that `octopi.txt` still happily suggests that raspicam is a valid option.

foosel, 2023-02-13:
For the record... Have the following ready for OctoPi-UpToDate (the image that you on OctoPrint's download page, and on the Raspberry Pi Imager):

- updated `octopi.txt` (uvc only version)
- ArduCam mjpg-streamer fork replacing the existing one

Will also see I can still make the rpi3 cam work (I finally got my hands on one).

foosel, 2023-02-16:
Some news on the RpiCam3/libcamera front. Experiments with the arducam mjpeg-streamer fork turned out rather disappointing, re performance and stability. Instead I have been building a replacement for the mjpg-streamer/webcamd combo based on camera-streamer the past few days, incl. multicam support. Currently running a Pi with a picam3 and two usb cams, working fine. USB stuff also is getting hot plug support.

guysoft, 2023-02-22:
Released, available on the release page and rpi-imager:
https://github.com/guysoft/OctoPi/tree/1.0.0

closed 2023-02-22 by guysoft

rubyredrick, 2023-02-23:
> Some news on the RpiCam3/libcamera front. Experiments with the arducam mjpeg-streamer fork turned out rather disappointing, re performance and stability. Instead I have been building a replacement [...]

So I guess this means that [OctoPi-UpToDate](https://github.com/OctoPrint/OctoPi-UpToDate/tree/main/scripts) doesn't reflect these experiments.  Yes?  If so do you have an ETA?

Thanks!

kevinwright, 2023-02-23:
Sure it does.

Details at https://github.com/OctoPrint/OctoPi-UpToDate/issues/2