guysoft/OctoPi #801: OctoPi won't boot without HDMI attached

sidamos, 2022-12-30:
#### What were you doing?
1. Flash OctoPi stable using Pi Imager on Windows
2. Boot Pi 4B

When I do not attach HDMI during very first boot, even resize fails and the system is not usable, even when I reboot with HDMI attached. Have to re-flash the card then.
If I attach HDMI for first boot, everything works fine. I can configure Octo Print and I can reboot multiple times (warm and cold).
If I then detach HDMI and try to boot, it does not boot fully. No ping. No ssh.

Tried 2 different SD cards. Tried 2 different power supply (original and UGreen 3A).

#### What did you expect to happen?
Boot without HDMI monitor attached.

#### What happened instead?
Looks like a boot loop, but cannot say for sure because not even ping works.

#### Did the same happen when running OctoPrint in safe mode?
Does not apply, I think.

#### Version of OctoPi
0.18.0.

#### Printer model & used firmware incl. version
Does not apply.

#### Screenshot(s)/video(s) showing the problem:
Does not apply.

sidamos, 2022-12-30:
Now I downloaded the Octo Pi image and flashed it with Linux and dd instead of Pi Imager on Windows. Same result.

cp2004, 2022-12-30:
Have you tried a normal RPi OS image?

sidamos, 2022-12-30:
Just tried the standard Raspberry PI OS 32 Bit (with desktop) and it behaves the same. Although I don't know, if this is normal for the desktop edition. I'll try next the light version.

sidamos, 2022-12-30:
Alright, seems to be a hardware issue. Sorry for the noise.

closed 2022-12-30 by sidamos