guysoft/OctoPi #802: Doesn't work with Pi Camera 3

Timmmm, 2023-01-19:
I had Octopi 0.18.0 running fine with a Pi Camera 2 on a Pi 4B. Camera works (though it is hella laggy due to MJPEG).

I upgraded to a Camera 3 and unfortunately it doesn't work. The `The webcam server is currently not running` page is a very nice touch so here are the logs:

```
Starting up webcamDaemon...

--- Configuration: ----------------------------
cfg_file:      /boot/octopi.txt
camera:        auto
usb options:   -r 640x480 -f 10
raspi options: -fps 10
http options:  -w ./www-octopi -n --listen 127.0.0.1

Explicitly USB device:
-----------------------------------------------

Found video devices:
/dev/video10
/dev/video11
/dev/video12
/dev/video13
/dev/video14
/dev/video15
/dev/video16
/dev/video18
config file='/boot/octopi.txt':USB device was not set in options, start MJPG-streamer with the first found video device: /dev/video10
/root/bin/webcamd: line 184: printf: 0x: invalid hex number
/root/bin/webcamd: line 184: printf: 0x: invalid hex number
<13>Jan 19 12:30:40 root: Starting USB webcam
Checking for VL805 (Raspberry Pi 4)...
  - It seems that you don't have VL805 (Raspberry Pi 4).
    There should be no problems with USB (a.k.a. select() timeout)
Running ./mjpg_streamer -o output_http.so -w ./www-octopi -n --listen 127.0.0.1 -i input_uvc.so -r 640x480 -f 10 -d /dev/video10
MJPG Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
 i: Using V4L2 device.: /dev/video10
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 10
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
Error opening device /dev/video10: video capture not supported.
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
Done bring up all configured video device

Goodbye...
```

Looks like there is a big in `webcamd` (not surprising given that it is Bash...).

I think it's failing because it does this:

```
$ cat /sys/class/video4linux/video10/device/uevent
DRIVER=bcm2835-codec
MODALIAS=platform:bcm2835-codec
```

And can't handle the fact that there is no `PRODUCT`.

Actually after further investigation I think maybe it needs a kernel upgrade to get V4L drivers for the new camera? I had similar issues with software support for Raspberry Pi devices for MotionEyeOS. I think maybe it's just a bad idea to have an entire distro dedicate to one application. Might try just running Octoprint on the official image.

cp2004, 2023-01-19:
Please see the discussion on the forums for supporting this camera:

https://community.octoprint.org/t/pi-camera-v3-not-working/49022

The Pi Cam 3 requires the use of libcamera, which mjpg streamer is not compatible with. There are users that have managed to get it working with an alternative streaming software on the forums.

ltlowe, 2023-01-21:
I've written a script based on Dark360's comments on the octoprint forum.  It will install everything and reconfigure boot and systemd to use camera-streamer.

```
# Combining work from octoprint forums:
# https://community.octoprint.org/t/pi-camera-v3-not-working/49022/16
# https://community.octoprint.org/t/add-support-for-raspberry-pi-camera-v3/49052/5
#
# Tested on a clean install of Jan 20, 2023 Nightly build:
# 2023-01-20_2022-09-22-octopi-bullseye-arm64-lite-1.0.0
```
[cam3install.sh.txt](https://github.com/guysoft/OctoPi/files/10472993/cam3install.sh.txt)

Timmmm, 2023-01-22:
Nicework @ltlowe! Unfortunately it didn't work for me. Running the `camera-streamer` command gives:

```
output/rtsp/rtsp.cc: ?: Running RTSP server on '8554'
[0:05:07.613951455] [794]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3406-e96d0201
device/libcamera/device.cc: CAMERA: No available cameras
device/libcamera/device.cc: CAMERA: Camera `/base/soc/i2c0mux/i2c@1/imx708@1a` was not found.
device/device.c: CAMERA: Can't open device: /base/soc/i2c0mux/i2c@1/imx708@1a
```

I'll probably have to try it with the official distro. :-/

Some minor comments on the script:

1. Probably want a `set -e` in there.
2. You've got an `apt-get install` and `apt install` separately for some reason?
3. TIL about `git clone -jN`! That's going to save me some time at work!
4. You can do `make -j3` to speed up compilation. Although in fairness it seems like their build system is a bit broken because `make install` builds some stuff (maybe rebuilds?) which is not how it is supposed to work.

ltlowe, 2023-01-22:
It's too bad it didn't work for you. A few people on octoprint forums seem to be having success, but I admit I haven't tested it thoroughly.   I simply pulled together two sets of manual instructions that seemed to work and decided to share to help others. I haven't even set up my github repo properly yet.

> |Probably want a set -e in there.

Good idea.

> |You've got an apt-get install and apt install separately for some reason?

Pulling from multiple sources and not clean yet :-)

> TIL about git clone -jN! That's going to save me some time at work!

I got that from one of the original threads, gonna have to use it at my work too :-)

> You can do make -j3 to speed up compilation. Although in fairness it seems like their build system is a bit broken because make install builds some stuff (maybe rebuilds?) which is not how it is supposed to work.

I'd try `make -j -l $(nproc)` for both and let it use all the cpu, although if install is doing build work, I wouldn't be surprised if the dag isn't proper either.

I'm unfortunately not looking at this for a while as I've had a major failure with my hotend that I need to deal with, but there is an ongoing discussion at https://community.octoprint.org/t/pi-camera-v3-not-working/49022

ltlowe, 2023-02-05:
The script has been updated and moved to https://raw.githubusercontent.com/ltlowe/misc/main/picam3/cam3install.sh

ivaniotchenko, 2023-03-14:
@ltlowe, man, you are awesome!
Struggled with Pi cam v3 streaming and Pi4, applied your script and it works like a charm

ltlowe, 2023-03-16:
> @ltlowe, man, you are awesome! Struggled with Pi cam v3 streaming and Pi4, applied your script and it works like a charm

Thank you @ivaniotchenko 
I'm glad I could help with an interim solution, but now that official support is underway, please try it from the link I posted above.

lpla, 2023-05-03:
How's compatibility with the new camera stack going? I saw some changes in this branch of the [OctoPi-UpToDate] (https://github.com/OctoPrint/OctoPi-UpToDate/tree/camera-streamer), but I don't know how upstreamable those changes are to this repository. In fact, I thought that the OctoPi-UpToDate repo was made to only provide updated parts, even beta/RC, of the latest OctoPi image (regarding kernel, bootloader and OctoPrint), not modifying or implementing features.

jneilliii, 2025-03-23:
@guysoft I just today stumbled across Arducam's fork of mjpg-streamer, which should add support for any libcamera device. It adds a libcamera input plugin and the releases to the repo include a drop-in replacement for webcamd. 

https://github.com/ArduCAM/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_libcamera/README.md

Wondering if it would make sense to build/use this version of mjpg-streamer instead of the original to be able to support the more modern line of cameras that only work that way?