guysoft/OctoPi #804: Network monitoring does not work if WiFi was configured outside of the `/boot/octopi-wpa-supplicant.txt` file

CoryCharlton, 2023-02-12:
#### What were you doing?

1. Trying to enable network monitoring to detect network outages

#### What did you expect to happen?

The wifi network to be restarted when it fails

#### What happened instead?

The script does not actually execute the check and no output indicates this failure

#### Did the same happen when running OctoPrint in safe mode?

Not applicable

#### Version of OctoPi

1.0.0 RC3

#### Printer model & used firmware incl. version

Not applicable

#### Screenshot(s)/video(s) showing the problem:

```
Feb 12 11:31:00 octopi-01 systemd[1]: Started Network Monitor.
Feb 12 11:31:00 octopi-01 systemd[1]: networkcheck.service: Succeeded.
```

I have read the FAQ. 👍 

---

I'll send over a PR to fix this shortly.

CoryCharlton, 2023-02-12:
Sent over the PR that fixes two issues

1. Add failure condition if network configuration is not found
2. Check `/etc/wpa_supplicant/wpa_supplicant.conf` for a network configuration in addition to `/boot/octopi-wpa-supplicant.txt`

guysoft, 2023-02-12:
Thanks, merged in