guysoft/OctoPi #807: [Feature] Moving to Kernel 6.1 tests.

bsimmo, 2023-02-13:
RPiOS will soon be moving to a newer Kernel 6.1
This will be the LTS version. 
I just thought I'd give a heads up if anyone wished to give it a try and report and problems back to RPi on their forum.
I guess also any problems that may crop could be added here? So devs can fix it. 
It's moved from next testing to testing, so you can just do sudo rpi-update.
I recommend cloning your current card/drive and testing that way.

Hopefully it doesn't break anything and may bring you new benefits, drivers etc. and MGLRU.
https://forums.raspberrypi.com/viewtopic.php?t=344246

guysoft, 2023-02-21:
I guess we could put it in the nightly for people to test @foosel ?

Though I think we should wait until RpiOS release a new image or the kernel officially, from experience they tend to change stuff and unless someone from rpi wants us to test this and report back. I think its worth waiting.

bsimmo, 2023-02-21:
The kernel will just come through with apt.
I've had no troubles with a Buster OctoPi on a Pi3 all updated as far as Buster allows with the OctoPi setup, so Bullseye should be no problem.
(Buster will not see 6.1) and I've not had the time to move this to Bullseye version for testing yet. 
The link provided gives info and a place to report back.
I was just giving a heads up so if people find anything not working well they can feedback before it goes live. (preferably on the link).
I was figuring they could just do it with apt.

No doubt some changes but most the testing was done in 'kernel next' 

`sudo rpi-update`
Will get the testing version. 

People can revert back to the current release with. 
```
sudo apt update
sudo apt install --reinstall raspberrypi-bootloader raspberrypi-kernel
```

guysoft, 2023-05-03:
Nightly would be using that

lpla, 2023-05-05:
Seems like it is not using it yet. Two last 64 bit builds are still named with `2022-09-22` base image.

guysoft, 2023-05-07:
@lpla looks like you are right, for some reason the raspberrypi mirror is directing the apt-get repo download to india, while the server is based in Israel, and it fails to pull the files.

cc @kaplanlior maintains that and might know.
I will try and overcome that here in the meantime.

Output here:
```
14:33:49 0 upgraded, 166 newly installed, 0 to remove and 0 not upgraded.
14:33:49 Need to get 19.2 MB/94.5 MB of archives.
14:33:49 After this operation, 299 MB of additional disk space will be used.
14:33:49 Err:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libgdk-pixbuf2.0-common all 2.42.2+dfsg-1+deb11u1
14:33:49   Connection failed [IP: 103.195.68.3 80]
14:33:50 Err:2 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libgdk-pixbuf-2.0-0 armhf 2.42.2+dfsg-1+deb11u1
14:33:50   Connection failed [IP: 103.195.68.3 80]
14:33:51 Err:3 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libatspi2.0-0 armhf 2.38.0-4+deb11u1
14:33:51   Connection failed [IP: 103.195.68.3 80]
14:33:52 Err:4 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf at-spi2-core armhf 2.38.0-4+deb11u1
14:33:52   Connection failed [IP: 103.195.68.3 80]
14:33:52 Err:5 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf dbus-user-session armhf 1.12.24-0+deb11u1
14:33:52   Connection failed [IP: 103.195.68.3 80]
14:33:53 Err:6 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libdatrie1 armhf 0.2.13-1+b1
14:33:53   Connection failed [IP: 103.195.68.3 80]
14:33:54 Err:7 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf librsvg2-2 armhf 2.50.3+dfsg-1
14:33:54   Connection failed [IP: 103.195.68.3 80]
14:33:55 Err:8 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf librsvg2-common armhf 2.50.3+dfsg-1
14:33:55   Connection failed [IP: 103.195.68.3 80]
14:33:55 Err:9 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libavahi-client3 armhf 0.8-5+deb11u2
14:33:55   Connection failed [IP: 103.195.68.3 80]
14:33:56 Err:10 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libcups2 armhf 2.3.3op2-3+deb11u2
14:33:56   Connection failed [IP: 103.195.68.3 80]
14:33:57 Err:11 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libayatana-indicator3-7 armhf 0.8.4-1+deb11u2
14:33:57   Connection failed [IP: 103.195.68.3 80]
14:33:58 Err:12 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libdrm-amdgpu1 armhf 2.4.104-1+rpi1
14:33:58   Connection failed [IP: 103.195.68.3 80]
14:33:58 Err:13 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libdrm-nouveau2 armhf 2.4.104-1+rpi1
14:33:58   Connection failed [IP: 103.195.68.3 80]
14:33:59 Err:14 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libdrm-radeon1 armhf 2.4.104-1+rpi1
14:33:59   Connection failed [IP: 103.195.68.3 80]
14:34:00 Err:15 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libevdev2 armhf 1.11.0+dfsg-1+rpi1
14:34:00   Connection failed [IP: 103.195.68.3 80]
14:34:01 Err:16 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libgdk-pixbuf2.0-bin armhf 2.42.2+dfsg-1+deb11u1
14:34:01   Connection failed [IP: 103.195.68.3 80]
14:34:01 Err:17 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libxxf86vm1 armhf 1:1.1.4-1+b2
14:34:01   Connection failed [IP: 103.195.68.3 80]
14:34:02 Err:18 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libllvm11 armhf 1:11.0.1-2+rpi1
14:34:02   Connection failed [IP: 103.195.68.3 80]
14:34:03 Err:19 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libxmu6 armhf 2:1.1.2-2+b3
14:34:03   Connection failed [IP: 103.195.68.3 80]
14:34:04 Err:20 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libmatchbox1 armhf 1.12+git20170224-2
14:34:04   Connection failed [IP: 103.195.68.3 80]
14:34:04 Err:21 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libxxf86dga1 armhf 2:1.1.4-1+b1
14:34:04   Connection failed [IP: 103.195.68.3 80]
14:34:05 Err:22 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf matchbox-window-manager armhf 1.2.2+git20200512-1
14:34:05   Connection failed [IP: 103.195.68.3 80]
14:34:05 W: --force-yes is deprecated, use one of the options starting with --allow instead.
14:34:05 E: Failed to fetch http://raspbian.mirror.net.in/raspbian/raspbian/pool/main/g/gdk-pixbuf/libgdk-pixbuf2.0-common_2.42.2+dfsg-1+deb11u1_all.deb  Connection failed [IP: 103.195.68.3 80]
```

bsimmo, 2023-05-07:
That's a know broken RaspberryPi mirror (see forum), you'll need to set it
yourself or update it and see if it selects another one.
No idea why it hasn't been removed or fixed.

guysoft, 2023-05-08:
@bsimmo What should I search for in the forum?

guysoft, 2023-05-08:
@bsimmo 
But before I close I would like to see the issue on the Rpi Forum to see if there is any input from there.
Fix deployed and you should have a nightly for now.

bsimmo, 2023-05-08:
Just search the IP address, it will pop up.