guysoft/OctoPi #809: Unable to setup wifi on Octopi Bullseye arm64 lite 1.0.0

flo269, 2023-02-22:
#### What were you doing?

1. Downloaded octopi-bullseye-**arm64**-lite-1.0.0.zip from releases
2. Write image to SD card with Raspberry Pi Imager v.1.7.3
3. Remount SD card and edit /boot/octopi-wpa-supplicant.txt with nano (removed the trailing #'s for WPA/WPA2 and country, then added wifi config)
4. Unmout
5. Boot Raspberry Pi (Pi 3 Model B Rev 1.2) from SD Card

#### What did you expect to happen?
Octopi connecting to specified wifi

#### What happened instead?
Not connected to wifi

I was able to connect via Ethernet and everything worked fine that way. I just wasn't able to get wifi working. Also not via raspi-config. Octopi-wpa-supplicant.txt shows the edits made before via nano.

"sudo iwlist wlan0 scan" shows me my wifi, so it shouldn't be a hardware issue.

Also, when I repeat the steps above with the 32bit image, same SD card and same raspberry pi, it connects to my wifi and everything works as expected.

Please let me know if I can provide additional info or a log of some sort. Thanks!

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi
1.0.0 arm64

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

flo269, 2023-02-22:
For the record, this is what the relevant parts of my octopi-wpa-supplicant.txt look like:
```
## WPA/WPA2 secured
network={
  ssid="WifiSSID"
  psk="SecretOPassphrase"
}

## Open/unsecured
#network={
#  ssid="put SSID here"
#  key_mgmt=NONE
#}

## WEP "secured"
##
## WEP can be cracked within minutes. If your network is still relying on this
## encryption scheme you should seriously consider to update your network ASAP.
#network={
#  ssid="put SSID here"
#  key_mgmt=NONE
#  wep_key0="put password here"
#  wep_tx_keyidx=0
#}

# Uncomment the country your Pi is in to activate Wifi in RaspberryPi 3 B+ and above
# For full list see: https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
#country=GB # United Kingdom
#country=CA # Canada
country=DE # Germany
#country=FR # France
#country=US # United States
```

doucettom, 2023-02-23:
I have the same exact problem.

doucettom, 2023-02-23:
> For the record, this is what the relevant parts of my octopi-wpa-supplicant.txt look like:
>
> ```
> ## WPA/WPA2 secured
> network={
>   ssid="WifiSSID"
>   psk="SecretOPassphrase"
> }
>
> ## Open/unsecured [...]

You can try without specifying the country code.

doucettom, 2023-02-25:
Ok I did further testing. I also edited the file with either nano or mousepad if that helps. I tried the 32bits version and it works like it always did so the issue is solely with the 64bits build. I didn't try the latest 64bits nightly though.

Uro1, 2023-02-25:
@doucettom  There were issues reported in the [rpi-imager repo](https://github.com/raspberrypi/rpi-imager) about Raspberry Pi Imager failing to escape password strings correctly in a few different cases.

You could try the pre-release v1.7.4 of Raspberry Pi Imager linked to in this issue comment to see if that fixes your Wifi issues as it seems to have fixed it for others: https://github.com/raspberrypi/rpi-imager/issues/541#issuecomment-1403794458

I also had problems getting this new release imaged correctly with v1.7.3 of Raspberry Pi Imager, every time it would fail to connect to my WiFi network, after double and triple checking my Wifi credentials in rpi-imager I resorted to unplugging it from my printer and hooking it up to a monitor to see what was going on, in the end with nothing obvious out of the ordinary I reset the WiFi credentials locally via raspi-config before I found the linked issue, it's not ideal having to tear down a headless printer in order to set the WiFi credentials. 😅 

Please do report back if this fixes anything for you, as it may nudge the rpi-imager maintainers to release an official 1.7.4 build.

doucettom, 2023-02-25:
> @doucettom There were issues reported in the [rpi-imager repo](https://github.com/raspberrypi/rpi-imager) about Raspberry Pi Imager failing to escape wifi password strings correctly in a few different cases.
> 
> You could try the pre-release v1.7.4 of Raspberry Pi Imager linked to in this issue comment to see if that fixes your Wifi issues as it seems to have fixed it for others: [raspberrypi/rpi-imager#541 (comment)](https://github.com/raspberrypi/rpi-imager/issues/541#issuecomment-1403794458)
> 
> I also had problems getting this new release imaged correctly with v1.7.3 of Raspberry Pi Imager, every time it would fail to connect to my WiFi network, after double and triple checking my Wifi credentials in rpi-imager I resorted to unplugging it from my printer and hooking it up to a monitor to see what was going on, in the end with nothing obvious out of the ordinary I reset the WiFi credentials locally via raspi-config before I found the linked issue, it's not ideal having to tear down a headless printer in order to set the WiFi credentials. sweat_smile
> 
> Please do report back if this fixes anything for you, as it may nudge the rpi-imager maintainers to release an official 1.7.4 build.

Thanks for your reply. I am not using Windows to write the image to the SD card and the problem happened with a good old dd as well. I'll still give 1.7.4 a try if I can find a linux version.

Uro1, 2023-02-26:
There is a Linux .deb of the pre-release rpi-imager 1.7.4 in this comment, I should have also linked it in my previous comment:
https://github.com/raspberrypi/rpi-imager/issues/486#issuecomment-1345625533

If you are not keen on installing pre-release/unoffical releases then you could always build it from source, the build instructions are in the README.md for the repo: 
https://github.com/raspberrypi/rpi-imager/blob/qml/README.md

spanzetta, 2023-02-26:
I am not sure it is the same issue.. but I have Octoprint on Raspberry PI 3A+ running from over 2 years.. and it was always working..
Now it doesn't connect to wifi .. and the problem is that I configured (a while ago) the automatic Desktop start.. and when Desktop start the Keyboard doesn't work.. 
So now I am in stalemate condition: 
- cannot access from remote with SSH as I always did (because of no wifi/no network)
- cannot login to desktop (with HDMI monitor connected) because as soon as it appears the desktop the keyboard stop working..
- tried adding init=/bin/bash to config.txt but then raspi-config cannot write changes (would like to disable automatic desktop start)
Any hints how to solve the problem?
Thanks in advance

cp2004, 2023-02-26:
If you have not just installed the OctoPi 1.0.0 64 bit image and you don't have the same problem as this issue, don't sidetrack this issue to your own problem. Head on over to the [community forums](https://community.octoprint.org) or [discord server](https://discord.octoprint.org) where helpful people can help you troubleshoot.

flo269, 2023-02-26:
> @doucettom There were issues reported in the [rpi-imager repo](https://github.com/raspberrypi/rpi-imager) about Raspberry Pi Imager failing to escape wifi password strings correctly in a few different cases.

I used the rpi-imager to write the sd, but I did not use the inbuilt functionality to preset the wifi credentials. So rpi-imager should not have changed the files written to the sd, it should just have "copied" them over. 

What I failed to mention at the beginning is that I had also tried writing the image with dd and had the same (not working) result.

What I'm trying to say is, I don't think rpi-imager is the problem here. But I'm happy to be proven wrong. ;-)

doucettom, 2023-02-27:
> I used the rpi-imager to write the sd, but I did not use the inbuilt functionality to preset the wifi credentials. So rpi-imager should not have changed the files written to the sd, it should just [...]

Exact Same here but seemingly some special characters are causing troubles. I might try to convert it to hex to see if it goes through.

flo269, 2023-02-27:
I just tried to connect to a wifi with a passphrase containing only upper and lowercase letters - no special characters. But same result: No connection. :-(

WolfiMuc, 2023-02-27:
I had exactly the same issue with Octopi 1.0 on a Raspberry 4 installed using the Pi Imager.

Diving a bit deeper into it I discovered that only some of the wifi parameters from Pi Imager got transferred into the system. I'm using a hidden SSID and am living in "DE" - that's what I also specified in the Imager. After booting the Pi wifi was not up and I could only connect through ethernet. Turns out, in the system's native file "/etc/wpa_supplicant/wpa_supplicant.conf" the country was still set to GB and the switch "scan_ssid=1" in the network settings was missing. The SSID-name and password where correct, though. I changed the country-setting and added the scan_ssid parameter and voila - after a reboot wifi was working as expected.

WolfiMuc, 2023-02-27:
> I had exactly the same issue with Octopi 1.0 on a Raspberry 4 installed using the Pi Imager.
>
> Diving a bit deeper into it I discovered that only some of the wifi parameters from Pi Imager got [...]

FYI - I also tried setting these parameters in the "/boot/octopi-wpa-supplicant.txt" without any luck...

doucettom, 2023-03-05:
I was able to make it work by logging in via Ethernet port and do some manipulation with iwconfig but this won't work with a Raspberry Pi 3A+ or a Pi Zero 2W. I am not really able to plug my 3A+ on the think we'll have to get a new release from @guysoft that fixes this and luckily will also include 1.8.7 :)

tdoucet-ecotel, 2023-03-25:
Had a bit of spare time today, tried the nightly and still the same issue sadly.

guysoft, 2023-03-25:
@doucettom do you remember what you had to change?

doucettom, 2023-03-26:
> @doucettom do you remember what you had to change?

Today's testing help me to find out that adding rfkill unblock wifi to rc.local make the wifi work even on headless Pi. This is a workaround and yes, the country domain is set. Wireless interface was soft blocked somehow

doucettom, 2023-03-26:
I also tried removing /var/lib/systemd/rfkill/ two files prior to booting my flashed instance and it worked. This was reported by a user in RC2 "main" issue thread.

potokslow, 2023-07-15:
I've encountered this issue when flashing Pi 3B with the 64bit OctoPi, version 1.0.0.

The fix, in my case, turned out to be quite simple - after filling in the Wi-Fi details, setting up the country code and then finding out that Wi-Fi is blocked by rfkill, I removed the SD card and copied the _octopi-wpa-supplicant.txt_ to _wpa-supplicant.txt_

On next boot, Pi connected to network automatically and without issues.

I have not tested if this would work the same way if the file was simply renamed right after the imaging.

TL;DR - Try renaming _octopi-wpa-supplicant.txt_ to _**wpa-supplicant.txt**_

dek3rr, 2023-10-06:
this cmd did it for me
`rfkill unblock wlan`

zahansellars, 2023-11-26:
> @zahansellars<https://github.com/zahansellars> You mean the file wpa-supplicant right? Or you mean the pi uses now NetworkMangaer?

Pi isn't using WPA-Supplicant.
3x separate pis tested (1x pi3b and 2x pi4bs).
WiFi info from initial setup will work but doesn't push through to WPA supplicant file and any changes to that file don't seem to work. Connecting to an additional WiFi network from the optional raspian front end/gui also works.

Gaffalover, 2023-12-18:
Hi, just wanted to +1 on this.
Exactly the same: flashed the 1.0.0 64 Bit Image to my 3B via balena etcher and Wifi does not connect.
Ethernet works, problem persists after apt update && apt upgrade -y && reboot.

Wifi is softblocked in rfkill, as doucettom mentioned.

octopi-wpa-supplicant.txt contains 60 char PSK with up and lowercase letters and digits, set countrycode to DE, which is correctly applied in /etc/wpa_supplicant/wpa_supplicant.conf other than WolfiMuc mentioned.

When flashing the 32bit image things work out of the box.

guysoft, 2024-01-30:
Ah, ok, thats the issue. Rpi by default block the wifi and have a service to unblock it when the country. code is set.

In CustomPiOS I remove the code here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/network/start_chroot_script#L31

Can you look if there is anything here?:
```
ls /var/lib/systemd/rfkill/
```

Kjubyte, 2024-04-23:
I can confirm this issue. Works with armhf image, but not with arm64. After I set WLAN country with raspi-config and reboot, wlan starts working.

> Can you look if there is anything here?:
> 
> ```
> ls /var/lib/systemd/rfkill/
> ```
```
pi@octopi:~ $ ls -l /var/lib/systemd/rfkill/
total 12
-rw-r--r-- 1 root root 2 Sep 22  2022 platform-3f300000.mmcnr:wlan
-rw-r--r-- 1 root root 2 Sep 22  2022 platform-fe300000.mmcnr:wlan
-rw-r--r-- 1 root root 2 Sep 22  2022 platform-soc:bluetooth
```
Both wlan files contain 1, after fix its 0.

guysoft, 2024-06-16:
On bookworm I get:
```
~ $ ls /var/lib/systemd/rfkill/
platform-1001100000.mmc:wlan  platform-107d50c000.serial:bluetooth
```