guysoft/OctoPi #810: sudo /home/pi/scripts/install-desktop on OctoPi 1.0.0 doesn't install everything necessary

b-morgan, 2023-02-25:
#### What were you doing?
1. Install OctoPi 1.0.0 image
2. sudo /home/pi/scripts/install-desktop
3. sudo reboot

#### What did you expect to happen?

Desktop appears

#### What happened instead?

Nothing. I had to install lightdm to get the desktop to appear.

#### Did the same happen when running OctoPrint in safe mode?

N/A

#### Version of OctoPi

1.0.0

#### Printer model & used firmware incl. version

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

Yes, I have read the FAQ.

guysoft, 2023-02-26:
Could indeed be something changed in Debian bullseye. Will need to test.

The script output would help

b-morgan, 2023-02-26:
This will take a while for me to capture as I'll have to start over from scratch with a new image.

b-morgan, 2023-02-26:
Here is the script output (captured with `screen -L`)

[install-desktop.zip](https://github.com/guysoft/OctoPi/files/10834091/install-desktop.zip)

This time it worked. Feel free to close this issue.

guysoft, 2023-02-27:
Looks like that indeed worked. Good to hear!
Closing for now.

closed 2023-02-27 by guysoft

Ruhel786, 2023-04-16:
Hi. I know this issue is closed but I'm having the same issue and can't seem to resolve it. I've installed the most recent prerelease version of OctoPi and it doesn't seem to go to desktop once the script is run.

guysoft, 2024-01-15:
@Ruhel786 What is your output when you run it?