guysoft/OctoPi #811: Raspberry Pi Camera v2 Not Working

anonymousaga, 2023-02-25:
#### What were you doing?

1. Configure octopi.txt with camera="raspi" with Octopi 0.18 (Buster). Works fine.
2. Update to Octopi 1.0.0 (bullseye) and copy over octopi.txt
3. The camera stream, does not work.

#### What did you expect to happen?

I have 2 cameras, one USB and one Raspi. Raspi is configured in octopi.txt, and USB is configured in /boot/octopi.conf.d/webcam2.txt. Both should have displayed at proper resolution.

#### What happened instead?

Port 8080 for raspi cam showed nothing, and strangely port 8081 (usb cam) shows the raspi cam, but at a 480p resolution (both cameras are configured to be 720p).

#### Did the same happen when running OctoPrint in safe mode?
Have not tried yet.

#### Version of OctoPi
1.0.0

#### Printer model & used firmware incl. version

Ender 3 Neo running stock firmware

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

I saw a pull request on error #790, but my webcamd was already fixed with that.

cp2004, 2023-02-25:
I am not sure that `camera="raspi"` is supported any more on the Bullseye image (maybe that was only 64 bit?), but you can still use these cameras in 'USB'/auto mode as far as I know. USB mode is really just using the UVC/V4l2 driver, it doesn't have to be a USB camera. I believe mentions of the `raspi` mode have been removed from writing in the `octopi.txt` file already.

anonymousaga, 2023-02-25:
Ok, I just configured it to use USB mode with v4l2 driver. After this print finishes I will restart and tell you.

anonymousaga, 2023-02-26:
Yep, got it working that way. Thanks!

closed 2023-02-26 by anonymousaga