guysoft/OctoPi #812: Offical RPI 7" Touchscreen does not working with OctiPi 1.0

JumpMaster, 2023-03-01:
Hi,

I'm using OctoPI on a CM4 in a waveshare cm4-io-base-a carrier along with an official Pi 7" touchscreen.  As of version 1.0 the touch screen is not functional.  The touchscreen is used with OctoDash.  I'm going to further look into why but hopefully this lets other people know.

Can you confirm that v1.0 changed from Raspian based on Debian 10 to the later Debian 11 based version?

guysoft, 2023-03-01:
Hey yes I can, its written in the release notes:
https://github.com/guysoft/OctoPi/releases/tag/1.0.0

> Uses Debian "bullseye" RpiOS 2022-09-22

JumpMaster, 2023-03-01:
Hi @guysoft 

I think this is a very specific problem with Raspbian based on Debian 11 when using a waveshare cm4-io-base-a carrier.  It is not related to OctoPi at all.

I'll close this issue.

Thanks!

closed 2023-03-01 by JumpMaster

JumpMaster, 2023-03-04:
Just in case anyone else has this issue.  The waveshare cm4-io-base-a board before V4 (has Rev 4 printed on the board) uses DSI0 instead of DSI1.  Bullseye only has support for DSI1.  I did have to force the display driver for the 7" display as automatic detection did not work.  To do so make the following change in /boot/config.txt to comment our auto_detect and force the 7" display.

```
# Automatically load overlays for detected DSI displays
#display_auto_detect=1
dtoverlay=vc4-kms-dsi-7inch
```