guysoft/OctoPi #813: Trying to build Armbian for Cubieboard 1

SimonIT, 2023-03-05:
I'm trying to build OctoPi with the [Sid minimal Armbian image](https://www.armbian.com/cubieboard-1/).

After researching, that the armbian image has to go into `src/image-armbian` (and not into `src/image-variants` like mentioned in some issues) it finally started to build with `sudo bash -x ./build_dist armbian`.

But after some waiting, it failed with this error:

```
+ sudo -u pi pip3 install --user virtualenv-tools3
sudo: unable to resolve host simonslaptop: System error
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
++++ echo_red 'build failed, unmounting image...'
++++ echo -e -n '\e[91m'
++++ echo build failed, unmounting image...
build failed, unmounting image...
++++ echo -e -n '\e[0m'
```

SimonIT, 2023-03-12:
@guysoft

guysoft, 2023-03-12:
Hey, where did you get this line to come from?
```
sudo -u pi pip3 install --user virtualenv-tools3
```
Its not part of the armbian variant code which is from here:
https://github.com/guysoft/CustomPiOS/tree/devel/src/variants/armbian

Can you provide the full build log?

SimonIT, 2023-03-12:
Here we go:
[build.log](https://github.com/guysoft/OctoPi/files/10952305/build.log)

guysoft, 2023-03-13:
Ok, I can see where that happens now

It looks like the line from this commit: [e5fc9bfb](https://github.com/guysoft/OctoPi/commit/e5fc9bfb10a5de518459a16f3b7bfb23778c3b5e)

Line:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/start_chroot_script#L340

@cp2004 Any reason to install this as user? can we install it globally?

Considering to install it not as user.

cp2004, 2023-03-13:
Not actually sure on that one - the user-fix script would need a modification if it ended up in a different place, since that uses the user installation. As long as the permissions on OctoPrint's virtualenvironment aren't changed it should be fine.

guysoft, 2023-03-14:
Fixed. Try pulling latest devel and build now. I tested using ``Armbian_23.02.0-trunk_Cubieboard_sid_edge_6.1.11-1.1.0`` and it builds here now.

SimonIT, 2023-03-14:
Okay, I tried building it, and it worked :tada:
The thing is, somehow I can't access it. It doesn't connect to the network via Ethernet (it doesn't have WiFi).
I tried the original CLI and minimal armbian images, and both show up.
I also hooked up a monitor, but there I got only an output with the android from the onboard flash

guysoft, 2023-03-16:
I have no hardware to test - does flashing the base image work?

SimonIT, 2023-03-19:
Yes, the unchanged CLI an minimal images.
I tried finding a log of the boot from the altered images, but couldn't find any meaningful in the /var/log directory.
I also couldn't find any place to buy this board

commit 0e211ef9 (2023-10-22): Fix build on bookworm, might also fix #813
closed 2023-10-22 by guysoft