guysoft/OctoPi #814: userconf not working on 25 March 64bits nightly

doucettom, 2023-03-26:
I tried to create a user with userconf.txt and while the file was already present and filled with pi and what I guess is encrypted "raspberry" password, editing it does nothing and the pi user isn't working either even though the home folder and other stuff are there.

guysoft, 2023-04-04:
Hey, did you test the 32bit as well? Is it only 64bit or is that what you happened to test?

doucettom, 2023-04-05:
> Hey, did you test the 32bit as well? Is it only 64bit or is that what you happened to test?

Only tried 64bits as I was trying to figure out the [wireless issue](https://github.com/guysoft/OctoPi/issues/809)

Kjubyte, 2024-04-23:
I couldn't change the default user either. I tried armhf and arm64 image. I think userconf.txt isn't supported and only used to set password for the 'pi' user.

guysoft, 2024-06-10:
Found an issue with pi user being changed, it makes to octoprint service not work, investigating