guysoft/OctoPi #815: Configured in Raspberry Pi Imager WiFi country "US" is ignored and instead "GB" is used.

ArkBrj, 2023-03-27:
#### What were you doing?

I am installing fresh OctoPi on my Raspberry Pi 4B using official Raspberry Pi Imager tool on Windows.
According to the tool it is OctoPi 1.0.0 with OctoPrint 1.8.7 (build 20230302160739) Released 2023-03-02.
I have configured my WiFi in the tool's "Advancedd options" dialog specifying among other things the "Wireless LAN country" as "US".
Inspection of the generated firstboot.sh on the SD card shows that this settings gets properly propagated:
/usr/lib/raspberrypi-sys-mods/imager_custom set_wlan 'MYSSID' 'mypassword' 'US'
However after first boot of Raspberry Pi WiFi is not available and /etc/wpa_supplicant/wpa_supplicant.conf contains the wrong country (GB). Rest of WiFi information is correct. Changing the country to US and reboot fixes the problem.
This problem does not happen if I install a pure Raspberry Pi OS (tried 32-bit Lite version). In this case wpa_supplicant.conf does not have any country specified which I guess defaults to US.
I found that octopi-wpa-supplicant.txt somehow contributes to this problem - it has country=GB in there and I guess this is somehow takes precedence over whatever I configure in the imager tool.
If I comment out that line right after creating the SD card, Raspberry Pi boots fine and WiFi is available.

The repro steps are simple:
1) Start Raspberry Pi Imager (Windows version).
2) Configure WiFi in "Advanced options" making sure that WiFi country is US.
3) Boot Raspberry Pi
4) Check country in /etc/wpa_supplicant/wpa_supplicant.conf

#### What did you expect to happen?

Expected WiFi to work without extra actions.

#### What happened instead?

WiFi did not work due to wrong country (GB) in /etc/wpa_supplicant/wpa_supplicant.conf

#### Did the same happen when running OctoPrint in safe mode?

N/A since it happens before I can select safe mode.

#### Version of OctoPi

1.0.0

#### Printer model & used firmware incl. version

Not important since this happens before I even connect the printer.

#### Screenshot(s)/video(s) showing the problem:

Not applicable.

I have read the FAQ.

guysoft, 2023-04-02:
Can you post the contents of `` /etc/wpa_supplicant/wpa_supplicant.conf``?

Thanks

ArkBrj, 2023-04-02:
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
**country=GB**

network={
        ssid="<redacted>"
        psk=<redacted>
}

Note that in the Raspberry Pi Imager the "Wireless LAN Country" is configured as "US".

guysoft, 2025-01-07:
might be fixed now that bookworm moved to network manager