guysoft/OctoPi #816: Have a big disclaimer - this will wipe your Pi

hexitex, 2023-04-09:
#### What were you doing?
looking to install octoprint 

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2023-04-09:
When following the instructions on https://octoprint.org/download, using the RPi imager, I am presented with this:
[image]

Garr-R, 2023-04-10:
Welcome to installing Raspberry Pi images...? It always wipes your SD card

guysoft, 2023-04-10:
This is expected behavior. Octopi is an image which is written on the sd card. Erasing all the data on it

closed 2023-04-10 by guysoft