guysoft/OctoPi #817: Webcamd restart in loops then lost video stream

Syl20-94, 2023-05-08:
Hello I switch to Octoprint 1.0.0 654b on my rapsberry pi 4b connected to a MK3 prusa printer,

so I did a backup from previous install in Octoprint 1.8.7 to save installation time. and it works well.

in usage this is OK but I faced a strange behave that I did not have on 0.18 version, I have 2 webcam and on a regular time, image froze and I need to switch from one camera to other to reconnect - I assume that I lost video stream.

I had a look on log and figure out that webcamd generate a lot of logs especially by restarting the deamon each 2 min.

I tried a lot of twicking of the conf file but I can not fix the hint, 

Does someone else have the same issue ? 

the log look like this

[logs_webcamd.txt](https://github.com/guysoft/OctoPi/files/11421045/logs_webcamd.txt)

#### What did you expect to happen?

obviously that videostream are constant 

#### What happened instead?

#### Version of OctoPi
Octopi 1.0.0 64b Octoprint 1.8.7 Python 3.8.10

guysoft, 2023-05-09:
What kind of camera?

Syl20-94, 2023-05-09:
> What kind of camera? Model?

main one is a Logitec C270 and the other is a C930e which works without problem on my previous octopi install (32 bits, Octopi 0.18 )

guysoft, 2023-05-10:
Says here you need to set settings for the C270. I'd start with one and then see how to add the second one:
https://community.octoprint.org/docs?search=c270&topic=21149

Source that indicates changes are needed:
https://www.reddit.com/r/3Dprinting/comments/pjeif0/comment/hbwnmlf/?utm_source=reddit&utm_medium=web2x&context=3

Also you might get better responses on the community forum, since I am not sure this is an issue I can resolve, because its either related to configuration or changes in Raspberry Pi OS. It might be, but you might get faster responses there from people who have the hardware:
https://community.octoprint.org/

Syl20-94, 2023-05-10:
Thank a lot for your help, I will test these point and provide feedback

Syl20-94, 2023-05-18:
Hello I checked again and can't fix it, anyway I figure out I had this behaviour on my previous Octopi 0.18 install so I close the point as this is working like this since a while and is a minor hont

closed 2023-05-18 by Syl20-94