guysoft/OctoPi #818: Cancel print shuts down marlin 2 totally

handsomedude, 2023-05-26:
Using the latest octopi, when I cancel a print, it totally shuts down my marlin 2 board and I have to go press the reset button on the printer board to start it up again.

When I had pi os 64bit installed with octoprint, I cancel the print and I can continue to use printer without pressing the reset button.

Any reason why octopi completely shuts off my marlin 2 board.

guysoft, 2023-05-26:
This is an OctoPrint issue not an OctoPi issue, suggest you post it on https://community.octoprint.org/ and also make a note of what printer you are using and what exact version of Marlin.

closed 2023-05-26 by guysoft

handsomedude, 2023-05-26:
Ok thanks 

Found it strange that when I use pi os and the exact same octoprint version the issues is not there.

Using octopi with exactly same octoprint the issues arrises.

So I didn't think it was octoprint as I'm using the same exact version of octoprint on  both pi os and octopi.

Hence why I posted the query here.

Tha KS for answering

guysoft, 2023-05-28:
Can you please post in that case the versions of both, and octoprint.log?