guysoft/OctoPi #822: OctoPi doesn't allow AP mode WiFi change anymore

LtNeX, 2023-10-03:
In the past I used octopi 0.18 with raspiwifi (https://github.com/jasbur/RaspiWiFi) that allowed me an AP mode whenever my Pi wasn't connected to a known network. 
But since the latest octopi version it doesn't work anymore.
When my pi is in AP mode, I can still connect to it's wifi but it only shows me the direct ocotoprint screen. I tried to chance ports for the AP webpage but didn't work.
Any ideas what can be the issue?

guysoft, 2023-11-05:
RaspiWiFi was not updated for over 5 years.
The 1.0.0 OctoPi image uses bullseye debian and the older uses buster.
There is an issue about it there:
https://github.com/jasbur/RaspiWiFi/issues/116

guysoft, 2024-06-02:
Closing due to no reply

closed 2024-06-02 by guysoft