guysoft/OctoPi #823: Debian 12 Bookworm / Raspberry Pi 5 support

guysoft, 2023-10-27:
Hey,
I haven't had the time to test the nighties with bookworm - has anyone happen to test it?

Also I have no Raspberry Pi 5 to test since they did not arrive to my county. If anyone could update on that it would help too.

idaband, 2023-11-23:
will test the nightly this weekend.  I'll let you know if you haven't already figured it out

NewsGuyTor, 2023-12-08:
> will test the nightly this weekend. I'll let you know if you haven't already figured it out

Any updates? Anyone?

idaband, 2023-12-08:
Didn't work.

Been having heck of time with bookworm and pi 5.

Still don't have working arducam with camera-streamer too ( different issue) but would still want to use that on the pi 5.

InsaneNemesis, 2023-12-09:
I am running 2023-12-05_2023-10-10-octopi-bookworm-arm64-lite-1.1.0.zip on my Pi 5. Seems to be fairly stable. Only issue i am having is i cant get the camera resolution to change. It seems to be stuck on 640x480.

idaband, 2023-12-10:
Are you using the new streaming service too?

kristofferR, 2023-12-20:
> I am running 2023-12-05_2023-10-10-octopi-bookworm-arm64-lite-1.1.0.zip on my Pi 5. Seems to be fairly stable. Only issue i am having is i cant get the camera resolution to change. It seems to be stuck on 640x480.

Same issue here. Using the default streaming service.

adam5zki, 2023-12-21:
I am trying to get a OV5647 wide angle camera to work on a Pi5 with the Octopi Nightly 20.12. Does not even list the camera. 64bit bookworm lite did show the camera, but I could not figure out how to stream it - neither ustreamer nor camera-streamer seemed to work (at least not in a way I could figure out). Since I used octo-deploy there, it felt like the whole thing was funny, so I switched over to the nightly. USB camera works like a charm, except I would prefer the OV5647 with the wider FOV. Not sure what I am doing differently :/

kristofferR, 2023-12-27:
> I am running 2023-12-05_2023-10-10-octopi-bookworm-arm64-lite-1.1.0.zip on my Pi 5. Seems to be fairly stable. Only issue i am having is i cant get the camera resolution to change. It seems to be stuck on 640x480.

Fixed it. Octopi.txt isn't active at all, it's not linked properly.

Login to the pi on ssh, then 

`sudo ln -s /boot/firmware/octopi.txt /boot/octopi.txt`

InsaneNemesis, 2023-12-28:
> > I am running 2023-12-05_2023-10-10-octopi-bookworm-arm64-lite-1.1.0.zip on my Pi 5. Seems to be fairly stable. Only issue i am having is i cant get the camera resolution to change. It seems to be stuck on 640x480.
> 
> Fixed it. Octopi.txt isn't active at all, it's not linked properly.
> 
> Login to the pi on ssh, then
> 
> `sudo ln -s /boot/firmware/octopi.txt /boot/octopi.txt ; sudo reboot`

This does fix the camera resolution issue for me. Im able to run 1280x720 at 20fps. Testing other resolutions now.

apbarratt, 2024-01-17:
Joining the thread.  I had to commandeer my Pi4 for another project so bought a Pi5 8GB for OctoPi (figured dual cameras could be nice).
I'm about to download the 64bit nightly for 2024-01-15 21:52, will keep you posted how I get on.
I'm still awaiting long camera cables for the smaller fitting, so I won't have news on that front immediately I'm afraid.

apbarratt, 2024-01-17:
It seems to be working well for me so far and restored from my backup perfectly.
I'll try and test cameras tomorrow.

One thing I am noticing is that the install desktop script does not appear to have been updated in seven years.
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/home/pi/scripts/install-desktop

I'm quite certain I read when Pi OS bookworm was announced that there's been some fairly heavy work done in the desktop environment.  I do wonder if this install-desktop will need to be revisited to bring it up to spec with whatever PiOS is providing out of the box these day.

apbarratt, 2024-01-19:
New ribbon cables for cameras arrives last night so testing today.

@guysoft Is there anything in particular you're needing tested?
I see above the editing of the octopi.txt file for camera settings will be one thing to check.  Anything else?

On inspection of that install-desktop script, it looks like it's fairly universal for getting whatever Pi has released so you can likely ignore me on the notion that it might need updating :)

apbarratt, 2024-01-19:
hmmm, I'm having no luck getting the camera to show up.
```
pi@octopi:~ $ libcamera-hello --list-cameras
No cameras available!
```
I have one of these night vision 160° cameras with the OV thingy at the start of its sensor name.  I see @adam5zki is reporting a similar experience :/

Will quickly try the second port, but not looking good so far...

[image]

apbarratt, 2024-01-19:
Confirmed no different on the second camera port.  Not that I really expected it to make a difference :/

apbarratt, 2024-01-19:
Okay, I used `sudo nano /boot/config.txt` and changed
`camera_auto_detect=0` to `camera_auto_detect=1`
and rebooted.  Still no stream however:
pi@octopi:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/axi/pcie@120000/rp1/i2c@80000/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]

apbarratt, 2024-01-19:
Does the nightly come with the new camera stack?  I see in the downloads folder there were extra files involving imaging.  Perhaps I am meant to be doing something with those? 

[image]

apbarratt, 2024-01-19:
Hmmm, no, that ffmpeg seems to match what's already in the nightly.

apbarratt, 2024-01-19:
Hmmm, running `libcamera-jpeg -o test.jpg` confirms that the camera is in fact working (after I've added that `camera_auto_detect=1`, which I note is enabled by default on stock PiOS).
It's just not showing up in Octoprint and visiting http://octopi.local/webcam/?action=stream is presenting me with a `The webcam server is currently not running` page, with instructions on rebooting/switching to USB camera.
I'd thought once I got the auto-detect working it'd resolve but it seems not.  Most odd.

Here's the output of `sudo systemctl status webcamd.service`

```
● webcamd.service - the OctoPi webcam daemon with the user specified config
     Loaded: loaded (/etc/systemd/system/webcamd.service; enabled; preset: enabled)
     Active: active (running) since Fri 2024-01-19 16:03:28 GMT; 5h 1min ago
   Main PID: 750 (webcamd)
      Tasks: 2 (limit: 9262)
        CPU: 4.670s
     CGroup: /system.slice/webcamd.service
             ├─ 750 /bin/bash /root/bin/webcamd
             └─5363 sleep 120

Jan 19 21:02:00 octopi mjpg_streamer[5329]: MJPG-streamer [5329]: TV-Norm...........: DEFAULT
Jan 19 21:02:00 octopi mjpg_streamer[5329]: MJPG-streamer [5329]: init_VideoIn failed
Jan 19 21:04:01 octopi mjpg_streamer[5360]: MJPG-streamer [5360]: starting application
Jan 19 21:04:01 octopi mjpg_streamer[5360]: MJPG-streamer [5360]: MJPG Streamer Version.: 2.0
Jan 19 21:04:01 octopi mjpg_streamer[5360]: MJPG-streamer [5360]: Using V4L2 device.: /dev/video0
Jan 19 21:04:01 octopi mjpg_streamer[5360]: MJPG-streamer [5360]: Desired Resolution: 640 x 480
Jan 19 21:04:01 octopi mjpg_streamer[5360]: MJPG-streamer [5360]: Frames Per Second.: 10
Jan 19 21:04:01 octopi mjpg_streamer[5360]: MJPG-streamer [5360]: Format............: JPEG
Jan 19 21:04:01 octopi mjpg_streamer[5360]: MJPG-streamer [5360]: TV-Norm...........: DEFAULT
Jan 19 21:04:01 octopi mjpg_streamer[5360]: MJPG-streamer [5360]: init_VideoIn failed
```

Though with this being MJPG stuff, I'm a little confused as to where the new camera stack stuff is hiding or if it's even installed :/

apbarratt, 2024-01-19:
Okay, I've finally found some brief mention in another ticket that the nightlies do not include camera-streamer, though where they saw that written I've no idea.  So perhaps I'm just wasting my time.  Perhaps this was the wrong thread, it'd be nice to be told by someone if I am in the wrong place or if I'm just shouting into the wind :(

3driffs, 2024-01-24:
Desktop install fails on the latest nightly - after a reboot, rapid blinking cursor and nothing else.

3driffs, 2024-01-24:
In case others want to install the desktop, here's what worked for me:

Install the latest nightly of octopi
sudo tasksel
Select the default desktop (top option)
sudo apt install raspberrypi-ui-mods
raspi-config
Change it to launch desktop with password
reboot

And then you have a desktop running on pi 5/bookworm.  Note that RealVNC isn't supported, and TigerVNC has issues.

tsh-xx, 2024-02-03:
I didn't plan to be on the cutting edge, but I bought a Pi5 and it needs to be running octoprint for now.

I tired 2024-01-16_2023-12-11-octopi-bookworm-arm64-lite-1.1.0.zip but this complains about needing os_check==0, and not having the right dtb. I satisfied these two, and I get a blank screen. What am I doing wrong?

guysoft, 2024-02-05:
It might be that Rpi changed the device tree because they sourced different components for different Rpi 5 builds.

3driffs, 2024-02-25:
Just an update - Had one hard crash in the middle of a print, rebooted and it's been stable. But while octoprint is still working, the desktop is non-responsive.  Definitely bleeding edge here :-)

guysoft, 2024-02-25:
@3driffs People are reporting the nightly is working for them (though I didn't test it, this weekend was FullPagOS, will try another weekend to test OctoPi and prepare an RC)

3driffs, 2024-02-26:
Octoprint works- it’s the desktop that froze.  Have you updated the desktop install script that I missed?

guysoft, 2024-02-26:
I didn't update. If anyone has the time to figure out what needs to go there it would help me a lot! I don't really use that script and not sure what to expect. And rather focus on Octoprint working.

apbarratt, 2024-02-26:
A fair comment that 😄

doucettom, 2024-02-28:
Are the 64bits nightlies coming with the new camera stack ? TIA

guysoft, 2024-02-28:
@foosel AFAIK you are enforcing 32bit kernel on OctoPi-UpToDate. Where is the new camera stack image built? I could move it in to OctoPi's nightlies.

foosel, 2024-03-06:
I'm not sure you really want to move that into the nightlies. First of all I first need to swap the apt key (and that looks like it's harder than I thought at first, which is why I didn't manage to do it before my surgery last week from which I'm still recovering), there's still a stability issue with the camera-streamer server itself for which I'm waiting for feedback from upstream, and all in all I'd prefer to not have that forked at this point because that will make updates & support for me even trickier.

In case you want to look at it however: https://github.com/OctoPrint/OctoPi-UpToDate/tree/camera-streamer

guysoft, 2024-03-06:
Thanks for the info and hope you have a speedy recovery!

cp2004, 2024-03-07:
There is also the slightly added complication that the RPi 5 does not have the hardware encoder that `camera-streamer` was designed for - I am not sure what would happen if you tried to build it targeting the HW encoder and then ran that on a Pi 5. Maybe it would *just work*, I don't know. Adds another nice dimension to upgrading camera streaming setup in general.

lpla, 2024-04-25:
`ustreamer` project seems to be more mature than `camera-streamer` in this aspect: https://github.com/pikvm/ustreamer/issues/248

blair287, 2024-05-19:
Nighties just seem broken for me installed several different ones all boot connect to wifi but octoprint server never starts

guyshe-jfrog, 2024-05-26:
@blair287 Are you using the nightly from rpi imager? Where from?
The nighties should work.

guysoft, 2024-05-26:
@blair287 Are you using the nightly from rpi imager? Where from?
The nighties should work.

guysoft, 2024-06-04:
Related, wpa-supplicant option does not work anymore
https://github.com/guysoft/CustomPiOS/issues/219

geerlingguy, 2024-06-07:
@guysoft - I just installed the nightly on my Pi 5 (64 bit lite), `2024-06-07_2024-03-15-octopi-bookworm-arm64-lite-1.1.0.zip` from http://unofficialpi.org/Distros/OctoPi/nightly-arm64/ — it wasn't entirely clear from reading this thread, but it sounds like the Pi camera streaming functionality might not be working in this image quite yet? (At least on Pi 5.)

Right now would [running something like `go2rtc`](https://community.octoprint.org/t/octoprint-rpi5-pi-camera-v3/59391/3) be the best bet to pass through the Pi camera as a 'standard webcam' locally?

geerlingguy, 2024-06-07:
Sorry for the rubber-ducking, but I found out the image has `libcamera-apps-lite` installed, [not the full suite of `libcamera-apps`](https://github.com/raspberrypi/rpicam-apps/issues/489#issuecomment-1467304601). So I installed with:

```
sudo apt install lib camera-apps
```

And now it gets further... but I wind up seeing 'Unable to set controls: Device or resource busy', so I think something else is broken here. Maybe something in the image is still trying to grab the camera so `go2rtc` can't use it?

guysoft, 2024-06-09:
Ill try build with 
```
libcamera-apps
``` 
installed and we can test that nightly

jneilliii, 2024-07-02:
@geerlingguy ...first and foremost love your videos. curious, did you also disable the 

I have [these notes](https://gist.github.com/jneilliii/fb40f42d16887649b58699f002fe48ec) from when I tested out restreamer in the past, which lists the services I disable.

KruseLuds, 2024-07-21:
Noob trying to get Debian running on RPI5. Once you get to a stable version... If I am installing HA will it see "Debian" or "Octopi" (?)

guysoft, 2024-07-21:
@KruseLuds What do you mean installing HA?

KruseLuds, 2024-07-23:
HA = Home Assistant

guysoft, 2024-08-06:
Started working on a wpa-supplicant/networkmanager solution, moving slow but I have a plan. After that is done I think I am ready to release a RC using the beta branch of CustomPiOS v2

closed 2024-11-14 by guysoft

evanfoster, 2024-11-27:
Wooo! That's awesome news, thanks for doing this!