guysoft/OctoPi #824: netboot pxe boot issues with 32 bit nightly

Philius, 2023-12-30:
#### What were you doing?
Netbooting a nightly image - 2023-12-11-octopi-bookworm-armhf-lite-1.1.0

I've got several PIs running this way around the house.
The key is to rename the nfs directory to that of the CPU ID, this is what the PI provides when it requests files during netboot.
That way dnsmasq knows where to look.

#### What did you expect to happen?
Clean boot, octoprint service started

#### What happened instead?
The "boot" included user prompts so I needed a monitor and keyboard. This ended up not working as intended.
The octoprint service did not start.

I've attached an abbreviated **journalctl -b** output which will hopefully serve as a pointer to some issues.

For example I didn't expect to be prompted to rename a user and have no way to say "no thanks" or "maybe later".

#### Did the same happen when running OctoPrint in safe mode?
n/a

#### Version of OctoPi
The same as that included in the nightly image.

#### Printer model & used firmware incl. version
n/a

I have read the FAQ.

[journalctl.txt](https://github.com/guysoft/OctoPi/files/13798982/journalctl.txt)

guysoft, 2024-01-16:
The error is:
```
Dec 30 18:41:00 b5c83920 networkcheck[1762]: /usr/local/bin/networkcheck: line 3: /boot/octopi.txt: No such file or directory
Dec 30 18:41:00 b5c83920 networkcheck[1762]: /usr/local/bin/networkcheck: line 5: [: ==: unary operator expected
```
The issue is caused by this:
https://github.com/guysoft/OctoPi/issues/823#issuecomment-1892131703

Philius, 2024-01-25:
Ok, got past that part.

I just downloaded [2024-01-16_2023-12-11-octopi-bookworm-arm64-lite-1.1.0.zip](https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2024-01-16_2023-12-11-octopi-bookworm-arm64-lite-1.1.0.zip) and setup nfs and tftpboot.

Looking at the HDMI output it insists on renaming the "pi" user to something else.

I can't say "no thanks" so I provided "pi2" as the new user name.

It prompted me for the new password twice, which is good practice.

Then it told me that the user "pi2" doesn't exist and I'm back to renaming the "pi" user.

guysoft, 2024-06-10:
@Philius did you solve it in the end? There seems to be a bug when changing the default user. still investigating.

Philius, 2024-06-14:
I'm onto other work.
Maybe if it didn't insist on renaming the "pi" user then there would be no problem.
Were you able to reproduce the problem?
I went the masquerading route for dns but Jeff Geerling went the [frigate route](https://www.youtube.com/watch?v=7wkVGcdI2vk).
I just want to implement a 'pi security camera setup + microphones with maybe image processing on a zfs server that hosts dns and tftpboot.
It doesn't look like there's a standard or "preferred" way to do this that I can find.

Philius, 2024-06-15:
Sorry, brain fart.
What I meant to say was "masquerading route for dhcp" which lets the server tell the client about the real dhcp server via a tftp boot response packet.
And zfs over nfs.
The 'pi's provide their cpu id to tftpboot which greatly simplifys providing an image for that particular 'pi.
Not sure how this is done for x86 and amd64 clients.
Maybe frigate is the only option there?
I can totally see an amd64 server (in my case an 8 core laptop) have its camera trained on a 3D print farm.
Being able to "fire up" different distros on regular PCs by changing a couple of links in tftpboot would be handy too.
Maybe this is more of a Raspberry Pi foundation thing, which I totally get.

Philius, 2024-06-15:
ChatGPT has the answer for regular PCs: [https://chatgpt.com/share/dabae425-4f46-4b92-a8b6-0db92b0e5a3e](https://chatgpt.com/share/dabae425-4f46-4b92-a8b6-0db92b0e5a3e)

guysoft, 2024-06-18:
@Philius that chatgpt answer goes on about how to set an NFS server, but then misses the point of how to set a raspeberrypi to boot such a partition. Do you want your kernel to boot from a network or only the root folder (I don't think I have time to figure out how its done on a PI, just pointing out its not there as far as I can see).

Philius, 2024-06-19:
Thanks for taking a look.
I already have pi's pxe booting, that's where I ran into the problem I reported :)
I can merge my server setup with ChatGPTs insight or I could try getting the pi's their directory configs the same way as a regular PC.
My tftpboot and DHCP knowledge is limited.
I'm closer to the day where I would have the ability to manage a server farm with other cameras and microphones viewing the pi's printing.
That setup would be remarkably similar to a home security system.
Monitoring for other events and even an intercom between an operator and the admin would only require microphones on the pi's running octoprint.
Then the issue of "who owns the camera" becomes an issue for timelapses, although that could be replaced by security system feature.
I don't know enough, home assistant? Multiple OBS instances on the server? WebRTC?
I'm happy to help, but more in terms of following instructions although of course I could provide some design input.