guysoft/OctoPi #825: libcamera-hello --list-cameras returns ERROR: rpicam-apps currently only supports the Raspberry Pi platforms on nightly build

b-morgan, 2024-01-06:
#### What were you doing?

1. Download and image 2024-01-04_2023-12-11-octopi-bookworm-armhf-lite-1.1.0.zip
2. Install on RPi 4
3. Type `libcamera-hello --list-cameras`

#### What did you expect to happen?
```
Available cameras
-----------------
0 : imx708 [4608x2592 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]
```

#### What happened instead?

`ERROR: rpicam-apps currently only supports the Raspberry Pi platforms on nightly build`

#### Did the same happen when running OctoPrint in safe mode?

Safe mode doesn't apply as this has nothing to do with OctoPrint

#### Version of OctoPi

2024-01-04_2023-12-11-octopi-bookworm-armhf-lite-1.1.0.zip

#### Printer model & used firmware incl. version

No physical printer installed.

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.
Yes

guysoft, 2024-01-15:
There seems to be some workaround here:
https://forums.raspberrypi.com/viewtopic.php?p=2163959#p2163959

Does that solve your issue?
@foosel Do we need to change the defaults in config.txt?

b-morgan, 2024-01-15:
The error message is misleading as it is displayed on an official Raspberry Pi 4B platform (unless I have a fake). If the error message is being produced by code coming from the official Raspberry Pi OS bookworm release, then there's probably nothing OctoPi can do about it.

As imaged, `/boot/config.txt` contains `camera_auto_detect=0` and `libcamera-hello` returns:
```
pi@rpi4b:~ $ libcamera-hello --list-cameras
ERROR: rpicam-apps currently only supports the Raspberry Pi platforms.
Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps.
```
Change `/boot/config.txt` to `camera_auto_detect=1` and `libcamera-hello` returns:
```pi@rpi4b:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx708 [4608x2592 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]
```
So, IMO, `/boot/config.txt` should probably be changed or some documentation for the user to make the change.

Since the nightly builds do not include `camera-streamer` I can't go any further with testing OctoPi 1.1.0

b-morgan, 2024-01-15:
I imaged and booted the "official" Raspberry PI OS (32-BIT) and `/boot/config.txt` contains `camera_auto_detect=1`. This is even stronger evidence that the OctoPi 1.1.0 image should include `camera_auto_detect=1` as well.

b-morgan, 2024-01-15:
I'm sure you will be on a machine that can do it long before I could figure out how to do a PR 😁

guysoft, 2024-11-22:
I am really swamped, if you can PR and have the time please do