guysoft/OctoPi #826: Swap size of 100MB to small to render Timelapse on Pi Zero 2

thomaslazar, 2024-01-14:
#### What were you doing?

On a Raspi Zero 2 I tried to render a Timelapse and got an error "Rendering of Timelapse failed with return code -9" same as described here: https://github.com/OctoPrint/OctoPrint/issues/4447

#### What did you expect to happen?

Timelapse being rendered and offered for download. 

#### What happened instead?

Error was thrown. 

#### Version of OctoPi

OctoPrint 1.9.3 Python 3.9.2 OctoPi* 1.0.0cam (build 2023.10.09.154319)

#### Workaround 

As describe in https://github.com/OctoPrint/OctoPrint/issues/4447#issuecomment-1604272892 extending the swap size from 100MB to 512MB solved the issue. It would be nice if that would be changed out of the box if that doesn't break anything else. Can't really assess that.

guysoft, 2024-01-15:
Hey,
there is no official Pi Zero 2 build (yet). 
There is an alpha branch I am working on for multi device build. Did you build it yourself? what base image did you use?

cp2004, 2024-01-15:
Quite a few people use the Pi Zero 2 - I believe it was compatible with OctoPi straight away on release because it uses the same chip as one of the Pi 3 models.

thomaslazar, 2024-01-15:
I used the raspberry pi imager as described in the docs on the official site, selected the pi zero 2 and got offered an image to write to the sd card.

kiwi-x, 2024-04-27:
I had the same issue on a 3B+
apt upgrade with kernel updates didn't ran and I had to reboot the raspberry pi.

After resizing swap size to 1024 anything worked

See
https://pimylifeup.com/raspberry-pi-swap-file/

-- kiwix