guysoft/OctoPi #829: [BUG] public key is not available

cminton-hub, 2024-03-22:
#### What were you doing?

Tried to install influxdb2 on an octopi image to only have one pi running in the household.

1. Sudo apt update && sudo upgrade
2. wget -q https://repos.influxdata.com/influxdata-archive_compat.key
3. echo '393e8779c89ac8d958f81f942f9ad7fb82a25e133faddaf92e15b16e6ac9ce4c influxdata-archive_compat.key' | sha256sum -c && cat influxdata-archive_compat.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/influxdata-archive_compat.gpg > /dev/null
4. echo 'deb [signed-by=/etc/apt/trusted.gpg.d/influxdata-archive_compat.gpg] https://repos.influxdata.com/debian stable main' | sudo tee /etc/apt/sources.list.d/influxdata.list
5. sudo apt-get update && sudo apt-get install influxdb2

#### What did you expect to happen?

#### What happened instead?
`Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease                                                         
Get:3 https://apt.octoprint.org/debian bullseye InRelease [4,067 B]                                                       
Hit:4 https://repos.influxdata.com/debian stable InRelease                
**Err:3 https://apt.octoprint.org/debian bullseye InRelease**
  **The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 418208F07AD011A2**
Reading package lists... Done
**W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://apt.octoprint.org/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 418208F07AD011A2
W: Failed to fetch https://apt.octoprint.org/debian/dists/bullseye/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 418208F07AD011A2**
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package influxdb2`

#### Did the same happen when running OctoPrint in safe mode?

Doesn't matter.

#### Version of OctoPi

See supportbundle (system info zip)

#### Printer model & used firmware incl. version

Doesn't matter

#### Screenshot(s)/video(s) showing the problem:

See PNG.

I have read the FAQ.

cminton-hub, 2024-03-22:
There is also no date on ubuntu keyserver.

[image]

shofeur, 2024-04-22:
There is a possible solution to this problem (worked for me) posted on the OctoPrint forum @ [https://community.octoprint.org/t/apt-update-doesnt-work-and-getting-no-public-key/57618](url)

Hope this helps,
Steve

cminton-hub, 2024-04-22:
@shofeur Thank you for pointing out the solution. I have been looking for the solution for hours. The question in the forum was asked after my search and so I couldn't find it. Apparently I was one of the first to not be able to solve the problem myself :D
SOLUTION: `curl -s --compressed "https://apt.octoprint.org/octoprint.gpg.key" | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/octoprint.gpg > /dev/null`
[https://apt.octoprint.org/](url)

closed 2024-04-22 by cminton-hub

shofeur, 2024-04-22:
Not the only one ;) I just was stubborn enough to look yesterday and someone finally had posted something that worked! Glad I could help.

Steve