guysoft/OctoPi #833: [Request] trying to build octopi with Longan Pi3h custom image

elasticdotventures, 2024-07-07:
I'm trying to get OctoPi working on a LonganPi 3b with a custom ubuntu image. 
https://github.com/sipeed/LonganPi-3H-SDK/issues/17

I've forked, and created a custom ubuntu image with the LonganPi images and i've created custom action(s) that build the ubuntu image and then trigger a workflow to build my octopi fork.  https://github.com/elasticdotventures/OctoPi-longan-pi3h

the build for the ubuntu found here:
https://github.com/elasticdotventures/LonganPi-3H-SDK/tree/feature/octoprint

I'm running into an issue where guysoft/CustomPIoS & specifically OctoPi seems to require the image to be in a NOOBS file system format.  (reference https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained)

I'm asking for guidance on the best way to proceed to get octopi working on a non rpi noobs image, and then eventually upstream that into octopi and *hopefully* any support into custompios so that it's possible to use a much cheaper & smaller longan pi3h instead of rpi.  
  
I read in the docs that octopi and custompios support armbian distro which afaik doesn't use NOOBS format but it's not obvious to me how to enable that workflow.  

```
BUILD STARTED @ Sun Jul 7 07:00:16 UTC 2024!

+++ echo -e -n '\e[0m'
+++ '[' -n 2000 ']'
+++ which sfdisk
+++ mkdir -p /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/workspace
+++ mkdir -p /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/workspace/mount
+++ rm -rf '/home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/workspace/*.tar.gz'
+++ install_cleanup_trap
+++ set -e
+++ trap cleanup SIGINT SIGTERM
+++ install_fail_on_error_trap
+++ set -e
+++ trap 'echo_red "build failed, unmounting image..." && cd $DIST_PATH && ( unmount_image $BASE_MOUNT_PATH force || true ) && echo_red -e "\nBUILD FAILED!\n"' ERR
+++ unmount_image /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/workspace/mount force
+++ mount_path=/home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/workspace/mount
+++ force=
+++ '[' 2 -gt 1 ']'
+++ force=force
+++ sync
+++ '[' -n force ']'
++++ sudo lsof -t /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/workspace/mount
++++ sudo mount
++++ grep /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/workspace/mount
++++ awk '{print $1}'
++++ sort -r
++++ awk -F ' on ' '{print $2}'
+++ pushd /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/workspace
/home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/workspace /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src
+++ '[' -e '*.img' ']'
+++ '[' '!' -f /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/image/sdcard-raspios.img.xz ']'
+++ '[' /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/image/sdcard-raspios.img.xz == '' ']'
+++ [[ /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/image/sdcard-raspios.img.xz =~ \.img$ ]]
+++ 7za x -aoa /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/image/sdcard-raspios.img.xz

7-Zip (a) [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=C,Utf16=off,HugeFiles=on,64 bits,4 CPUs AMD EPYC 7763 64-Core Processor                 (A00F11),ASM,AES-NI)

Scanning the drive for archives:
1 file, 226153328 bytes (216 MiB)

Extracting archive: /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/image/sdcard-raspios.img.xz
--
Path = /home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/repository/src/image/sdcard-raspios.img.xz
Type = xz
Physical Size = 226153328
Method = LZMA2:26 CRC64
Streams = 1
Blocks = 16

Everything is Ok

Size:       3154116[608](https://github.com/elasticdotventures/LonganPi-3H-SDK/actions/runs/9825382432/job/27125472670#step:10:609)
Compressed: 226153328
++++ ls
++++ grep '.img$\|.raw$'
++++ head -n 1
+++ BASE_IMG_PATH=sdcard-raspios.img
+++ '[' '!' -f sdcard-raspios.img ']'
++++ basename sdcard-raspios.img
+++ export CUSTOM_PI_OS_BUILDBASE=sdcard-raspios.img
+++ CUSTOM_PI_OS_BUILDBASE=sdcard-raspios.img
+++ '[' -n 2000 ']'
+++ enlarge_ext sdcard-raspios.img 2 2000
+++ image=sdcard-raspios.img
+++ partition=2
+++ size=2000
+++ echo_green 'Adding 2000 MB to partition 2 of sdcard-raspios.img'
+++ echo -e -n '\e[92m'
+++ echo Adding 2000 MB to partition 2 of sdcard-raspios.img
Adding 2000 MB to partition 2 of sdcard-raspios.img
+++ echo -e -n '\e[0m'
++++ sfdisk --json sdcard-raspios.img
++++ jq '.partitiontable.partitions[] | select(.node ==  "sdcard-raspios.img2").start'
+++ start=
/home/runner/work/LonganPi-3H-SDK/LonganPi-3H-SDK/CustomPiOS/src/common.sh: line 282: *512: syntax error: operand expected (error token is "*512")
+ exit 1
```

guysoft, 2024-07-07:
Hey @elasticdotventures .
Will try and help you out. though you might need to adapt stuff.
Can you post the partition table of the image? Or a link so I can download it and take a look?

I have a bet its likely that there are no 2 partitions like in the Pi image. First is boot and the second the root.
Its defined here:
https://github.com/guysoft/CustomPiOS/blob/feature/meta-modules/src/modules/base/config#L65

However, you might want to go though a different much more tested route.

There are two methods also to build an ubuntu image. It was built for the Pi image but is much closer to what you are trying to build. There is an OctoPi [ubuntu_arm64 variant](https://github.com/guysoft/OctoPi/tree/devel/src/variants/ubuntu_arm64) you might want to take a look at. It might solve some issues you are having.

The second one is in the works. I am working on CustomPiOS v2 where you can define boards and it should work better in the future.
Its WIP, you can follow it here:
https://github.com/guysoft/CustomPiOS/issues/214

I don't have for that a just run X yet though.

elasticdotventures, 2024-07-09:
I think there's actually a problem with my image build.  I'll follow guysoft/custompios#214 and let that stabilize.

guysoft, 2024-07-09:
It might take a while, how long do you have?